realsenseai/librealsense #12262: UE5 Plugin Release

ffreality, 2023-10-05:
### Issue Description
After license changing of Unreal Engine for non-game projects, we decided to add other engines to our stack to secure ourselves.
For this reason, there is no need to hide this plugin. We just want to solve our problems and use that know-how on Open3DEngine as soon as possible.

https://github.com/FF-Plugins-Active/FF_RealSense/
You can access it from this link.

**Details about it.**
- We are not realated with Intel or RealSense product family.
- We just created a plugin to get RGB, Depth, Infrared Streams and Distance with D435i and D455. We don't have other cameras and we don't have a plan about buying them.
- We used C library because when we use C++ library, we errors from Visual Studio.
- We refactored it with FRunnableThread. We didn't try its Android compability but our main goal is supporting Android devices because Meta/Oculus Quest support. Before we refactor it, it worked with Android.
- We support only current engine versions. We update its engine compability after one month of new Epic Launcher Release.
(So, currently it supports only 5.3.X and newers. After one month of 5.4 release, it will only support 5.4)
- We are not responsible with librealsense related problems. We just expose "C" library to blueprints.
- Our development platform is Windows. We don't and won't use Linux or Mac.
- This is not a actor-drag-drop plugin which you can use it easily. If you are not proficient with blueprints, you can have hard time to use it.

- We integrated ZXing based QR decoder. But in order to use, you have two options
1. You have to buy a specific plugin for that. we shared its link in our repo. We are not related with ZXing plugin's developer but we use it as standalone plugin for our other projects. So, we don't want library clashes and we just include it as module to our plugin. 
2. If you don't want to buy it, you can build ZXing for yourself and change .build.cs and .uplugin to integrate it.
Decode functions and C++ includes are not related with plugin. So, you don't have to change codes. (It is in Rs_Thread.cpp)

ffreality, 2023-10-05:
As we know realsense D series don't have auto focus and some developers (I think you) suggested increasing "Gain". If we have a sample about it with C API, we can implement it.

We also tried to use OpenCV but UE doesn't package for Android when we use it. Because it causes a typedef clash with AndroidPlatform.h

Other than that it will be very nice to have if we can solve point cloud.

There are some runtime importers for UE, too. This can opens us some realworld scene inspection feature.

After learning Open3DEngine, we will develop one plugin for it, too.

MartyG-RealSense, 2025-05-11:
Closed due to no further comments since October 2023.

closed 2025-05-11 by MartyG-RealSense