realsenseai/librealsense #12263: Pyrealsense rs.align API got a bad performance on Raspberry pi 4

gary88888888, 2023-10-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version |   Raspbian 64-bit Desktop | 
| Kernel Version (Linux Only)    |  6.1| 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            |  2.54.2                           | 
| Language                            |  python }                          | 
| Segment			|  others                           | 

### Issue Description
Hi Dear mentainers, I follow this tutorials(https://github.com/datasith/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435) to install pyrealsense on my raspberry pi 4 device, then all the package is working fine, except the align function, i have been test the code with align and without align, it has bad performance apparently in code with align (around 6 fps), 28~29 fps in code without align, could someone tell me what the problem is, any solutions ?
here is my code 
```
import sys
import time
import cv2
import numpy as np
import pyrealsense2 as rs
from PyQt5.QtWidgets import QApplication, QMainWindow, QLabel, QVBoxLayout, QWidget
from PyQt5.QtGui import QImage, QPixmap
from PyQt5.QtCore import Qt, QTimer

class DepthCameraApp(QMainWindow):
    def __init__(self):
        super().__init__()

        self.initUI()

    def initUI(self):
        self.setWindowTitle("Depth and RGB Camera App")
        self.setGeometry(100, 100, 1280, 480)

        self.central_widget = QWidget(self)
        self.setCentralWidget(self.central_widget)

        self.layout = QVBoxLayout()
        self.central_widget.setLayout(self.layout)
        self.fps_label = QLabel(self)
        self.layout.addWidget(self.fps_label)

        self.depth_label = QLabel(self)
        self.rgb_label = QLabel(self)
        self.layout.addWidget(self.depth_label)
        self.layout.addWidget(self.rgb_label)

        self.timer = QTimer(self)
        self.timer.timeout.connect(self.update_frames)
        self.timer.start(1000 // 30)  # 30 FPS
        self.last_time = 0
        self.pipeline = rs.pipeline()
        self.config = rs.config()
        self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
        self.config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
        
        self.colorizer = rs.colorizer()

        self.profile = self.pipeline.start(self.config)
        depth_profile = self.profile.get_stream(rs.stream.depth)
        color_profile = self.profile.get_stream(rs.stream.color)

        self.align_to = rs.stream.color
        self.align = rs.align(self.align_to)

    def update_frames(self):
 
        frames = self.pipeline.wait_for_frames()
        self.aligned_frames = self.align.process(frames)
        depth_frame = self.aligned_frames.get_depth_frame()
        color_frame = self.aligned_frames.get_color_frame()

        if depth_frame and color_frame:

            # Convert depth frame to numpy array
            depth_image = np.asanyarray(self.colorizer.colorize(depth_frame).get_data())

            # Convert color frame to numpy array
            color_image = np.asanyarray(color_frame.get_data())

            # Calculate and display FPS 
            current_time = time.time()
            self.fps_label.setText(f"FPS: {int(1 / (current_time - self.last_time))}")
            self.last_time = current_time

 
            # Create QImage from numpy arrays
            depth_image = cv2.cvtColor(depth_image, cv2.COLOR_BGR2RGB)
            color_image = cv2.cvtColor(color_image, cv2.COLOR_BGR2RGB)

            depth_height, depth_width, _ = depth_image.shape
            color_height, color_width, _ = color_image.shape

            bytes_per_line_depth = 3 * depth_width
            bytes_per_line_color = 3 * color_width

            q_depth_image = QImage(depth_image.data, depth_width, depth_height, bytes_per_line_depth, QImage.Format_RGB888)
            q_color_image = QImage(color_image.data, color_width, color_height, bytes_per_line_color, QImage.Format_RGB888)

            # Display the QImage
            self.depth_label.setPixmap(QPixmap.fromImage(q_depth_image))
            self.rgb_label.setPixmap(QPixmap.fromImage(q_color_image))

    def closeEvent(self, event):
        self.pipeline.stop()
        event.accept()

if __name__ == '__main__':
    app = QApplication(sys.argv)
    window = DepthCameraApp()
    window.show()
    sys.exit(app.exec_())

```

MartyG-RealSense, 2023-10-06:
Hi @gary88888888  Slow performance on Pi boards during depth-color alignment is a known issue.  So it is likely that your code is fine and the Pi hardware is the cause.  You may be able to improve performance if you use 15 FPS for the streans instead of 30.

gary88888888, 2023-10-06:
@MartyG-RealSense Thank you for your response. So, if I switch to X86 platform board, will it resolve this issue, or would I also need a GPU card to avoid this problem?

also i tried to use 15 FPS, and it didn't improve FPS

MartyG-RealSense, 2023-10-06:
I cannot make guarantees about performance on a particular computer but x86 is less likely to have alignment lag than Pi, yes.  

The integrated graphics that is built into modern CPU chips such as Intel Core is usually sufficient for processing alignment.

gary88888888, 2023-10-11:
@MartyG-RealSense  thank you so much, i will try on x86 board(up 4000 board), when the result comes out, i will let you know

MartyG-RealSense, 2023-10-11:
You are very welcome.  I look forward to your next update.  Good luck!

gary88888888, 2023-10-18:
@MartyG-RealSense  hi ~ I have tested it on UP4000 X86 board, and it works perfect
this board's CPU is intel Celeron N3350 with 4 GB ram, the FPS can reach 30 FPS

[image]

MartyG-RealSense, 2023-10-23:
Hi @gary88888888  Do you require further assistance with this case, please?  Thanks!

gary88888888, 2023-10-23:
> Hi @gary88888888  Do you require further assistance with this case, please?  Thanks!

For now there is no problem, thank u 👍

closed 2023-10-23 by gary88888888