realsenseai/librealsense #12264: Example of IR stereo image output and IMU

Userpc1010, 2023-10-06:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D455 } | 
| Firmware Version                   | (5.15.0.2) | 
| Operating System & Version |   { Linux (Ubuntu 20.04)}   | 
| Kernel Version (Linux Only)    |  (5.15.0-83-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { v 2.54.1 }                          | 
| Language                            |  {C/C++/opencv/ }                          | 
| Segment			|  { others }                          | 

### Issue Description

I'm trying to output IR stereo image and imu data with OpenCV imshow, but I only get a dark screen:
```

#include <signal.h>
#include <stdlib.h>
#include <iostream>
#include <algorithm>
#include <fstream>
#include <chrono>
#include <ctime>
#include <sstream>

#include <condition_variable>

#include <opencv2/opencv.hpp>

#include <librealsense2/rs.hpp>
#include "librealsense2/rsutil.h"

using namespace std;

bool b_continue_session;

void exit_loop_handler(int s){
    cout << "Finishing session" << endl;
    b_continue_session = false;

}

void interpolateData(const std::vector<double> &vBase_times,
                     std::vector<rs2_vector> &vInterp_data, std::vector<double> &vInterp_times,
                     const rs2_vector &prev_data, const double &prev_time);

rs2_vector interpolateMeasure(const double target_time,
                              const rs2_vector current_data, const double current_time,
                              const rs2_vector prev_data, const double prev_time);

static rs2_option get_sensor_option(const rs2::sensor& sensor)
{
    // Sensors usually have several options to control their properties
    //  such as Exposure, Brightness etc.

    std::cout << "Sensor supports the following options:\n" << std::endl;

    // The following loop shows how to iterate over all available options
    // Starting from 0 until RS2_OPTION_COUNT (exclusive)
    for (int i = 0; i < static_cast<int>(RS2_OPTION_COUNT); i++)
    {
        rs2_option option_type = static_cast<rs2_option>(i);
        //SDK enum types can be streamed to get a string that represents them
        std::cout << "  " << i << ": " << option_type;

        // To control an option, use the following api:

        // First, verify that the sensor actually supports this option
        if (sensor.supports(option_type))
        {
[... 422 lines cut ...]
        Mat imRightCV = cv::Mat(cv::Size(width_img, height_img), CV_8U, (void*)(ir_frameR.get_data()), cv::Mat::AUTO_STEP);

        // Update the window with new data
        cv::imshow("img_l", imCV);
        cv::imshow("img_r", imRightCV);

    }

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}

```
[image]

I can't figure out what I'm doing wrong? I didn't find an example for stereo + imu so I turned here.

MartyG-RealSense, 2023-10-07:
Hi @Userpc1010  The RealSense SDK has a C++ example program called **rs-data-collect** that streams depth, infrared and color and automatically additionally streams IMU data from camera models that are equipped with an IMU.

https://github.com/IntelRealSense/librealsense/tree/master/tools/data-collect

Another approach for streaming IMU data at the same time as other stream types such as infrared is to use a **callback**, as described in the C++ discussion at https://github.com/IntelRealSense/librealsense/issues/6426

Userpc1010, 2023-10-07:
I found out what the problem was, it was missing: `cv::waitKey(1);` very important additions I believe. 
Thank you for your time, we can close!

MartyG-RealSense, 2023-10-08:
You are very welcome.  I'm pleased to hear that you achieved a solution.  As you suggested, I will close this case.  Thanks again!

closed 2023-10-08 by MartyG-RealSense