realsenseai/librealsense #12267: [D415] Does the IR data I obtained contain visible brand information?

thraster, 2023-10-09:
Hi there,
Currently I was trying to use the D415 IR data for Face anti-spoofing. I noticed that the depth module actually is consist of two imager without IR filter. This means the left imager can provied IR data. 

What I'm curious about is that: 
_**Does the IR data I obtained contain information from the visible brand?**_ 

Since the left imager can also return the RGB data.
_**Does the IR data is generated simply by RGB2GRAY using the RGB data from left imager? 
or just extract the red channel of RGB data from left imager as IR data?**_

From the datasheet, we know that left imager don't have a IR filter, but 
**_is there any special design on its bayer filter? 
Does the left imager using the RGGB or some special designed bayer filter like RGBI(infrared)?_**

Thank you very much!

MartyG-RealSense, 2023-10-09:
Hi @thraster  The infrared image does not use RGB data at all.  The left sensor is capable of outputting an RGB image from the left infrared imager on certain RealSense camera models (D405, D415, D455) but that is a separate mode from the infrared image.

The left and right imagers can provide an infrared image.  Depth frames are constructed using images from both infrared imagers, which is why D415 is called a 'stereo' depth camera.

At https://github.com/IntelRealSense/librealsense/issues/1070#issuecomment-366405803 a RealSense team member advises that the D415 has a Bayer pattern filter to extract color but it does not block infrared.

thraster, 2023-10-10:
Hi @MartyG-RealSense , Thanks for your quick response. It's quite useful!

I'd like to know that the _**IR data from the left infrared imager is the pure IR data? Dose it contains the information from the visible wavelength?**_  

If it's possible, I'd like to know the **_how the IR data from the left infrared imager is generated?_** 

Because currently I am doing some research on given wavelength, I want to know the spectral band of  the IR data I obtained.

Thanks a lot!

MartyG-RealSense, 2023-10-10:
The infrared sensors of the D415, D435, D435i and D455 can see all visible wavelengths, including visible and near-infrared.

'f' models with a near-infrared pass filter such as D435f, D435if and D455f can allow near-infrared frequencies to enter into the sensor but block the lower visible-light frequencies.

The D405 has an IR Cut filter that can see visible light frequencies but blocks near-IR frequencies, making them invisible to the camera.

Intel have a guide to using optical filters with RealSense cameras and how different types of filters affect the frequencies.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400

The infrared sensors capture an infrared frame and if the Y8 infrared stream format is selected then it is adjusted (**rectified**) by the Vision Processor D4 hardware inside the camera before being transmitted along the USB cable to the computer.  If the Y16 infrared format is selected then the infrared image is not rectified because Y16 is used for camera calibration.

If the camera's IR projector component is enabled then a semi-random dot pattern is overlaid onto the infrared image.  The camera uses this dot pattern, which is cast onto surfaces in the real-world scene, to help to analyze the surfaces for depth information by using the dots as a readable surface texture.

thraster, 2023-10-11:
Hi @MartyG-RealSense ,

So the IR data I received from the left IR imager of D415 by using the realsense SKD is actually including **visible and near-infrared?**

Dose the adjustment you mentioned of the Y8 IR stream contains any modification of the spectrum band?

Thank you!

MartyG-RealSense, 2023-10-11:
Yes, with the D415 camera model it is possible for the camera to use light in visible or near-infrared frequencies ranging from 400-1000 nm as its IR sensors are not equipped with a filter.  Rectification adjustment will not affect the frequencies.  The link below describes what is involved in rectification.

https://en.wikipedia.org/wiki/Image_rectification

thraster, 2023-10-11:
Well, thank you @MartyG-RealSense , This really helps me better undersanding it.

And by the way I want to ask that what's the relationship between the **RGB data from the left IR imager** and the **IR data from the left IR imager**?

Are they generated from two different pipelines? Like RAW2RGB and RAW2IR?

MartyG-RealSense, 2023-10-11:
When RGB from the left IR imager is enabled, the left IR image cannot be accessed.  This is likely because although it is an RGB image, the infrared stream is being used and you cannot have two of the same infrared stream active at the same time.

Changing the **Color Correction** values in the Advanced Controls section of the Viewer changes the colors of the RGB from left IR image.  The left IR image is essentially a colorized infrared view.

The Y8 and RGB8 (RGB from left infrared) views can be compared below.

**Y8**

[image]

**RGB8**

[image]

More information about color from left IR can be found in the section of Intel's camera tuning guide linked to below.

https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance#use-the-left-color-camera

thraster, 2023-10-11:
Ok, I will do some experiments to find out more. Thank you for your help!