realsenseai/librealsense #12268: Timestamp: 1696831425043.935303 Severity: Error Description: Motion Module failure

Rocky-YU, 2023-10-09:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435I|
|**Serial Number**|243322070802|
|**Firmware Version**|5.15.0.2|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|322743060569|
|**Firmware Update Id**|322743060569|

Please provide a description of the problem

Rocky-YU, 2023-10-09:
09/10 14:21:48,342 ERROR [545578274560] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 09/10 14:21:48,353 INFO [547978225792] (uvc-streamer.cpp:28) endpoint 130 read buffer size: 815104
 09/10 14:21:48,357 WARNING [546584907520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/10 14:21:48,368 WARNING [546584907520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/10 14:21:48,376 ERROR [545578274560] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 09/10 14:21:48,677 WARNING [546584907520] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Resource temporarily unavailable, number: 11
 09/10 14:21:50,059 INFO [547978225792] (frame-archive.h:193) The user was holding on to 7 frames after stream 0x0x55a5172ac0 stopped
 09/10 14:21:50,059 INFO [547978225792] (frame-archive.h:193) The user was holding on to 7 frames after stream 0x0x55a5172ac0 stopped
 09/10 14:21:50,059 INFO [547978225792] (frame-archive.h:193) The user was holding on to 7 frames after stream 0x0x55a5172ac0 stopped
 09/10 14:21:50,059 INFO [547978225792] (frame-archive.h:193) The user was holding on to 7 frames after stream 0x0x55a5172ac0 stopped
 09/10 14:21:55,861 INFO [547978225792] (ds-calib-parsers.cpp:236) Depth Module V2 IMU Accelno valid intrinsic available, use default values.
 09/10 14:21:55,861 INFO [547978225792] (ds-calib-parsers.cpp:248) Depth Module V2 IMU Gyrointrinsic not valid, use default values.
 09/10 14:21:55,862 INFO [547978225792] (sensor.cpp:1594) Request: MOTION_XYZ32F Accel, 
Resolved to: MOTION_XYZ32F Accel, 
 09/10 14:21:55,863 INFO [547978225792] (ds-calib-parsers.cpp:236) Depth Module V2 IMU Accelno valid intrinsic available, use default values.
 09/10 14:21:55,863 INFO [547978225792] (ds-calib-parsers.cpp:248) Depth Module V2 IMU Gyrointrinsic not valid, use default values.
 09/10 14:21:55,864 INFO [547978225792] (sensor.cpp:1594) Request: MOTION_XYZ32F Gyro, 
Resolved to: MOTION_XYZ32F Gyro,

MartyG-RealSense, 2023-10-09:
Hi @Rocky-YU  Does the motion module failure still occur if you add a camera reset instruction to your script, please?  C++ and Python code for doing so can be found at https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

closed 2023-10-11 by Rocky-YU