realsenseai/librealsense #12277: cannot update even though connected to Ethernet

monajalal, 2023-10-10:
| | |
|---|---|
|**librealsense**|2.50.12 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|819112070232|
|**Firmware Version**|05.13.00.50|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|D400|
|**Asic Serial Number**|815213022437|
|**Firmware Update Id**|815213022437|

Please provide a description of the problem

[image]

(base) mona@mona-ThinkStation-P7:~$ uname -a
Linux mona-ThinkStation-P7 6.2.0-34-generic #34~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 13:12:03 UTC 2 x86_64 x86_64 x86_64 GNU/Linux

(base) mona@mona-ThinkStation-P7:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

MartyG-RealSense, 2023-10-10:
Hi @monajalal  It looks as though you are using the Viewer for the FRAMOS industrial 'e' range of cameras such as D435e but have an Intel RealSense D435 camera attached.  The FRAMOS software was designed for their own range of RealSense-compatible cameras with an ethernet connector and not Intel's cameras.

If a FRAMOS D435e camera was attached then the Viewer would look this this:

[image]

Intel RealSense cameras should be used with Intel's own software, which can be installed from packages using the instructions at the link below:

https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

Or compiled from source code with these instructions:

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

It is possible to use Intel's version of the Viewer with an ethernet connection by building the software from source code with networking support.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras

This ethernet tool is only supported up until version 2.53.1 of the RealSense software though, as it has now been removed as it is due to be replaced by a new ethernet networking interface in an upcoming software release.

monajalal, 2023-10-10:
@MartyG-RealSense thanks for the link. When I followed the link you shared, I got this error:

```
(base) mona@mona-ThinkStation-P7:~$  sudo apt-get install librealsense2-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  chromium-codecs-ffmpeg-extra cuda-cccl-11-7 cuda-command-line-tools-11-7 cuda-compiler-11-7 cuda-cudart-11-7 cuda-cudart-dev-11-7 cuda-cuobjdump-11-7 cuda-cupti-11-7 cuda-cupti-dev-11-7
  cuda-cuxxfilt-11-7 cuda-documentation-11-7 cuda-driver-dev-11-7 cuda-gdb-11-7 cuda-libraries-11-7 cuda-libraries-dev-11-7 cuda-memcheck-11-7 cuda-nsight-11-7 cuda-nsight-compute-11-7
  cuda-nsight-systems-11-7 cuda-nvcc-11-7 cuda-nvdisasm-11-7 cuda-nvml-dev-11-7 cuda-nvprof-11-7 cuda-nvprune-11-7 cuda-nvrtc-11-7 cuda-nvrtc-dev-11-7 cuda-nvtx-11-7 cuda-nvvp-11-7 cuda-sanitizer-11-7
  cuda-toolkit-11-7 cuda-toolkit-11-7-config-common cuda-toolkit-11-config-common cuda-toolkit-config-common cuda-tools-11-7 cuda-visual-tools-11-7 docker-scan-plugin ffmpeg gds-tools-11-7
  gstreamer1.0-vaapi libcublas-11-7 libcublas-dev-11-7 libcufft-11-7 libcufft-dev-11-7 libcufile-11-7 libcufile-dev-11-7 libcurand-11-7 libcurand-dev-11-7 libcusolver-11-7 libcusolver-dev-11-7
  libcusparse-11-7 libcusparse-dev-11-7 libdvdnav4 libegl-mesa0:i386 libegl1:i386 libflashrom1 libftdi1-2 libgbm1:i386 libgles2:i386 libgstreamer-plugins-bad1.0-0 libllvm13 liblua5.2-0 libmujs1
  libnpp-11-7 libnpp-dev-11-7 libnvjpeg-11-7 libnvjpeg-dev-11-7 libopengl0:i386 libplacebo192 libsixel1 libva-wayland2 libwayland-client0:i386 libwayland-server0:i386 mpv nsight-compute-2022.2.0
  nsight-systems-2022.1.3 python3-pycryptodome python3-pyxattr rtmpdump
Use 'sudo apt autoremove' to remove them.
The following additional packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libblkid-dev libcairo2-dev libdatrie-dev libdbus-1-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglib2.0-dev libglib2.0-dev-bin
  libgraphite2-dev libgtk-3-dev libharfbuzz-dev libharfbuzz-gobject0 libmount-dev libpango1.0-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpixman-1-dev librealsense2 librealsense2-gl
  libselinux1-dev libsepol-dev libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev libxcursor-dev libxdamage-dev libxfixes-dev libxi-dev libxinerama-dev libxkbcommon-dev libxtst-dev
  pango1.0-tools wayland-protocols
Suggested packages:
  libcairo2-doc libdatrie-doc libgirepository1.0-dev libglib2.0-doc libxml2-utils libgraphite2-utils libgtk-3-doc libpango1.0-doc libthai-doc graphicsmagick
The following NEW packages will be installed:
  libatk-bridge2.0-dev libatk1.0-dev libatspi2.0-dev libblkid-dev libcairo2-dev libdatrie-dev libdbus-1-dev libepoxy-dev libfribidi-dev libgdk-pixbuf-2.0-dev libglib2.0-dev libglib2.0-dev-bin
  libgraphite2-dev libgtk-3-dev libharfbuzz-dev libharfbuzz-gobject0 libmount-dev libpango1.0-dev libpcre16-3 libpcre3-dev libpcre32-3 libpcrecpp0v5 libpixman-1-dev librealsense2 librealsense2-gl
  librealsense2-utils libselinux1-dev libsepol-dev libthai-dev libxcb-render0-dev libxcb-shm0-dev libxcomposite-dev libxcursor-dev libxdamage-dev libxfixes-dev libxi-dev libxinerama-dev
  libxkbcommon-dev libxtst-dev pango1.0-tools wayland-protocols
0 upgraded, 41 newly installed, 0 to remove and 17 not upgraded.
Need to get 20.4 MB of archives.
After this operation, 83.8 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libglib2.0-dev-bin amd64 2.72.4-0ubuntu2.2 [117 kB]
Get:2 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libblkid-dev amd64 2.37.2-4ubuntu3 [185 kB]
Get:3 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libsepol-dev amd64 3.3-1build1 [378 kB]
Get:4 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libselinux1-dev amd64 3.3-1build2 [158 kB]
Get:5 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libmount-dev amd64 2.37.2-4ubuntu3 [14.5 kB]
Get:6 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpcre16-3 amd64 2:8.39-13ubuntu0.22.04.1 [164 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpcre32-3 amd64 2:8.39-13ubuntu0.22.04.1 [155 kB]
Get:8 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpcrecpp0v5 amd64 2:8.39-13ubuntu0.22.04.1 [16.5 kB]
Get:9 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libpcre3-dev amd64 2:8.39-13ubuntu0.22.04.1 [579 kB]
Get:10 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libglib2.0-dev amd64 2.72.4-0ubuntu2.2 [1,739 kB]
Get:11 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libatk1.0-dev amd64 2.36.0-3build1 [96.6 kB]
Get:12 https://librealsense.intel.com/Debian/apt-repo jammy/main amd64 librealsense2 amd64 2.54.2-0~realsense.10773 [2,844 kB]
Get:13 http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 libdbus-1-dev amd64 1.12.20-2ubuntu4.1 [188 kB]
Get:14 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libxfixes-dev amd64 1:6.0.0-1 [12.2 kB]
Get:15 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libxi-dev amd64 2:1.8-1build1 [193 kB]
Get:16 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libxtst-dev amd64 2:1.2.3-1build4 [16.3 kB]
Get:17 http://us.archive.ubuntu.com/ubuntu jammy/main amd64 libatspi2.0-dev amd64 2.44.0-3 [71.5 kB]
[... cut ...]
Unpacking librealsense2-utils:amd64 (2.54.2-0~realsense.10773) ...
dpkg: error processing archive /tmp/apt-dpkg-install-Zrpmec/40-librealsense2-utils_2.54.2-0~realsense.10773_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/realsense-viewer', which is also in package framos-librealsense2 2.50.12
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:
 /tmp/apt-dpkg-install-Zrpmec/40-librealsense2-utils_2.54.2-0~realsense.10773_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

monajalal, 2023-10-10:
@MartyG-RealSense  problem resolved 

```
(base) mona@mona-ThinkStation-P7:~/Downloads/FRAMOS_D400e_Software_Package$ dpkg --info FRAMOS-librealsense2-2.50.12-Linux64_x64.deb
 new Debian package, version 2.0.
 size 163935536 bytes: control archive=116827 bytes.
     216 bytes,    10 lines      control              
  420277 bytes,  3958 lines      md5sums              
     382 bytes,     5 lines      postinst             #!/bin/bash
 Package: framos-librealsense2
 Version: 2.50.12
 Section: devel
 Priority: optional
 Architecture: amd64
 Recommends: freeglut3
 Installed-Size: 432464
 Maintainer: FRAMOS GmbH
 Description: librealsense2 built using CMake
 
(base) mona@mona-ThinkStation-P7:~/Downloads/FRAMOS_D400e_Software_Package$ sudo dpkg -r framos-librealsense2
(Reading database ... 329902 files and directories currently installed.)
Removing framos-librealsense2 (2.50.12) ...
(base) mona@mona-ThinkStation-P7:~/Downloads/FRAMOS_D400e_Software_Package$ ls
total 171M
-rw-r--r-- 1 mona mona 9.5K Jul 21 07:21 ReadMe.txt
-rw-r--r-- 1 mona mona  11K Jul 21 07:21 FRAMOS_LibRealSense_Changelog.txt
-rw-r--r-- 1 mona mona 157M Jul 21 07:21 FRAMOS-librealsense2-2.50.12-Linux64_x64.deb
-rw-r--r-- 1 mona mona 4.7K Jul 21 07:21 FRAMOS_D400e_Software_Package_Changelog.txt
-rw-r--r-- 1 mona mona 9.4K Jul 21 07:21 FRAMOS_CameraSuite_Changelog.txt
-rw-r--r-- 1 mona mona  14M Jul 21 07:21 FRAMOS_CameraSuite_4.10.1.0-Linux64_x64.deb
drwxr-xr-x 2 mona mona 4.0K Jul 21 07:21 .
drwxr-xr-x 5 mona mona 4.0K Oct 10 10:35 ..
(base) mona@mona-ThinkStation-P7:~/Downloads/FRAMOS_D400e_Software_Package$ dpkg --info FRAMOS_CameraSuite_4.10.1.0-Linux64_x64.deb
 new Debian package, version 2.0.
 size 14579554 bytes: control archive=23884 bytes.
     290 bytes,    15 lines   *  config               #!/bin/sh
     315 bytes,    11 lines      control              
  104569 bytes,   734 lines      md5sums              
    1980 bytes,    61 lines      postinst             #!/bin/sh
      53 bytes,     5 lines      postrm               #!/bin/sh
     393 bytes,    11 lines      prerm                #!/bin/sh
    3155 bytes,    77 lines      templates            
 Architecture: amd64
 Description: Image acquisition library for GigE Vision standard digital cameras.
  CameraSuite library
 Maintainer: FRAMOS <info@framos.com>
 Package: framoscamerasuite
 Pre-Depends: gcc, make, libelf-dev, dkms, shim-signed
 Priority: optional
 Section: devel
 Version: 4.10.1.0
 Installed-Size: 45039
 
(base) mona@mona-ThinkStation-P7:~/Downloads/FRAMOS_D400e_Software_Package$ sudo dpkg -r framoscamerasuite
(Reading database ... 325242 files and directories currently installed.)
Removing framoscamerasuite (4.10.1.0) ...
```

and

```
(base) mona@mona-ThinkStation-P7:~$ sudo apt-get update
Get:1 file:/var/cuda-repo-ubuntu2204-11-7-local  InRelease [1,575 B]
Ign:2 http://10.82.164.106/debs ./ InRelease
Get:1 file:/var/cuda-repo-ubuntu2204-11-7-local  InRelease [1,575 B]
Ign:3 http://10.82.164.106/debs ./ Release           
Ign:4 http://10.82.164.106/debs ./ Packages
Ign:5 http://10.82.164.106/debs ./ Translation-en_US
Ign:6 http://10.82.164.106/debs ./ Translation-en
Ign:4 http://10.82.164.106/debs ./ Packages
[... 4 lines cut ...]
Ign:6 http://10.82.164.106/debs ./ Translation-en                                                                                 
Hit:4 http://10.82.164.106/debs ./ Packages                                                                                                             
Ign:5 http://10.82.164.106/debs ./ Translation-en_US                                                                                                    
Ign:6 http://10.82.164.106/debs ./ Translation-en                                                                                                           
Ign:5 http://10.82.164.106/debs ./ Translation-en_US                                                                                                   
Ign:6 http://10.82.164.106/debs ./ Translation-en                                                                                                                                                         
Ign:5 http://10.82.164.106/debs ./ Translation-en_US                                                                                                                                                      
Ign:6 http://10.82.164.106/debs ./ Translation-en                                                                                                                                                         
Ign:5 http://10.82.164.106/debs ./ Translation-en_US                                                                                                                                                      
Ign:6 http://10.82.164.106/debs ./ Translation-en                                                                                                                                                         
Hit:7 https://nvidia.github.io/libnvidia-container/stable/ubuntu18.04/amd64  InRelease                                                                                                                    
Hit:8 https://nvidia.github.io/nvidia-container-runtime/stable/ubuntu18.04/amd64  InRelease                                                                                                               
[... cut ...]
Hit:14 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                              
Hit:15 https://packages.microsoft.com/repos/azure-cli jammy InRelease                                                                                   
Hit:16 https://packages.microsoft.com/repos/ms-teams stable InRelease                            
Hit:17 http://packages.ros.org/ros2/ubuntu jammy InRelease
Hit:18 https://packages.microsoft.com/repos/code stable InRelease
Hit:19 https://librealsense.intel.com/Debian/apt-repo jammy InRelease
Reading package lists... Done
```

MartyG-RealSense, 2023-10-11:
realsense-viewer is installed by installing the **librealsense2-utils** package with `sudo apt-get install librealsense2-utils` and so if you had install problems with that package earlier then realsense-viewer may not be installed on your computer.

monajalal, 2023-10-11:
@MartyG-RealSense thanks a lot. I followed your instruction and was able to install the package and load the `realsense-viewer`. After installing, I had some bit of hiccup when updating the firmware. The first time updating the firmware it crashed but trying it for the second time it worked. Here are some screenshots for your reference.

Is on chip calibration necessary? I cannot calibrate it when I point the camera to the wall and when I point it elsewhere, the health score is -1. What about focal point calibration? Is that necessary? I am closing the issue because the main task is done. 

[image]

[image]
[image]
[image]
[image]
[image]
[image]

[image]
[image]
[image]

closed 2023-10-11 by monajalal

MartyG-RealSense, 2023-10-11:
It's great to hear that you were successful with installing realsense-viewer!

On-Chip calibration can improve the quality of the depth image, whilst Tare calibration improves depth measurement accuracy.  Focal Length calibration is a rarely used option.

Your camera had firmware version 5.13.0.50 installed.  This firmware is designed for SDK versions 2.50.0 and 2.51.1.  As you are using 2.54.2, the correct firmware to use with it is **5.15.1.0** and it sounds as though your camera has now been updated to that firmware.

If on-chip calibration is not working for you then you can alternatively try resetting the camera to its default factory-new calibration in realsense-viewer using instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487