realsenseai/librealsense #12278: Capturing uint16 depth video using Python Script from Intel RealSense 435 Camera (Ubuntu 22.04)

monajalal, 2023-10-10:
How can I capture depth video in uint16 format (raw)?
Currently, the following format sounds wrong (and the mkv depth video feels weird)

```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

color_path = 'V00P00A00C00_rgb.avi'
depth_path = 'V00P00A00C00_depth.mkv'
colorwriter = cv2.VideoWriter(color_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640,480), 1)
depthwriter = cv2.VideoWriter(depth_path, cv2.VideoWriter_fourcc('F', 'F', 'V', '1'), 30, (640, 480), isColor=False)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        if not depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        colorwriter.write(color_image)
        depthwriter.write(depth_image)
        
        cv2.imshow('Stream', color_image)
        
        if cv2.waitKey(1) == ord("q"):
            break
finally:
    colorwriter.release()
    depthwriter.release()

pipeline.stop()
```

I need it for this: https://github.com/NVlabs/BundleSDF/issues/82#issuecomment-1698248648 and I need simultaneous and synchronous captures of depth and RGB using Intel RealSense 435.

For the code above, here's the mediainfo of captures:
```
(capture) mona@mona-ThinkStation-P7:~/intel-realsense-captures$ mediainfo V00P00A00C00_depth.mkv 
General
Unique ID                                : 2599431746086664507319781809109936678 (0x1F4A1E148D1C0E1E283001A49AD2A26)
Complete name                            : V00P00A00C00_depth.mkv
Format                                   : Matroska
Format version                           : Version 4
File size                                : 4.40 MiB
Duration                                 : 4 s 333 ms
Overall bit rate mode                    : Variable
Overall bit rate                         : 8 514 kb/s
Writing application                      : Lavf59.27.100
Writing library                          : Lavf59.27.100
ErrorDetectionType                       : Per level 1

Video
ID                                       : 1
Format                                   : FFV1
Format version                           : Version 0
Codec ID                                 : V_MS/VFW/FOURCC / FFV1
Duration                                 : 4 s 333 ms
Bit rate mode                            : Variable
Bit rate                                 : 8 345 kb/s
Width                                    : 640 pixels
Height                                   : 480 pixels
Display aspect ratio                     : 4:3
Frame rate mode                          : Constant
Frame rate                               : 30.000 FPS
Color space                              : Y
Bit depth                                : 8 bits
Compression mode                         : Lossless
Bits/(Pixel*Frame)                       : 0.905
Stream size                              : 4.31 MiB (98%)
Default                                  : No
Forced                                   : No
coder_type                               : Golomb Rice

```

and 

```
(capture) mona@mona-ThinkStation-P7:~/intel-realsense-captures$ mediainfo V00P00A00C00_rgb.avi 
General
Complete name                            : V00P00A00C00_rgb.avi
Format                                   : AVI
Format/Info                              : Audio Video Interleave
File size                                : 550 KiB
Duration                                 : 4 s 333 ms
Overall bit rate                         : 1 039 kb/s
Writing application                      : Lavf59.27.100

Video
ID                                       : 0
Format                                   : MPEG-4 Visual
Format profile                           : Simple@L1
Format settings, BVOP                    : No
Format settings, QPel                    : No
Format settings, GMC                     : No warppoints
Format settings, Matrix                  : Default (H.263)
Codec ID                                 : XVID
Codec ID/Hint                            : XviD
Duration                                 : 4 s 333 ms
Bit rate                                 : 1 022 kb/s
Width                                    : 640 pixels
Height                                   : 480 pixels
Display aspect ratio                     : 4:3
Frame rate                               : 30.000 FPS
Color space                              : YUV
Chroma subsampling                       : 4:2:0
Bit depth                                : 8 bits
Scan type                                : Progressive
Compression mode                         : Lossy
Bits/(Pixel*Frame)                       : 0.111
Stream size                              : 541 KiB (98%)
Writing library                          : Lavc59.37.100

```

Currently the RGB video is in correct format.

```
(capture) mona@mona-ThinkStation-P7:~$ uname -a
Linux mona-ThinkStation-P7 6.2.0-34-generic #34~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 13:12:03 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
(capture) mona@mona-ThinkStation-P7:~$ lsb_release -a
LSB Version:	core-11.1.0ubuntu4-noarch:security-11.1.0ubuntu4-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy
(capture) mona@mona-ThinkStation-P7:~$ python
Python 3.10.13 (main, Sep 11 2023, 13:44:35) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
```

* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

monajalal, 2023-10-10:
also using something like astype doesn't work as shown by mediainfo and I think it's not suggested. Is there a way to directly get the 16bit of depth saved in a video file?
```
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

color_path = 'V00P00A00C00_rgb.avi'
depth_path = 'V00P00A00C00_depth.mkv'

colorwriter = cv2.VideoWriter(color_path, cv2.VideoWriter_fourcc(*'XVID'), 30, (640,480), 1)
depthwriter = cv2.VideoWriter(depth_path, cv2.VideoWriter_fourcc('F', 'F', 'V', '1'), 30, (640, 480), isColor=False)

pipeline.start(config)

try:
    while True:
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        
        if not depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        
        depth_image_uint16 = depth_image.astype(np.uint16)  # Convert depth image to uint16 format
        
        depthwriter.write(depth_image_uint16)
        colorwriter.write(color_image)
        
        cv2.imshow('Stream', color_image)
        
        if cv2.waitKey(1) == ord("q"):
            break
finally:
    colorwriter.release()
    depthwriter.release()

pipeline.stop()
```

MartyG-RealSense, 2023-10-11:
Hi @monajalal  The link below may be a helpful reference.

https://forum.opencv.org/t/python-representation-of-mat-with-pointer-to-user-data/2239/3