realsenseai/librealsense #12279: Flickering images rgb and depth

Nelsonagj, 2023-10-10:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|f0246135|
|**Firmware Version**|1.5.8.1|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|2.1|
|**Product Line**|L500|
|**Asic Serial Number**|0003accd658b|
|**Firmware Update Id**|0003accd658b|

Please provide a description of the problem
The images keep appearing and disappearing. I have tried every usb ports in 2 different computers and the same problem occurs. I have updated the firmware and I am using the latest SDK. Thanks for your help!

MartyG-RealSense, 2023-10-11:
Hi @Nelsonagj  If the depth image is appearing and disappearing in the same manner as a video at https://github.com/IntelRealSense/librealsense/issues/9505#issuecomment-1012009632 then this is unfortunately an unfixable fault with the L515 camera hardware.  The recommendation for this fault is to contact the retailer that it was purchased from to enquire about returning the camera if it was bought during the past 12 months.

MartyG-RealSense, 2023-10-19:
Hi @Nelsonagj  Do you require further assistance with this case, please?  Thanks!

Nelsonagj, 2023-10-19:
Hi MartyG, apparently there is nothing else to be done, the camera is broken and because it was bought from Intel more than 3 years ago there is no free replacement.
Thank you for your help.
Regards,

MartyG-RealSense, 2023-10-19:
You are very welcome.  I'm sincerely sorry for the outcome of this case.

closed 2023-10-19 by MartyG-RealSense