realsenseai/librealsense #12280: libusb seg faults when accessing realsenses concurrently on Ubuntu 20

danielkoohmarey-bc, 2023-10-11:
* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   |5.13.0.XX | 
| Operating System & Version  |   Ubuntu 20  | 
| Kernel Version (Linux Only)   |  5.10.120                                    | 
| Platform                                 | NVIDIA Jetson |
| SDK Version                          | n/a            | 
| Language                               |  C++                           | 
| Segment			| Robot        | 

### Issue Description
We recently migrated from Ubuntu 18 to Ubuntu 20, and are seeing some seg faults using this branch of librealsense: https://github.com/IntelRealSense/librealsense/tree/4f37f2ef0874c1716bce223b20e46d00532ffb04

We are accessing 2 realsense cameras concurrently in C++ in two separate threads. Each thread has its own `rs2::context` object. When the threads access frames concurrently, we receive a seg fault:
```
(gdb) bt
#0  0x0000fffff5a9cad4 in  () at /lib/aarch64-linux-gnu/libusb-1.0.so.0
#1  0x0000fffff5a9cca4 in  () at /lib/aarch64-linux-gnu/libusb-1.0.so.0
#2  0x0000fffff5a9d274 in libusb_get_device_list () at /lib/aarch64-linux-gnu/libusb-1.0.so.0
#3  0x0000fffff6dc89bc in librealsense::platform::usb_context::usb_context() (this=0xffffb8003c40) at /<redacted>/librealsense/src/libusb/context-libusb.cpp:18
#4  0x0000fffff6dd08d4 in __gnu_cxx::new_allocator<librealsense::platform::usb_context>::construct<librealsense::platform::usb_context>(librealsense::platform::usb_context*) (this=<optimized out>, __p=0xffffb8003c40) at /usr/include/c++/9/new:174
#5  std::allocator_traits<std::allocator<librealsense::platform::usb_context> >::construct<librealsense::platform::usb_context>(std::allocator<librealsense::platform::usb_context>&, librealsense::platform::usb_context*) (__a=..., __p=0xffffb8003c40)
    at /usr/include/c++/9/bits/alloc_traits.h:483
#6  std::_Sp_counted_ptr_inplace<librealsense::platform::usb_context, std::allocator<librealsense::platform::usb_context>, (__gnu_cxx::_Lock_policy)2>::_Sp_counted_ptr_inplace<>(std::allocator<librealsense::platform::usb_context>) (__a=..., this=0xffffb8003c30)
    at /usr/include/c++/9/bits/shared_ptr_base.h:548
#7  std::__shared_count<(__gnu_cxx::_Lock_policy)2>::__shared_count<librealsense::platform::usb_context, std::allocator<librealsense::platform::usb_context>>(librealsense::platform::usb_context*&, std::_Sp_alloc_shared_tag<std::allocator<librealsense::platform::usb_context> >) (__a=..., __p=<synthetic pointer>: <optimized out>, this=<synthetic pointer>) at /usr/include/c++/9/bits/shared_ptr_base.h:679
#8  std::__shared_ptr<librealsense::platform::usb_context, (__gnu_cxx::_Lock_policy)2>::__shared_ptr<std::allocator<librealsense::platform::usb_context>>(std::_Sp_alloc_shared_tag<std::allocator<librealsense::platform::usb_context> >)
    (__tag=..., this=<synthetic pointer>) at /usr/include/c++/9/bits/shared_ptr_base.h:1344
#9  std::shared_ptr<librealsense::platform::usb_context>::shared_ptr<std::allocator<librealsense::platform::usb_context>>(std::_Sp_alloc_shared_tag<std::allocator<librealsense::platform::usb_context> >) (__tag=..., this=<synthetic pointer>)
    at /usr/include/c++/9/bits/shared_ptr.h:359
#10 std::allocate_shared<librealsense::platform::usb_context, std::allocator<librealsense::platform::usb_context>>(std::allocator<librealsense::platform::usb_context> const&) (__a=...) at /usr/include/c++/9/bits/shared_ptr.h:702
#11 std::make_shared<librealsense::platform::usb_context>() () at /usr/include/c++/9/bits/shared_ptr.h:718
#12 librealsense::platform::usb_enumerator::query_devices_info() () at /<redacted>/librealsense/src/libusb/enumerator-libusb.cpp:83
#13 0x0000fffff6dd4940 in librealsense::platform::v4l_backend::query_usb_devices() const (this=<optimized out>) at /<redacted>/librealsense/src/linux/backend-v4l2.cpp:1819
#14 0x0000fffff6de4b7c in librealsense::polling_device_watcher::polling(dispatcher::cancellable_timer) (this=0xaaaaab1cd050, cancellable_timer=...) at /<redacted>/librealsense/src/types.h:1556
#15 0x0000fffff6de952c in std::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const (__args#0=..., this=<optimized out>) at /usr/include/c++/9/bits/std_function.h:683
#16 active_object<std::function<void (dispatcher::cancellable_timer)> >::do_loop()::{lambda(dispatcher::cancellable_timer)#1}::operator()(dispatcher::cancellable_timer) const (this=<optimized out>, ct=...)
    at /<redacted>/librealsense/src/concurrency.h:404
#17 std::_Function_handler<void (dispatcher::cancellable_timer), active_object<std::function<void (dispatcher::cancellable_timer)> >::do_loop()::{lambda(dispatcher::cancellable_timer)#1}>::_M_invoke(std::_Any_data const&, dispatcher::cancellable_timer&&)
    (__functor=..., __args#0=...) at /usr/include/c++/9/bits/std_function.h:300
#18 0x0000fffff6c55698 in std::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const (__args#0=..., this=0xffffebd9e6c8) at /usr/include/c++/9/bits/std_function.h:683
#19 dispatcher::dispatcher(unsigned int)::{lambda()#1}::operator()() const (__closure=0xaaaaab1b2458) at /<redacted>/librealsense/src/concurrency.h:231
#20 0x0000fffff73faf9c in  () at /lib/aarch64-linux-gnu/libstdc++.so.6
#21 0x0000fffff751a624 in start_thread (arg=0xfffff73faf80) at pthread_create.c:477
#22 0x0000fffff726849c in thread_start () at ../sysdeps/unix/sysv/linux/aarch64/clone.S:78
```
Additional separate backtrace, with libusb logs:
```
[Realsenserealsense_camera_front][info][74756] processFrame returned for Depth
[47.435637] [000123f2] libusb: debug [linux_get_parent_info] Dev 0xffffb8028100 (1-2.7) has parent 0xffffb80123d0 (1-2) port 7
[47.436823] [000123f2] libusb: debug [linux_get_device_address] getting address for device: 1-3 detached: 0
[47.436882] [000123f2] libusb: debug [linux_get_device_address] scan 1-3
[47.437064] [000123f2] libusb: debug [linux_get_device_address] bus=1 dev=3
[47.437110] [000123f2] libusb: debug [linux_enumerate_device] busnum 1 devaddr 3 session_id 259
[47.437153] [000123f2] libusb: debug [linux_enumerate_device] allocating new device for 1/3 (session 259)
[47.437318] [000123f2] libusb: debug [linux_get_parent_info] Dev 0xffffb80281f0 (1-3) has parent 0xffffb8027f00 (usb1) port 3
[47.437748] [000123f2] libusb: debug [linux_get_device_address] getting address for device: usb2 detached: 0
[47.437798] [000123f2] libusb: debug [linux_get_device_address] scan usb2
[47.437969] [000123f2] libusb: debug [linux_get_device_address] bus=2 dev=1
[47.438032] [000123f2] libusb: debug [linux_enumerate_device] busnum 2 devaddr 1 session_id 513
[47.438075] [000123f2] libusb: debug [linux_enumerate_device] allocating new device for 2/1 (session 513)
[47.438965] [000123f2] libusb: debug [linux_get_device_address] getting address for device: 2-3 detached: 0
[47.439022] [000123f2] libusb: debug [linux_get_device_address] scan 2-3
[47.439233] [000123f2] libusb: debug [linux_get_device_address] bus=2 dev=2
[47.439260] [000123f2] libusb: debug [linux_enumerate_device] busnum 2 devaddr 2 session_id 514
[... 70 lines cut ...]
#6  0x0000fffff6dd4940 in librealsense::platform::v4l_backend::query_usb_devices() const (this=<optimized out>) at //<redacted>//librealsense/src/linux/backend-v4l2.cpp:1819
#7  0x0000fffff6de4b7c in librealsense::polling_device_watcher::polling(dispatcher::cancellable_timer) (this=0xaaaaab1cd050, cancellable_timer=...)
    at //<redacted>//librealsense/src/types.h:1556
#8  0x0000fffff6de952c in std::function<void (dispatcher::cancellable_timer)>::operator()(dispatcher::cancellable_timer) const (__args#0=..., this=<optimized out>) at /usr/include/c++/9/bits/std_function.h:683
#9  active_object<std::function<void (dispatcher::cancellable_timer)> >::do_loop()::{lambda(dispatcher::cancellable_timer)#1}::operator()(dispatcher::cancellable_timer) const (this=<optimized out>, ct=...)
    at //<redacted>//librealsense/src/concurrency.h:404
[... cut ...] re any known issues/limitation to instantiating multiple `rs2::context` within the same process and running them in a thread?

We fetch frames by registering a callback with the `rs2::sensor`:
```
            rs2::sensor sensor = activeSensors[moduleName];
            sensor.open(sensor_profile.second);
            sensor.start(_sensorsCallback[moduleName]);  // how we fetch frames via callback
```
Any insight into improper usage resulting in the seg fault or how to avoid is appreciated, thanks!

MartyG-RealSense, 2023-10-11:
Hi @danielkoohmarey-bc   Which JetPack version are you using, please?  JetPack 4 is for Ubuntu 18.04, whilst JetPack 5 is the first to be compatible with Ubuntu 20.04.  The librealsense SDK officially supports JetPack **5.0.2** from SDK version 2.54.1 onwards.

Which librealsense SDK version are you currently using?

In regard to multiple cameras, you can either use **ctx.query_devices** to automatically build a list of all attached cameras and their serial numbers (as rs-multicam does), or instead manually define each individual camera and its serial number.  If the same cameras will not always be attached to the computer then automatically building a device list with ctx will be the best approach.

Usually there will only be one **rs2::context**.  You can though create multiple pipelines and assign a different camera to each.  The rs-multicam example creates multiple pipelines automatically, one for each attached device.

MartyG-RealSense, 2023-10-11:
Yes, ideally you should upgrade to 2.54.1 or newer to use JetPack 5 and Ubuntu 20.04.  

If you build librealsense again from packages then it should automatically be the latest version (currently 2.54.2).

If your build librealsense from source code then you can select the version that you want to use from the Releases page, and download the source code as a zipped folder from the **Assets** list at the bottom of each version's information listing.

https://github.com/IntelRealSense/librealsense/releases/

closed 2023-10-25 by MartyG-RealSense