realsenseai/librealsense #12284: Missing headers for realsense-file

GrubbyHalo, 2023-10-11:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |N/A | 
| Firmware Version                   |N/A| 
| Operating System & Version |  Arch Linux | 
| Kernel Version (Linux Only)    |  (6.5.6)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  { 2.53.1  }                          | 
| Language                            |  {C/C++ }                          | 
| Segment			|                            | 

Is there any reason why the SDK installs the realsense-file static library (librealsense-file.a) but no associated header files.

MartyG-RealSense, 2023-10-11:
Hi @GrubbyHalo  I will seek advice from my Intel RealSense colleagues about your question.  Thanks very much for your patience!

maloel, 2023-10-12:
Hi @GrubbyHalo 
`realsense-file` is a statically linked library that is not meant for public consumption. As such, it should **not** be installed, and should be used only within the librealsense build process.
The same is true for other libraries included in our build.
I will take a look and, as part of our next major release, will try to remove such artifacts. No promises, though :).

GrubbyHalo, 2023-10-12:
Hi @maloel 
For a recent project, I needed some of the extra rosbag file functionality (e.g. depth scale, intrinsics, fps)  exposed by the `realsense-file` code. I basically copied the directory in the SDK and plugged it into my own project. This seemed like a better option than trying to add  an  independent library from the ROS project or elsewhere just to obtain some rosbag functionality. I the interest of streamlining workflows it would be great if I could have the inclusion of the `realsense-file` code as a cmake option for instance.

maloel, 2023-10-12:
You can continue to copy the directory from our repo -- that won't change. But I'm not sure I understand what it is you're asking for: you want realsense-file as its own separate repo so that you can download and use it from your own cmake, without copying?

GrubbyHalo, 2023-10-13:
> you want realsense-file as its own separate repo so that you can download and use it from your own cmake, without copying?
No. I was saying that instead of removing `realsense-file` from the installation, that it would be great to keep it in and provide the headers as well because some of the extra functionality with regards to bag files can be useful. Maybe it could be optionally built and installed by specifying a cmake build flag when building the real-sense libs. e.g. 
`cmake .... -DBUILD_REALSENSE_FILE ..`

MartyG-RealSense, 2023-10-19:
Hi @maloel   Could you further assist @GrubbyHalo regarding their comment above please?  Thanks!

maloel, 2023-10-19:
@GrubbyHalo 

Currently, as long as we want to support record & playback, `realsense-file` is always a dependency, so is always built.

Releasing it in the Debians is there, but more libraries are coming and, rather than include them all in the Debians, we're more likely to remove the ones that aren't needed for the distribution.

But I understand what you're asking for. I will put it on the todo to discuss. So again, no promises :)

GrubbyHalo, 2023-10-19:
@maloel 

Ok.