realsenseai/librealsense #12285: Announcement: Introducing the RealSense D456 IP65-rated stereo depth camera

MartyG-RealSense, 2023-10-11:
Hi everyone,

A new model in the RealSense 400 Series stereo depth camera family, the **D456**, is now available for purchase in the official online RealSense Store.  The D456 has the same capabilities as the D455 model, with the addition of an IP65-rated casing that protects against dust and water.

https://www.intelrealsense.com/depth-camera-d456/

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d456.html

A new edition of the data sheet document for the 400 Series cameras that includes the D456 is also now available.

https://dev.intelrealsense.com/docs/intel-realsense-d400-series-product-family-datasheet

marameref, 2023-10-13:
To update the datasheet document requires you to login to the portal and have write access. Is this issue still open, i will like to work on it pls?

MartyG-RealSense, 2023-10-13:
Hi @marameref  The datasheet is a read-only PDF document that is not meant to be updated except by the Intel staff who work on it.

closed 2023-10-15 by MartyG-RealSense