realsenseai/librealsense #12286: Cannot Access https://egiintel.jfrog.io/artifactory/librealsense

Shaan-Mistry, 2023-10-12:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | N/A | 
| Operating System & Version |   MacOS  | 
| Platform                                 | Android Studio  |
| SDK Version                            |  { 2.54.2 }                          | 
| Language                            |  {Java/C++}                          | 
| Segment			|  {Smartphone }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to run both the Java and Native examples.  It seems that I need to download the lib from https://egiintel.jfrog.io/ui/repos/tree/General/librealsense. But this link requires a login. Do I need to create some sort of account for the the JFrog? When I follow the link it says "JFrog's Free-Tier subscription has been deactivated."

Here is some of the build errors:
```
Caused by: org.gradle.api.resources.ResourceException: Unable to load Maven meta-data from https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/maven-metadata.xml. 
```
or
```
Unable to load Maven meta-data from https://egiintel.jfrog.io/artifactory/librealsense/com/intel/realsense/librealsense/maven-metadata.xml. 
> org.xml.sax.SAXParseException; lineNumber: 1; columnNumber: 326; The reference to entity "display" must end with the ';' delimiter. 
```

MartyG-RealSense, 2023-10-12:
Hi @Shaan-Mistry  I will highlight this issue to my Intel RealSense colleagues.  Thanks very much for your patience!

Shaan-Mistry, 2023-10-16:
Hi is there any update on the possible cause of the issue?

MartyG-RealSense, 2023-10-17:
I have followed up again with my Intel RealSense colleagues and will let you know as soon as I have information.  Thanks very much for your patience!

MartyG-RealSense, 2023-10-17:
> I have followed up again with my Intel RealSense colleagues and will let you know as soon as I have information. Thanks again!

xMKx, 2023-10-23:
This is still an issue as far as I see now. How would we be able to download the library?

MartyG-RealSense, 2023-10-23:
Hi @xMKx  I have informed my colleagues on the RealSense team about this issue and will be sure to follow up about it this week.

Shaan-Mistry, 2023-10-26:
> Hi @xMKx I have informed my colleagues on the RealSense team about this issue and will be sure to follow up about it this week.

Are there any workarounds that we can look into for now while the team fixes the issue?

MartyG-RealSense, 2023-10-27:
Hi @Shaan-Mistry  I believe that the only alternative to the JFrog repository would be to build the Android wrapper from source code in gradle or in Android Studio using the procedure described at the link below.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android#instructions

MartyG-RealSense, 2023-10-30:
Hi again @Shaan-Mistry  I received feedback from my Intel RealSense colleagues regarding JFRog.  We are no longer releasing to JFrog or performing any testing on Android.  The reference to JFrog will therefore be removed in a future update.

Shaan-Mistry, 2023-11-02:
Thank you for the update. So if I wanted to build something similar to the C++ Native example, which example from here: https://github.com/IntelRealSense/librealsense/tree/master/wrappers/android#instructions would I follow or make changes to?

MartyG-RealSense, 2023-11-03:
I am unfamiliar with native NDK programming in the Android wrapper.  The best reference available for doing so is likely to be https://github.com/IntelRealSense/librealsense/issues/8277

When the above link talks about downloading from **bintray**, it means the repository that Intel used for Android downloads before it moved over to JFrog.

https://github.com/IntelRealSense/librealsense/issues/4091 also references programming of a native application.

Sincere apologies that I could not be of more help on this particular question.

Shaan-Mistry, 2023-11-06:
Great thanks for the help!

closed 2023-11-19 by MartyG-RealSense