realsenseai/librealsense #12292: How to solve RuntimeError by taking irregular photos: Frame didn't drive within 5000

12581djjd, 2023-10-13:
The project requires a d435 camera to take occasional photos, and I couldn't find a similar project on the official website. I am borrowing from video related projects, but it seems that I cannot avoid timeout issues

MartyG-RealSense, 2023-10-13:
Hi @12581djjd  One method of taking occasional photos would be to perform a **sleep** operation between captures so that the program does not do anything until the sleep period has elapsed (a number of seconds equal to the minutes that you want to wait before the next capture).

If you are having difficulties with timeouts then another method that has been used with other RealSense projects that take occasional captures is to have the camera wait for a **trigger pulse** signal to be received before initiating capture.  

An example of this is a project at https://github.com/IntelRealSense/realsense-ros/issues/1354 involving plants on a conveyor belt that took a capture every 15 minutes.

Another conveyor-based approach described at https://github.com/IntelRealSense/librealsense/issues/2179 is to have capture initiated when an object passes through a photoelectric switch.

12581djjd, 2023-10-14:
Thank you very much for your help，

closed 2023-10-25 by MartyG-RealSense