realsenseai/librealsense #12293: d455 - libusb crash in python

bricemiramont, 2023-10-13:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.13.00.55 | 
| Operating System & Version |  Linux for Jetson / Jetpack 5.1  | 
| Kernel Version (Linux Only)    | 5.10.104-tegra #6 SMP PREEMPT                                         | 
| Platform                                 | NVIDIA Jetson Xavier NX  |
| SDK Version                            |  2.51.1                         | 
| Language                            | python 3.8.10                           | 
| Segment			|  Computer vision  | 

### Issue Description
Hello, I have a multiprocess app written in Python for realtime image computing. Main process is acquiring frames (depth, rgb and motion) from the d455 then with shared memory variables makes the pointcloud and rgb image available to other sub-processes to process the data.
Several times per 24h, the main process crash with a log from the libusb. There are 2 different error messages at least, as indicated below:

```
libusb: debug [libusb_submit_transfer] transfer 0xffff08001818
./scripts/launch-detection.sh: line 26: 21214 Segmentation fault      (core dumped) python3 main-mp.py "$@"

libusb: debug [handle_bulk_completion] handling completion status 0 of bulk urb 1/1
./scripts/launch-detection.sh: line 26: 21214 Segmentation fault      (core dumped) python3 main-mp.py "$@"
```

libusb version is : v1.0.23.11397

Even though the program is using multi process libs, the realsense pipeline is only used in the main thread, then the frames are processed and the resulting arrays used in a shared memory variable as the rs2 frame cannot be directly used (to my knowledge) in a mp shared memory variable.

librealsense has been compiled with RSUSB API for communication (instead of Linux native kernel drivers), my main objective is to eliminate the crash. Shall I use native kernel drivers to avoid this problem or is there a correction available ?

This is a production environment so any indication for performance/stability improvement is welcome.

Thank you,

bricemiramont, 2023-10-13:
Tried to patch the kernel but there is an issue:
```
Jetson Board (proc/device-tree/model): NVIDIA Jetson Xavier NX Developer Kit
Jetson L4T version: 35.2.1
Unsupported JetPack revision 35.2.1 aborting script
```

Can I force it or am I stuck with Jetpack 5.0.2 ?

thank you

MartyG-RealSense, 2023-10-13:
Hi @bricemiramont  If you have a production environment then using 5.0.2 with SDK **2.54.1** instead of 2.51.1 may be the best option, as 2.54.1 officially supports JetPack 5.0.2 in its **patch-realsense-ubuntu-L4T.sh** patch script.

https://github.com/IntelRealSense/librealsense/blob/master/scripts/patch-realsense-ubuntu-L4T.sh

If you need to use 2.51.1 and a JetPack newer than 5.0.2 then the best option is likely to be either to build the SDK from source code with CMake and include the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction, or to build from source code but not apply a patch script to the kernel.

bricemiramont, 2023-10-13:
Ok thank you, but I think the RSUSB backend is at fault in my setup and I would like to avoid it.

Any timeline for Jetpack 5.1+ support for the librealsense ?

The only requirements is to use Jetpack 5.1 and native kernel drivers

thank you

MartyG-RealSense, 2023-10-13:
There is a discussion at https://github.com/IntelRealSense/librealsense/issues/12012 about JetPack 5.1.

Whilst I do not have information about a timeline for its official support at the time of writing this, a RealSense user in that discussion managed to edit the patch script to add 5.1 support themselves.  This is a method that has been used previously by users, such as the one who edited the patch script for PC computers to add kernel 5.11 support.

If you do not wish to edit the patch script or use RSUSB then I recommend not applying a patch script and seeing how it works.

MartyG-RealSense, 2023-10-19:
Hi @bricemiramont  Do you require further assistance with this case, please?  Thanks!

bricemiramont, 2023-10-19:
Hi, no it's ok I will close the issue. For the record here is what I did:
- uninstall previsous version of the sdk that was compiled from source with RSUSB. (sudo make uninstall && sudo make clean)
- edited the L4T patch script to add JetPack 5.1 support
- patched the kernel (I only have the uvcvideo patched, no sensor_hid etc..) -> worked fine
- compiled the 2.54.2 version of the sdk (with python support because I need python support) -> worked fine
- copied the pyrealsense2 and pybackend2 libs in the correct place and did the symbolic links etc.. to have it work with python.

Brice

closed 2023-10-19 by bricemiramont

MartyG-RealSense, 2023-10-19:
Thanks very much for the update and for sharing the details of your procedure!