realsenseai/librealsense #12294: Questions and confusion regarding the API

milan-r-shah, 2023-10-13:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Questions:
- I have been trying to understand the API as in-depth as possible. I have been mainly going through these two pages:
  1. https://intelrealsense.github.io/librealsense/doxygen/namespacers2.html
  2. https://github.com/IntelRealSense/librealsense/tree/master/doc
- However, I still cannot fully grasp the following concepts:
  1. [`rs2::device`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device.html)
  2. [`rs2::sensor`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1sensor.html)
  3. [`rs2::context`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1context.html)
  4. [`rs2::device_hub`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__hub.html)
  5. [`rs2::device_list`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__list.html)
  6. [`rs2::config`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1config.html)
  7. [`rs2::pipeline`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html)

<br>

- Here is the scenario: Let's say I have two RealSense cameras: 
  - Camera A - D435i 
  - Camera B - D455
- Both are connected to the same host computer. So, the following are my understanding and/or some questions. Please correct me if I'm mistaken.

1. [`rs2::device`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device.html):
   - The purpose of `rs2::device` is to control both of them independently, correct?

2. [`rs2::sensor`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1sensor.html):
   - The purpose of `rs2::sensor` is to control individual sensors (Depth camera, RGB camera, and IMU) in each camera, right?
   - Are the Left Grayscale Camera, Right Grayscale Camera, and IR emitter considered separate sensors, or they are just considered part of/collectively called the Depth Camera sensor?
   - Let's say I'm using C++ and somewhere I found this code snippet:
     ```
     rs2::device dvc;
     .
     .
     std::vector<rs2::sensor> sensors = dvc.query_sensors();
     for (const auto sen : sensors) {
     
     }
     ```
     But how can I get the details (or at least `cout` the names) of all the detected sensors?
     I know `rs2::sensor` has the [`get_info`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1sensor.html#ab8badd4183c42e86cf2f8ebd19fe460f) method. However, it needs [`rs2_camera_info`](https://intelrealsense.github.io/librealsense/doxygen/rs__sensor_8h.html#aa607d93c1bac40af3ec1c9b0345f59af) as an input argument. I couldn't find a way to get information!
   - In the above example, I even tried to print the size of `sensors` i.e. `sensors.size()`. Every time it says `3` but I don't know which three sensors. For now, I'm mainly interested in the IMU. But I don't know how can I get the IMU data through `rs2::device`. I did find a sample ([imutest](https://github.com/GruffyPuffy/imutest)), through the [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples) page. However, that one uses `rs2::frameset` i.e all the data (left grayscale image, right grayscale image, depth image, RGB image, accelerometer data, and gyroscope data) is synchronized, correct? I can't individually control the frequency of the Accelerometer and Gyroscope if I use `rs2::frameset`, right? Please correct me if I'm mistaken here.

3. [`rs2::context`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1context.html):
   - What exactly is this? I can't wrap my head around it. Because the documentation just says "default librealsense context class includes realsense API version as provided by RS2_API_VERSION macro"
   - In the aforementioned case, do I have to initialize two `rs2::context`s? One for D435i and one for D455?

4. [`rs2::device_hub`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__hub.html):
   - What is this? The documentation just says "[device_hub](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__hub.html) class - encapsulate the handling of connect and disconnect events"

5. [`rs2::device_list`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__list.html):
   - For this one, I assume it's a list of all the connected and detected devices, correct? So, in the aforementioned case, this one will have D435i and D455, right?
   - But then how does this differ from `rs2::device_hub`? When to use which one?

6. [`rs2::config`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1config.html):
   - The documentation says "The config allows pipeline users to request filters for the pipeline streams and device selection and configuration. This is an optional step in pipeline creation, as the pipeline resolves its streaming device internally. Config provides its users a way to set the filters and test if there is no conflict with the pipeline requirements from the device. It also allows the user to find a matching device for the config filters and the pipeline, in order to select a device explicitly, and modify its controls before streaming starts." Does this mean I need to initialize two `rs2::config`s for the aforementioned scenario of two cameras? 
   - However, in that case, what's the role of `rs2::device`?

7. [`rs2::pipeline`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html):
   - The documentation says "The pipeline simplifies the user interaction with the device and computer vision processing modules. The class abstracts the camera configuration and streaming, and the vision modules triggering and threading. It lets the application focus on the computer vision output of the modules, or the device output data. The pipeline can manage computer vision modules, which are implemented as a processing blocks. The pipeline is the consumer of the processing block interface, while the application consumes the computer vision interface." Does this mean I need to initialize two `rs2::pipelines`s for the aforementioned scenario of two cameras? 
   - However, in that case, what's the role of `rs2::device`? Because the default param in one of the [constructor](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1pipeline.html#aae168079fd9b73927185efb7a0556dad)s accepts empty `rs2::context`:
     ```
     rs2::pipeline::pipeline(context ctx = context())
     ```
     As I don't know much about `rs2::context`, I'm confused about how if would work out in the aforementioned scenario of two cameras.

<br/>
If the number of questions is too many for one issue, please let me know. In that case, I'll create multiple separate issues.
<br/>
Thanks in advance!

MartyG-RealSense, 2023-10-14:
1.  The **api_how_to.h** file of the RealSense SDK's 'sensor-control' example project states:

An **rs2::device** is a container of rs2::sensors that have some correlation between them.  For example: 

- A device where all sensors are on a single board
- A robot with mounted sensors that share calibration information

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L118-L123

2.  Yes, **rs2::sensor** enables you to access information about individual sensors and control their settings, like in the previously mentioned **sensor_control** example program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/rs-sensor-control.cpp

The left and right IR sensors are individual sensors.  The IR emitter is a separate component from the left and right IR sensors but is not a sensor.  It is  another name for the projector component that casts a semi-random pattern of dots onto the scene, which is overlaid on the IR image.  The camera uses these dots on the surface of objects as a texture source to aid in the analysis of that surface's depth.

************

The Depth stream and the left and right IR streams are usually placed together under the category name 'Stereo Module' or 'Depth Module'.  For example, in the RealSense ROS2 wrapper, 'depth_module' refers to the depth stream and Infrared 1 and Infrared 2 streams, whilst 'rgb_camera' refers to the RGB camera stream.

*********

The C++ example program **FluentRealSense** contributed by a RealSense user at the link below describes how to retrieve information from the different types of **RS2_CAMERA_INFO** instructions.

https://www.codeproject.com/Articles/1233892/FluentRealSense-The-First-Steps-to-a-Simpler-RealS

******

On a camera equipped with an IMU such as D455, the three sensors are (0) Stereo Module, (1) RGB Camera and (2) Motion Module (the IMU)

******

Streaming IMU data only is straightforward but streaming IMU and additional streams such as depth and RGB is more complicated and usually requires use of a **callback** to use IMU data alongside depth and RGB.  The C++ script at https://github.com/IntelRealSense/librealsense/issues/6424 may be useful.

**********

3.  **rs2::context** is typically used to manage multiple RealSense cameras instead of just a single camera.  It is often used in the instruction **rs2::context ctx**.  An example of its use is in the **rs-multicam** multiple camera example program.

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L15

Once the context has been defined as ctx then a single rs2::context can be used to automatically compile a list of all attached cameras (no matter what their model name is) using the instruction **ctx.query_devices()**

https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L23

*********

4.  rs2::device_hub handles USB connection and disconnection events.  It can listen to the USB status and identify when the camera connection is established and when it becomes disconnected.  If a list of cameras has been generated with query_devices then cameras can be automatically added to and removed from the list as they connect and disconnect.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1device__hub.html

*******

5.  rs2::device_list provides a list of devices.  The link below provides information about ways to use it.

https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L47-L61

As mentioned earlier, rs2::device_hub is for the handling of USB connect and disconnect events and so is not the same thing as rs2::device_list.

*****

6.  You only need to create separate configurations for each camera if you are defining different settings for each individual camera instead of using the same configuration for all cameras.  An example of this would be setting one camera's depth to 30 FPS and the other camera's depth to 15 FPS.

The purpose of rs2::device is described in the answer to question 1 above.

******

7.  An application that uses multiple camera 'ctx' code such as the rs-multicam example will automatically create a separate pipeline for each camera, as it can also automatically detect the serial number of each individual camera.

It is possible though to manually define separate pipelines if you provide the serial number of the camera that you are assigning to a particular pipeline instead of having the camera detect the serial number automatically.  This approach is less flexible than automatic pipeline creation though, as you would have to always use the same set of camera units because of the need to manually define the cameras' serial numbers.

milan-r-shah, 2023-10-17:
Thank you so much @MartyG-RealSense for such a detailed and quick response. I appreciate it :pray: 

Further questions/confusion/clarifications:

1. In your response, you mentioned
   > The left and right IR sensors are individual sensors.
   
   But you also mentioned
   > On a camera equipped with an IMU such as D455, the three sensors are (0) Stereo Module, (1) RGB Camera, and (2) Motion Module (the IMU)

   That part is still a bit confusing. I think you meant "The left and right IR sensors are individual **physical** sensors but if we talk about `rs2::sensor` objects then there are three in D455 (namely, (`0`) Stereo Module, (`1`) RGB Camera and (`2`) Motion Module)" correct?

2. Let's say hypothetically we have the D430 module, not the camera. It does not have an RGB camera. In that case, does the "Motion Module" still get assigned index `2` or it will have index `1`?

3. What exactly is the use of [`rs2::options`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1options.html)? I can see that `rs2::sensor` inherits `rs2::options` but couldn't find more documentation or idea behind it. `rs2::options` has nothing to do with the sensor settings, correct?

4. For IMU, as I'm dealing with other streams as well, the example link (#6424) you mentioned is useful. However, I still don't quite understand the concept of [`rs2::stream_profile`](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1stream__profile.html):
   - What exactly is `rs2::stream_profile`? I mean it's just a way of specifying settings for different streams of data in a given sensor? or it's some sort of data bus?
   - If it is some sort of data bus, does that mean I can have the same type of data e.g. Accelerometer data, at different frequencies on different buses at the same time (or say can I have the different resolutions of RGB images on different buses at the same time)?
   - For the "Motion Module" (of type `rs2::sensor`) when I run `get_stream_profiles()`, it gives me 4 different stream profiles i.e. `std::vector<rs2::stream_profile>` consisting of 4 `rs2::stream_profile` objects. Through some investigation, I found out that two of them are for the Accelerometer and two of them are for the Gyroscope:
     | type | format | fps |
     |-----|----|----|
     | Accel | MOTION_XYZ32F | 250Hz |
     | Accel | MOTION_XYZ32F | 63Hz |
     | Gyro  | MOTION_XYZ32F | 400Hz |
     | Gyro  | MOTION_XYZ32F | 200Hz |
     
     But somewhere on the web, I had also read about the `RS2_FORMAT_MOTION_RAW` format. I couldn't find it in my case. Do you know why? 

5. The #6424 helped me a bit but I still don't fully understand how can I set the frequency of different streams. For example, somewhere I saw the procedure:
   1. For a given `rs2::sensor`, get all the stream profiles i.e. `std::vector<rs2::stream_profile>`
   2. Compare each of them with the desired requirements and save the selected ones into a new vector e.g. `std::vector<rs2::stream_profile> new_strm_prof_vec`
   3. For example, I want Accelerometer data at 250 Hz and Gyroscope data at 400 Hz. So, from the original vector of stream profiles, I would select the first and third object and put them into the aforementioned new vector.
   4. Use that new vector while opening a sensor i.e. `motion_module.open(new_strm_prof_vec)`
   5. Is this the correct procedure? 
   6. Because I noticed that even if I pass the conflicting stream profiles, for example, let's say the vector I pass to the `sensor.open()` function looks like this:
     `{ {Accel, MOTION_XYZ32F, 250Hz}, {Accel, MOTION_XYZ32F, 63Hz}, {Accel, MOTION_XYZ32F, 400Hz} }`
      I don't get any errors or warnings. I just get the accelerometer data at 250 Hz.
   7. In the vector, if I specify 63 Hz first and then 250 Hz i.e. 
       `{ {Accel, MOTION_XYZ32F, 63Hz}, {Accel, MOTION_XYZ32F, 250Hz}, {Accel, MOTION_XYZ32F, 400Hz} }`
       I get the Accelerometer data at 63 Hz!

6. Does `sensor.start(callback)` creates a new thread? I tried looking into the source code but couldn't find it.

7. How can I differentiate between accelerometer data and gyroscope data because both of them are of the same type `rs2::frame` or `rs2::motion_frame`? For example, in #6424, 
   ```
   auto callback = [&](const rs2::frame& frame) {
   }
   imuSsr.start(callback);
   ```
   Here both accelerometer and gyroscope data are of the same type `rs2::frame`. So, what can I do? Should I just try to get a corresponding `stream_profile` and that's it?

Thanks again!

MartyG-RealSense, 2023-10-18:
1.  There are four physical sensors on D435i and D455 - left IR, right IR, RGB and the motion module (IMU).  These sensors are categorized under the index numbers (0) Stereo Module - the two IR sensors, (1) RGB Camera and (2) Motion Module.

2.  The IMU is located on the Vision Processor D4 board and not the camera board.  So if you have a D430 then it will be identified by the firmware driver as D430 if the V1 version of the D4 board is used, and as D430i if the V2 version of the D4 board is used (as the D4 V2 has an IMU integrated on it).

I do not know how the index numbers would be assigned on a D430i without an RGB sensor, because having the camera identify as D430i is usually a situation that should be avoided if possible as D430i is not a 'proper' configuration.  It mostly exists to enable the camera's depth, IR and IMU streams to still work if the RGB sensor cannot be detected by the firmware for some reason. 

The librealsense SDK supports D430i but the RealSense ROS wrapper does not.  So if ROS were going to be used and the IMU was required then using D450 and the IMU-equipped D4 V3 would likely be best, so that the firmware should identify this combination as D455 (which is supported by the ROS wrapper).

3.  The stream_profile class stores the profile of a stream and enables that stream to be queried for information about it such as its format, FPS and intrinsics.

https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1stream__profile.html

rs2::options enables the setting of sensor options, yes.  This includes the hardware sensors and also 'software sensors' such as the software-device interface that enables a camera to be simulated virtually.

4.  rs2::stream_profile is covered by answer 3. above.

RS2_FORMAT_MOTION_RAW is raw data from the motion sensor.  Not all formats supported by the firmware are accessible to the user through the SDK.

5.  This question about vectors is outside of my programming knowledge, unfortunately.  I do apologize.  Usually, only one FPS speed at a time can be set for a stream, and if you want to change the speed then you have to stop the stream, change the configuration and restart the stream.

6.  The SDK's callback example program rs-callback states in its code that "the callback is executed on a sensor thread and can be called simultaneously from multiple sensors".

https://github.com/IntelRealSense/librealsense/blob/master/examples/callback/rs-callback.cpp#L21

7.  If the accel and gyro streams are given fixed index numbers such as 1 and 2 then this can cause gyro values to be reported for the accel stream and accel values to be reported for the gyro stream.  This is because the two IMU streams should be configured 'dynamically' - see https://github.com/IntelRealSense/librealsense/issues/6896#issuecomment-665274907

MartyG-RealSense, 2023-10-23:
Hi @milan-r-shah  Do you require further assistance with this case, please?  Thanks!

milan-r-shah, 2023-10-23:
@MartyG-RealSense Thanks a lot again for your comments. Sorry, I didn't get a chance to respond to your comments sooner. Yes, I have a few further questions:

1. You said:
   > ... because having the camera identify as D430i is usually a situation that should be avoided if possible as D430i is not a 'proper' configuration

   What do you mean by "_proper_ configuration"?

2. > RS2_FORMAT_MOTION_RAW is raw data from the motion sensor. Not all formats supported by the firmware are accessible to the user through the SDK.

   Out of curiosity, if that format is not accessible through the SDK, how exactly can you access it? Or it's not possible at all?

3. > "The callback is executed on a sensor thread and can be called simultaneously from multiple sensors"

   Oh, so it runs on its own thread on the realsense CPU rather than the CPU of the host device, correct?

4. > https://github.com/IntelRealSense/librealsense/issues/6896#issuecomment-665274907

   This example uses `pipeline` and `frameset`. However, in my case, as I'm trying to use Depth, RGB, and IMU data, I have to use callbacks. So, I don't think I can assign an index to Accelerometer and Gyroscope, correct?

MartyG-RealSense, 2023-10-24:
1.  D430i is not advertised as a RealSense product configuration like the depth modules in the image below, because it usually occurs unintentionally because of a problem rather than being a configuration that is deliberately sought.

[image]

There was once a customer who used the D430i configuration deliberately on their robot because they needed the IMU but not an RGB sensor.  So it is not wrong to deliberately use that configuration but it is highly rare to do so, and D430i will not be able to be used with the RealSense ROS wrapper.

2.  Some instructions will be usable by the camera firmware driver to perform a particular camera operaton but are not exposed in the RealSense SDK's 'High-Level API' that most scripts operate in and so are not accessible by scripts.  There may also be functions that are only accessible by Intel engineers with cameras that have special debug permissions, such as the ability to program the GPIO hardware sync pins.

The **Low-Level API** enables the camera sensor hardware to be accessed directly, though there is little documentation for doing so.

https://dev.intelrealsense.com/docs/api-architecture

The Low-Level API can be accessed through **callbacks**, as described at https://github.com/IntelRealSense/realsense-ros/issues/1409#issuecomment-696017011

3.  Threads will be processed on the host device's CPU and not the camera hardware.  The High / Low Level API page linked to above discusses threads.

4.  My knowledge of callback programming is admittedly limited, so I do not have an answer for that question.  Usually though, it is recommended to assign an index number to accel and gyro **dynamically** instead of giving them a fixed 1 and 2 number, otherwise the gyro stream may show the accel stream values and vice versa.

milan-r-shah, 2023-10-27:
@MartyG-RealSense I think I got the answers to most of my questions. For a few specific ones (e.g. callbacks v/s assigning an index number to accel and gyro, etc.), I'll create new issue(s).
Thank you so much again :pray:

closed 2023-10-27 by milan-r-shah

MartyG-RealSense, 2023-10-28:
You are very welcome  :)