realsenseai/librealsense #12300: get_xu(ctrl=1) when setting visual_preset

ZhaoXianglin, 2023-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

in enable_camera
    depth_sensor.set_option(rs.option.visual_preset, Preset.HighAccuracy)
RuntimeError: get_xu(ctrl=1) failed! Last Error: No error

ZhaoXianglin, 2023-10-17:
Hi Marty,
Thank you for your reply. I've used the D415 camera and successfully updated it to the latest firmware. However, despite trying the script you provided, my problem still persists.

I added `time.sleep(0.5)` after the code `pipeline.start(config)` and the problem has been solved. Is there any other solution to this problem?

MartyG-RealSense, 2023-10-17:
Did you install the librealsense SDK by building it from source code with CMake and including the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction, please?

If you did then an RSUSB build of librealsense can take longer to respond to a request to change a setting's status with a **set_option** instruction than it takes with a package build or with a non-RSUSB souce code build.

An RSUSB build can also sometimes miss an option change request made by a set_option instruction.

It is possible to edit the SDK's source code to respond faster when the SDK has been built in RSUSB mode, as described in the link below.

https://support.intelrealsense.com/hc/en-us/community/posts/9267755586323/comments/9268051567379

Otherwise, your sleep solution may be the best approach to solving the problem.

MartyG-RealSense, 2023-10-29:
Case closed due to no further comments received.

closed 2023-10-29 by MartyG-RealSense

ZhaoXianglin, 2023-11-15:
I followed your suggestion to build the source code, and the problem has been solved. Many thanks for your help MatryG.

MartyG-RealSense, 2023-11-15:
That's excellent news that your problem was solved.  Thanks very much for the update!

ruixucs, 2025-11-06:
Hi, I also met the problem, and the link you mentioned is not valid anymore, can you help me find the solution?

ruixucs, 2025-11-06:
Thank you very much, I will try this later today, have a good day