realsenseai/librealsense #12302: Dockerfile cannot be built with Ubuntu 23.10

cngzhnp, 2023-10-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
When Ubuntu 23.10 base image is used to compile 2.54.2 version, there is a compilation error which could be seen below:

#0 6.140 /usr/src/librealsense/third-party/rsutils/include/rsutils/version.h:23:13: error: 'uint64_t' does not name a type
#0 6.140    23 |     typedef uint64_t number_type;
#0 6.140       |             ^~~~~~~~
#0 6.140 /usr/src/librealsense/third-party/rsutils/include/rsutils/version.h:7:1: note: 'uint64_t' is defined in header '<cstdint>'; did you forget to '#include <cstdint>'?

In this header file, actually uint64_t requires <cstdint> system header file and it is missing.

MartyG-RealSense, 2023-10-17:
Hi @cngzhnp  The most recent Ubuntu version currently supported by the librealsense SDK is **22.04**.

It is possible to build librealsense from source code with CMake for non-supported Linux versions by including the flag **-DFORCE_RSUSB_BACKEND=TRUE** in the CMake build instruction.  An RSUSB build of the SDK is not dependent on Linux versions or kernel versions and does not need the kernel to be patched.

An example of instructions for performing an RSUSB build from source code can be found at https://github.com/IntelRealSense/librealsense/issues/9931#issuecomment-964289692

cngzhnp, 2023-10-17:
Hi @MartyG-RealSense, thanks for your answer. 

However, I do not use anything else than Dockerfile in the scripts folder for build. Only change I did was, upgrading Ubuntu version from 20.04 to 23.10. In my opinion, with new compiler that comes from newer Ubuntu can find more problems in the codebase.

Simply, missing header file for uint64_t data type which is <cstdint> not included on version.h. Only this would be enough for fixing right now, I guess. If you say that, third-party library is not managed by you, please let me know. I will try to contact corresponding person.

MartyG-RealSense, 2023-10-17:
This error has not been previously reported in relation to librealsense installation, though it has appeared in a number of non-RealSense cases.  In a lot of those cases, a common factor is that **GCC 13** is being used.  Are you using GCC 13, please?

cngzhnp, 2023-10-17:
Yes, I am using `GCC 13.2.0 `in Docker container which comes with Ubuntu Mantic by default, I guess.

MartyG-RealSense, 2023-10-18:
I will highlight this issue to my Intel RealSense colleagues.  Thanks very much for your patience!

MartyG-RealSense, 2023-10-28:
Hi @cngzhnp  Thanks very much for your patience.  I had a response from my Intel RealSense colleagues (below).

*********

Including `<cstdint>` at the top of the version.h should fix the error. 

https://github.com/IntelRealSense/librealsense/blob/master/third-party/rsutils/include/rsutils/version.h

[image]

cngzhnp, 2023-10-28:
Thanks for the reply @MartyG-RealSense. 

Do you plan when this fix will be merged into main branch? Adding one necessary header file, does not affect on current behavior, IMHO.

MartyG-RealSense, 2023-10-28:
I have enquired to my colleagues about your question above regarding whether the change will be added to the master branch.

MartyG-RealSense, 2023-11-02:
Hi again @cngzhnp  My Intel RealSense colleagues have informed me that the issue has been fixed and included in the development branch - please see https://github.com/IntelRealSense/librealsense/pull/11917 for details.

The development branch will become the next master branch release of the SDK when published at a future date.

MartyG-RealSense, 2023-11-09:
Hi @cngzhnp  Do you require further assistance with this case, please?  Thanks!

cngzhnp, 2023-11-13:
Hello @MartyG-RealSense, no thanks a lot. Only one more thing, do you have any plan when it will be merged into main and when you do release a new version?

MartyG-RealSense, 2023-11-13:
I have contacted my Intel RealSense colleagues regarding whether there is a time estimate available for the next master release.  I will let you know as soon as I have a reply.

MartyG-RealSense, 2023-11-21:
Hi @cngzhnp  As the GCC fix has been merged ito the master branch (see the comment above), do you require further assistance with this case please?  Thanks!

cngzhnp, 2023-11-21:
Hello @MartyG-RealSense, 

Thanks for your fast & informative support. IMHO, we can live this change right now. No required any further assistance. 

Best Regards.

closed 2023-11-21 by cngzhnp

MartyG-RealSense, 2023-11-21:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!