realsenseai/librealsense #12305: calibrating the camera

Mamatharun, 2023-10-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version | Linux (Ubuntu 22.04.3)  | 
| Kernel Version (Linux Only)    |  5.15.0-86-generic                                        | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.54.2 }                          | 
| Language                            |  python                         | 
| Segment			|  others                        | 

### Issue Description
Hello,

I need to perform the Intel real-sense D435 camera calibration to get the extrinsic parameters to get the accurate results in the computer vision applications like object detection. Could you please guide me how can I calibrate the camera?

Thank you.

MartyG-RealSense, 2023-10-17:
Hi @Mamatharun  The camera can be calibrated using an **On-Chip Calibration** tool that can be accessed from the 'More' option near the top of the options side-panel of the RealSense Viewer tool.   A guide for this tool can be found at the link below.

https://dev.intelrealsense.com/docs/self-calibration-for-depth-cameras

Mamatharun, 2023-10-17:
Thank you for your response. But, I'm getting the following error when I tried on-chip calibration.

[image]

MartyG-RealSense, 2023-10-17:
If On-Chip does not work on your particular computer then there is an alternative, just as capable calibration tool called **Dynamic Calibration**.  The tool's user guide at the link below has instructions on page 14 onwards for installing it on Linux.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

Mamatharun, 2023-10-17:
Thank you very much. Can I do it programitically by using opencv-python ? 

1. Capture the checkerboard images,
2. Identify the checkerboard corners,
3. calibrating them to get intrinsic's and extrinsic's of both RGB and Depth sensors.

I tried in doing in this way. But, it's hard to find the checkerboard pattern in the depth images. So, if we can do it programmatically, can we do it?

MartyG-RealSense, 2023-10-17:
The OpenCV chessboard calibration tutorial at the link below is a helpful reference.

https://docs.opencv.org/4.x/d4/d94/tutorial_camera_calibration.html

Mamatharun, 2023-10-17:
Thank you for the reference. I can get the Intrinsics of the RGB sensor with the opencv-method. But, not with the depth frames. As I mentioned already it is hard to detect the checkerboard corners in the depth images. So, could you please help me how can I get the intrrinsics of the Depth and extrinsic as well?

Mamatharun, 2023-10-17:
Thank you very much. I'll check that. 

According to this comment " If On-Chip does not work on your particular computer then there is an alternative, just as capable calibration tool called Dynamic Calibration. The tool's user guide at the link below has instructions on page 14 onwards for installing it on Linux.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html"

I tried installing the tool and calibrating the camera. Unfortunately, I'm getting the error mentioned in the screenshot. It is strange as I already using the USB-3.0. Could you please help me? And, also you can find the Device info another screenshot. Please help me.

[image]
[image]

MartyG-RealSense, 2023-10-17:
If there is a part of your USB setup that is USB 2.1 then the camera will be detected as a USB 2.1 device.  This could occur if using a USB cable of your own choice that is a USB 2 cable, or if the camera is plugged into a USB 2.1 hub.

Mamatharun, 2023-10-18:
Thank you very much for your support here. I'll try with different cable.

MartyG-RealSense, 2023-10-29:
Hi @Mamatharun  Do you require further assistance with this case, please?  Thanks!

Tharunguvvala67, 2023-10-29:
Hello. Thank you very much for your response. I got this problem resolved yesterday. Thank you very much for your response again.

MartyG-RealSense, 2023-11-04:
Case closed due to solution achieved and no further comments received.

closed 2023-11-04 by MartyG-RealSense

monajalal, 2023-12-04:
@MartyG-RealSense  similar to this issue, should I also dismiss that I can't do on-chip calibration?

[image]

```
(base) mona@ada:~$ realsense-viewer
 04/12 08:47:34,411 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video0
 04/12 08:47:34,412 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video1
 04/12 08:47:34,412 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video2
 04/12 08:47:34,412 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video3
 04/12 08:47:34,412 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video4
 04/12 08:47:34,413 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video5
 04/12 08:47:34,746 INFO [140685774360832] (synthetic-stream-gl.cpp:80) Initializing rendering, GLSL=1
 04/12 08:47:34,746 INFO [140685774360832] (synthetic-stream-gl.cpp:89)  0 GPU objects initialized
 04/12 08:47:34,746 INFO [140685774360832] (synthetic-stream-gl.cpp:132) Initializing processing, GLSL=1
 04/12 08:47:34,890 INFO [140685774360832] (synthetic-stream-gl.cpp:146)  0 GPU objects initialized
 04/12 08:47:35,015 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video0
 04/12 08:47:35,016 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video1
 04/12 08:47:35,016 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video2
 04/12 08:47:35,016 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video3
 04/12 08:47:35,017 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video4
 04/12 08:47:35,017 INFO [140685774360832] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video5
 04/12 08:47:35,017 INFO [140685774360832] (context.cpp:336) Found 1 RealSense devices (mask 0xff)
 04/12 08:47:35,050 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video0
 04/12 08:47:35,050 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video1
 04/12 08:47:35,051 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video2
 04/12 08:47:35,051 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video3
 04/12 08:47:35,051 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video4
 04/12 08:47:35,051 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video5
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video0
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video1
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video2
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video3 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video3
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video4 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video4
 04/12 08:47:35,052 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video5 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.3/video4linux/video5
 04/12 08:47:35,055 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video0 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video0
 04/12 08:47:35,055 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video1 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video1
 04/12 08:47:35,055 INFO [140685606807104] (backend-v4l2.cpp:657) Enumerating UVC video2 realpath=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8.4/1-8.4:1.0/video4linux/video2
[... cut ...]
 04/12 08:48:22,698 ERROR [140685445416512] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 04/12 08:48:22,765 ERROR [140685445416512] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 04/12 08:48:22,832 ERROR [140685445416512] (synthetic-stream.cpp:51) Exception was thrown during callback: Out of frame resources!
 04/12 08:48:22,894 INFO [140685774360832] (environment.cpp:130) Found 2 unreachable streams, 284 extrinsics deleted

```

MartyG-RealSense, 2023-12-04:
@monajalal  It looks as though you are using Windows.  Therefore, does on-chip calibration work if you download and run the older **2.49.0** version of the RealSense Viewer by selecting **Intel.RealSense.Viewer.exe** from the 'Assets' file list at the link below?

https://github.com/IntelRealSense/librealsense/releases/tag/v2.49.0

[image]

monajalal, 2023-12-04:
@MartyG-RealSense I am actually on Ubuntu 22.04
I only have this problem if I use realsense-viewer GUI
I don't have a problem with ros noetic capture and now I am confused if I mainly use ros noetic ros realsense for my captures, if I should worry about this?

MartyG-RealSense, 2023-12-05:
@monajalal  Thanks very much for the clarification about Ubuntu.

As you are able to generate a 3D pointcloud image in the Viewer, this suggests that your Ubuntu computer's video drivers are okay.

If you are unable to install librealsense version 2.49.0 to test on-chip calibration in version 2.49.0 of the Viewer then you could try resetting your camera's calibration to its factory-new default values using the instructions at https://github.com/IntelRealSense/librealsense/issues/10182#issuecomment-1019854487