realsenseai/librealsense #12306: There is a question about the process of calculating the  “Z-Accuracy (Signed)”

pgswag, 2023-10-19:
### Issue Description
What is "Median_Value"?  I need to find out the median of the "Depth_Error" array in this step or  to multiply each value by 50%.
[image]
[image]

MartyG-RealSense, 2023-10-19:
Hi @pgswag  This section of information comes from the data sheet document for the **SR300** camera model.  Are you using an SR300 camera, or one of the newer models from the 400 Series such as D435 please?

pgswag, 2023-10-19:
hi, I'm using D435.Does this calculation not apply to D435?

MartyG-RealSense, 2023-10-19:
The SR300 and 400 Series are different depth camera technologies.  SR300 is **coded light** and the 400 Series is **stereo** (because it has a pair of left and right infrared sensors that are used to calculate depth).

On 400 Series cameras that measure distance from the front glass instead of the sensor lense, the ground truth Z-value can be calculated simply by adding a fixed minus-value to the depth value.  This minus-value represents the distance between the sensor lenses and the front glass that covers the lenses.

On the D435 camera model the ground truth value can be found by adding **-4.2** (mm) to the depth value.

[image]

pgswag, 2023-10-20:
For this step：''Median_Value = 50% percentile value of Depth_Error''  I want to know  how the "Median_Value"  is calculated.
1.each of "value of Depth_Error" * 0.5    
or  2. Get the median of the depth array. (e.g.  1 3 3 5 6 7 9     the median=5)

MartyG-RealSense, 2023-10-20:
There is not any information available about calculating median_value except for the psuedo-code that you quoted at the beginning of this case.  Reading that code though, median_value is depth_error multiplied by 0.5 as you said in '1'.

***********

Z-accuracy on the 400 Series cameras relates to a phenomenon called **RMS Error**, where error starts at around zero at the camera lenses and increases linearly as distance from the camera increases.  It can be calculated with the formula below.

[image]

pgswag, 2023-10-20:
About "Z-Accuracy"  ,also can see in https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality
[image]
But this calculation seems to be different from the one in pseudocode.I want to know how about the "Median" work in calculating Z-Accuracy.

MartyG-RealSense, 2023-10-20:
SR300 and D435 (the camera that you have) are different camera technologies that work very differently.  The pseudocode in the SR300 data sheet document is not relevant to D435.

The formula above is used to calculate error for D435 and other models in the 400 Series range.

pgswag, 2023-10-20:
Well, thank you for your answer.Have a nice day!

closed 2023-10-22 by pgswag