realsenseai/librealsense #12307: failed to set power state macosx ventura

CG-Spectre, 2023-10-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.15.1 | 
| Operating System & Version | MacOS 13.5 Apple Silicon  | 
| Platform                                 | Mac Realsense Viewer |
| SDK Version                            | v2.50.0 | 

I built the SDK for mac as instructed in the documentation, however whenever I open realsense viewer with `sudo realsense-viewer`, I am bombarded with errors stating `UNKNOWN in rs2_get_option(options:0x6000036dc960, option:Stereo Baseline): failed to set power state` except replace `Stereo Baseline` with many different features each time the error pops up I click "ok" and a new one shows up.

**What I've tried**

I tried building libusb separately as stated in [https://lightbuzz.com/realsense-macos/](https://lightbuzz.com/realsense-macos/)
I've seen posts telling me to copy 99-realsense-libusb.rules, but Im not sure where to copy it for MacOS. 

**Additionally,**

When I try to see if the camera is recognized in pyrealsense2-macosx, it states that one realsense device is connected, but it gives me an error saying no device is found when i try to use it in Python.

MartyG-RealSense, 2023-10-20:
Hi @CG-Spectre  A RealSense user with MacOS Ventura at https://github.com/IntelRealSense/librealsense/issues/11815#issuecomment-1620581786 also had the **Failed to set power state** error when using the RealSense Viewer tool and problems with the Python wrapper.

An alternative approach to the Lightbuzz guide for installing librealsense on Apple Silicon for Ventura is to use the command **brew install librealsense**

https://formulae.brew.sh/formula/librealsense

One RealSense user suggested using **brew install libusb** first before doing brew install librealsense.

jen-co, 2023-10-20:
I am having the same issues on m2 mac with Ventura. I know using sudo is meant to fix it but the USB power issues are still there, albeit it intermittently does work at a very low frame rate. I have tried installing with homebrew with no such luck. Any other ideas?

jen-co, 2023-10-20:
Hi @MartyG-RealSense. Thanks so much for the quick reply! I have tried using both a USB c to c cable and a USB A to C adaptor, without success.

In the past I was able to get the same camera reliably working on an Intel Mac with this USB a to c adaptor (for this to work I had to always use sudo).

I have been forced to upgrade to the m2 and now cannot get it to reliably work unfortunately. Sometimes it runs with sudo but most of the time not.

CG-Spectre, 2023-10-20:
> Hi @CG-Spectre  A RealSense user with MacOS Ventura at https://github.com/IntelRealSense/librealsense/issues/11815#issuecomment-1620581786 also had the **Failed to set power state** error when using [...]

Thanks I'll try doing these things as soon as I can, though I already have tried installing libusb beforehand.

MartyG-RealSense, 2023-10-21:
@CG-Spectre  I look forward to your next update.  Good luck!

@jen-co  If you were using an Intel Mac before and switched to M2 then the problem may be with Apple Silicon in general rather than the M2 version of Apple Silicon specifically.  Which installation method did you use to install the librealsense SDK on your Mac, please?

CG-Spectre, 2023-10-21:
> @CG-Spectre I look forward to your next update. Good luck!
>
> @jen-co If you were using an Intel Mac before and switched to M2 then the problem may be with Apple Silicon in general rather than the M2 [...]

Installing via Brew, unfortunately, didn't have much of an effect, however, I'm now trying to test to see if it works with my own XCode project. I was able to link the header files, but even after linking the dylib file that was generated after building librealsense, I'm getting the error `Library 'realsense2.2.54.2' not found. Even though the dylib file does appear under the Frameworks folder of my project.
`

jen-co, 2023-10-21:
@MartyG-RealSense thanks for clarifying. I installed via brew.

MartyG-RealSense, 2023-10-22:
The librealsense SDK has not officially supported MacOS since 2017's 'High Sierra' version, though the SDK has continued to work well with modern Macs up until Apple Silicon models and MacOS Monterey. 

Monterey was very difficult to get working with RealSense.  The RealSense MacOS community exchanged advice with each other to work through the problems, and eventually the Lightbuzz installation guide was created.

Ventura also has problems with librealsense, and the solution is again likely to come from the community of Ventura users as Intel no longer provides Mac support for RealSense and has no Mac computers to test with unfortunately.

MartyG-RealSense, 2023-10-26:
Hi @CG-Spectre  You are very welcome.  Thanks very much for the update.

MartyG-RealSense, 2023-11-09:
Case closed due to no further comments received.

closed 2023-11-09 by MartyG-RealSense

kirienkomaxym, 2023-11-15:
Facing the same issue with macos sonora 14.1

Tried almost everything, no matter what it simply ends-up with `segmentation fault` or `failed to set power state`

The lib installed from brew (`brew install librealsense`) gives following logs:
`
 15/11 15:28:52,992 INFO [0x1e6145ec0] (context.cpp:336) Found 1 RealSense devices (mask 0xff)
 15/11 15:28:53,068 INFO [0x1e6145ec0] (rs.cpp:2697) Framebuffer size changed to 1209 x 781
 15/11 15:28:53,068 INFO [0x1e6145ec0] (rs.cpp:2697) Scale Factor is now 1
[1]    59744 segmentation fault  sudo realsense-viewer
`
Another approach was to build everything from source using the script here (does almost the same suggested here: https://lightbuzz.com/realsense-macos/) -- https://github.com/yugasun/pyrealsense2-mac/blob/main/build.sh
This solution faces the issue of `failed to set power state`.

If anyone have any solution to make this work on macos sonoma with apple m1-m3 processors please let me know

philchao, 2024-03-05:
@kirienkomaxym  I wonder if you may have found a solution to your issue? I am faced with the same problem currently. Thanks

zumpchke, 2024-03-27:
Woww.. Can't believe it's so difficult to get working with M1.

Nabla7, 2025-03-28:
2025 and this issue still persists it would seem, ran into the same error on the m4 max chip.

josiahbjorgaard, 2025-06-08:
Still having issues

MartyG-RealSense, 2025-06-08:
The most recent fixes were implemented in August 2024 for librealsense SDK 2.56.1 onwards, fixing source code compilation on Mac 'M' processors and adding support for MacOS 14 (Sonoma) - please see https://github.com/IntelRealSense/librealsense/pull/12972 for more information.