realsenseai/librealsense #12312: How to implement Post-Processing with Intel® RealSense SDK API in MATLAB

pgswag, 2023-10-22:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Operating System & Version |  win10| 
| Platform                                 | PC |
| SDK Version                            | 2.0                          | 
| Language                            |  MATLAB                         | 

### Issue Description
I want to use Spatial-Filters、hole-filing and temporal-filter in MATLAB, how do I add them in code.

MartyG-RealSense, 2023-10-22:
Hi @pgswag  A user of the RealSense MATLAB wrapper shared their post-processin filter script at https://github.com/IntelRealSense/librealsense/issues/4400#issuecomment-511197098

MartyG-RealSense, 2023-10-29:
Hi @pgswag  Do you require further assistance with this case, please?  Thanks!

pgswag, 2023-10-29:
Thanks for your help，my problem is solved.

closed 2023-10-29 by pgswag

MartyG-RealSense, 2023-10-29:
That's great to hear.  Thanks very much for the update!