realsenseai/librealsense #12317: Regarding the use of Halcon for hand eye calibration of six axis robotic arms and realseneD405, there was a significant position error

zhanglepy, 2023-10-25:
[image]

zhanglepy, 2023-10-25:
By searching for someone who said the camera was set to mirror, just uncheck the mirror. But I don't know where the mirror settings of the realsense D405 camera are and how to modify them.

zhanglepy, 2023-10-25:
[image]

MartyG-RealSense, 2023-10-25:
Hello @zhanglepy  I see that you are using the SCARA robot hand-eye calibration software at the link below.

http://www.skcircle.com/?id=1241

RealSense cameras do not have a mirror setting, so I am not certain what that refers to.

Can you tell me more about how hand-eye calibration is performed with Halon and the D405, please?

closed 2023-11-09 by MartyG-RealSense