realsenseai/librealsense #12319: pyqt5,一但引用了pyrealsense2，就无法打开QtWidgets.QFileDialog

FranticEr, 2023-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (11)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
使用python，pyqt5,一但引用了pyrealsense2，就无法打开QtWidgets.QFileDialog

MartyG-RealSense, 2023-10-26:
**Using python, pyqt5, once pyrealsense2 is referenced, QtWidgets.QFileDialog cannot be opened.**

Hello @FranticEr  The discussion at https://github.com/IntelRealSense/librealsense/issues/6174 may be a helpful reference.

你好@FranticEr https://github.com/IntelRealSense/librealsense/issues/6174 上的讨论可能是一个有用的参考。

MartyG-RealSense, 2023-11-04:
Hello @FranticEr  Do you require further assistance with this case please?

您好@FranticEr 对于此案您需要进一步帮助吗？

FranticEr, 2023-11-04:
问题得到了解决，非常感谢！

MartyG-RealSense, 2023-11-04:
You are very welcome.  That is excellent news that your problem has been solved.  Thanks very much for the update!

非常欢迎您。 这是个好消息，您的问题已得到解决。 非常感谢您的更新！

closed 2023-11-04 by MartyG-RealSense