realsenseai/librealsense #12321: HW synchronisation 3 time D415

andyatphilips, 2023-10-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |  5.15.1 | 
| Operating System & Version |   Win 10  | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | PC  |
| SDK Version                            | 2.54.2                         | 
| Language                            |  C++                          | 
| Segment			| VR/AR                         | 

### Issue Description
Use case: 3 x  D415 real sense cameras with depth and RGB image all synchronized.

I've been reading a lot of issues on this site related to real sense HW/SW synchronization.

This issue is to confirm my findings:

1. Only depth sensors can be synchronized thru HW synchronization.
2. RGB sensors cannot be synchronized thru HW synchronization.
3. The only way to synchronize RGB sensors is thru timestamp algorithms.

So, my questions:

1. Is above correct?
2. For the D415 depth sensor and RGB sensor are on the same circuit board. Depth sensor can be HW synchronized. Is the start of the RGB sensor exposure somehow derived from the start of the depth sensor start? And if so, how?

Thanks in advance.

MartyG-RealSense, 2023-10-26:
Hi @andyatphilips   For the most part, only depth is synched.  An exception to this is Inter Cam Sync Mode '3', which is called **Full Slave**.  It was intended for use with the D415 model to sync depth and RGB.

However, it was unreliable and Intel removed support for it, though the mode is still usable in the RealSense SDK.  It should behave like Mode 2 (Slave, which is depth only).

Regarding sync between depth and color on the same camera, the color sensor always wants to be the master over depth.  making it problematic in a multiple camera setup - please see https://github.com/IntelRealSense/librealsense/issues/2637#issuecomment-434203140 for more information about this,

andyatphilips, 2023-10-26:
Hi @MartyG-RealSense, thanks for your very quick response!

I think I have all the info needed to determine our approach.
Thanks again, you may close this issue.

MartyG-RealSense, 2023-10-26:
You are very welcome.  Thanks very much for the update!

closed 2023-10-26 by MartyG-RealSense

ahjcahkl, 2024-09-04:
Hello! In the process of hardware synchronization with D435 and D455 cameras, I can observe the millisecond offset of their respective channels, but the mapping speed of these two cameras is not the same, the drawing time of D435 is about 33ms, but the drawing time of D455 is about 20ms, does this realize hardware synchronization?

MartyG-RealSense, 2024-09-04:
Hi @ahjcahkl  Are you hardware syncing two different models of RealSense camera, please (D435 and D455).  If you are then whilst syncing different models can work, it is not recommended as it is preferable to use the same model for all cameras in the multiple camera setup.  This is because hardware differences between different models could result in the kind of discrepancy that you describe.