realsenseai/librealsense #12323: Camera is accessible in the IP-tool but cannot be detected in realsense viewer

MuningXu, 2023-10-26:
I'm trying to get access to a framos D453E camera. I can see it accessible in the IP-tool in camerasuite, but it cannot be detected in the realsense viewer. What's the problem?

The system is Ubuntu 20.04 in VMware. Is the problem due to the virtual machine?

[image]
[image]
[image]

MartyG-RealSense, 2023-10-27:
Hi @MuningXu  Intel do not provide technical support for the FRAMOS range of RealSense-compatible 'e' cameras as they are not Intel products.  Please contact FRAMOS support with your question by visiting their support page for their 'e' camera models at the link below and clicking the 'Contact Us' or 'Let's Talk' buttons on the page.

https://www.framos.com/en/industrial-depth-cameras#downloads

The RealSense ROS wrapper does not have support for the FRAMOS 'e' cameras.

[image]

The RealSense SDK and Viewer version used with the 'e' cameras should be the **software package** version provided at the FRAMOS download link above.  The download button on that page is not currently enabled but I obtained the download links.

**tar.gz - Linux**
https://www.framos.com/framos3d/D400e/Software/Latest/FRAMOS_D400e_Software_Package_Linux64_x64_latest.tar.gz

**zip- Windows**
https://www.framos.com/framos3d/D400e/Software/Latest/FRAMOS_D400e_Software_Package_Win64_x64_latest.zip

MuningXu, 2023-10-27:
Thank you for the reply. I followed the instructions here https://support.framos.com/support/solutions/articles/48001217984-setup-ros-wrapper-from-librealsense2-and-debugging-tips and it used D435e as the example.

MartyG-RealSense, 2023-10-27:
I do not have advice that I can provide on this subject, unfortunately.  We do not provide support for 'e' cameras or have any to test with.  I do apologize.

MartyG-RealSense, 2023-11-04:
Case closed due to issue being FRAMOS related.

closed 2023-11-04 by MartyG-RealSense