realsenseai/librealsense #12336: Is there a way to build the plugin for UE 4.21.2? RealSenseUE and RunTimeMeshComponent are built with a different engine version

ColinCoderUmich, 2023-10-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-10-29:
Hi @ColinCoderUmich   The RealSense UE4 wrapper had support for UE 4.19 and newer, up a maximum of 4.22, until SDK version **2.41.0**.  

https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2410

After that the wrapper had an update that enabled UE 4.26 support but introduced breakages for UE versions older than 4.26.

There were not updates to the UE4 wrapper after the 4.26 support.  A member of the RealSense community shared a new plugin created for UE5 at https://github.com/IntelRealSense/librealsense/issues/12262

MartyG-RealSense, 2023-11-05:
Hi @ColinCoderUmich  Do you require further assistance with this case, please?  Thanks!

ColinCoderUmich, 2023-11-06:
Hi, is there support for Unreal Engine 5.3 as well? Thank you!

ColinCoderUmich, 2023-11-06:
That answers my questions. Thank you!

MartyG-RealSense, 2023-11-06:
You are very welcome.  As your questions are answered, I will close this case.  Thanks again!

closed 2023-11-06 by MartyG-RealSense