realsenseai/librealsense #12338: Is there any tools or code for python to get the Pose Transformation Matrix of D400 with IMU

Easenyi, 2023-10-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |                                      | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.<54>.<2> }                          | 
| Language                            |  {python}                          | 
| Segment			|  {Robot/VR/AR }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2023-10-31:
Hi @Easenyi  Only the **RealSense T265 Tracking Camera** model, which also has an IMU, supports a 6DOF pose data stream.  This model is now retired and no longer supported by the RealSense SDK though.

It is possible to obtain the IMU tilt and rotation angles of an IMU-equipped D400 camera such as D435i with Python code - as demonstrated at https://github.com/IntelRealSense/librealsense/issues/4391 - but the camera does not know its own position.

Easenyi, 2023-10-31:
@MartyG-RealSense Thank you for your response. However, what I meant to ask is whether there is ready-made python code or tools available to calculate the transformation matrix between two points in time based on the tilt and rotation angles from an IMU.
Because I have tried to do this  
`config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
    config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 200)
    config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)`
`aligned_frames = align.process(frames)
    d = aligned_frames.get_depth_frame()
    c= aligned_frames.get_color_frame()`
`   initial_translation = np.array([0.0, 0.0, 0.0])
    initial_rotation = Rotation.from_euler('XYZ', [0, 0, 0], degrees=True)
    time_interval = 1.0 / 200  
    accel_frame = frames.first_or_default(rs.stream.accel)
    gyro_frame = frames.first_or_default(rs.stream.gyro)
    # Validate that both frames are valid

    accel_data = accel_frame.as_motion_frame().get_motion_data()
    gyro_data = gyro_frame.as_motion_frame().get_motion_data()

    rotation_rate = np.array([gyro_data.x, gyro_data.y, gyro_data.z])
    rotation = Rotation.from_euler('XYZ', rotation_rate * time_interval, degrees=True)
    initial_rotation = initial_rotation * rotation

    acceleration = np.array([accel_data.x, accel_data.y, accel_data.z])
    position_change = 0.5 * acceleration * (time_interval ** 2)
    initial_translation += position_change

    rotation_matrix = initial_rotation.as_matrix()
    translation_vector = initial_translation
    pose = np.eye(4)
    pose[:3, :3] = rotation_matrix
    pose[:3, 3] = translation_vector`
1、 Because the sampling frequencies of the image data stream and the IMU data stream are different, I am unsure how to align them. 
2、 I am uncertain if my method for calculating the rotation matrix is correct, as the quality of the reconstructed point cloud is poor. It is possible that the noise and jitter from the IMU itself are contributing factors.

MartyG-RealSense, 2023-10-31:
I am not aware of an existing code sample that demonstrates the transformation that you described, unfortunately.

However, each IMU data packet is timestamped using the depth sensor hardware clock to allow temporal synchronization between gyro, accel and depth frames.  So the IMU accel and gyro timestamps should already be in sync to depth.

In regard to aligning IMU timestamps to color, https://github.com/IntelRealSense/librealsense/issues/3205 recommends using **hardware metadata** to do so.

MartyG-RealSense, 2023-11-09:
Hi @Easenyi  Do you require further assistance with this case, please?  Thanks!

Easenyi, 2023-11-09:
@MartyG-RealSense No further assistance is currently needed.Thank you.

MartyG-RealSense, 2023-11-09:
You are very welcome.  As you do not currently require further assistance, I will close this case.  Thanks very much for the update!

closed 2023-11-09 by MartyG-RealSense