realsenseai/librealsense #12341: RSUSB vs v4l2 and kernel patches.

GrubbyHalo, 2023-10-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |  (6.0)                                         | 
| Platform                                 | Raspberry Pi  |
| SDK Version                            |  2.54.1                           | 

What is the situation with the realsense camera's being supported by the linux kernel version 6.0 and up ? I have noticed that unless i am building with RSUSB enabled I wont get the required metadata to enable the HDR functionality on the 400 series camera's to work properly (i.e. when using the v4l2 backend). I dont really like using the RSUSB UVC backend  as I have experienced issues with it. 

So I am assuming that in order to provide kernel support the patches located in the scripts folder still need to be applied however some of them seem out of date. Is it possible to update the scripts for the patches so that they are more current ? e.g. the last debian script was for debian 10 . We are now on 12.

Cheers

MartyG-RealSense, 2023-10-31:
Hi @GrubbyHalo  There have been a couple of past cases where RealSense users have successfully edited a kernel patch script to add support for a kernel not yet officially supported.  An example of this is a PR at https://github.com/IntelRealSense/librealsense/pull/9727 which added kernel 5.11 support when the most recent official version supported at that time was 5.8. 

Another example of an edit was one at https://github.com/IntelRealSense/librealsense/commit/13094ea0b8cfb6955ce6e69d18138afc125f56aa which added support for JetPack 4.6 in the Nvidia Jetson patch script.

GrubbyHalo, 2023-10-31:
@MartyG-RealSense . Thanks. Following issues referenced in the links above, answered my questions.

closed 2023-10-31 by GrubbyHalo

MartyG-RealSense, 2023-10-31:
You are very welcome.  I'm pleased that I could help.  :)