realsenseai/librealsense #12347: Issue with Exporting PLY Data from Intel RealSense D455 Camera

sytong02, 2023-11-02:
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D455|
|**Serial Number**|234222301584|
|**Firmware Version**|5.15.1|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|226523134184|
|**Firmware Update Id**|226523134184|

I am an architecture student using the D455 as part of my thesis. I have no programming experience, but I am facing difficulties using the `rs.convert(ply)` tool. In my attempt to address this issue, I referred to GitHub issue #7678 and tried to export PLY data instead of bag data using the sample Python script called `export_ply_example.py` available on this GitHub page:

[https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py](https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py)

I ran the script in Visual Studio Code, and I encountered the following error message:
`PS C:\Users\xx> & C:/Users/xx/AppData/Local/Programs/Python/Python38/python.exe c:/Users/xx/Downloads/export_ply_example.py
Traceback (most recent call last):
  File "c:/Users/xx/Downloads/export_ply_example.py", line 32, in <module>
    frames = pipe.wait_for_frames(1000)
RuntimeError: Frame didn't arrive within 1000`

Could you please check if there are any issues with the following steps:

1. Open Intel RealSenseViewer v2.54.2.
2. Turn on the Intel RealSense D455.
3. Run the `export_ply_example.py` script.

I would appreciate any assistance you can provide in resolving this issue. Thank you.

sytong02, 2023-11-03:
Hi, thank you for your reply!

1. **Objective**: My ultimate goal is to export a series of PLY files from a recording, capturing a sequence of frames.

2. **Using Python Script**: I have installed the `pyrealsense2` library on my computer, which is necessary for interfacing with the RealSense camera. I attempted to run the provided Python script `export_ply_example.py` via the command prompt, as follows:

```plaintext
C:\Users\xx>pip install pyrealsense2
Requirement already satisfied: pyrealsense2 in c:\users\xx\appdata\local\programs\python\python38\lib\site-packages (2.54.2.5684)

C:\Users\xx>cd C:\Users\xx\Downloads

C:\Users\xx\Downloads>python export_ply_example.py
Saving to 1.ply...
Done
```

3. **Issue with Export**: The script executed successfully without any errors. However, I expected to see multiple PLY files saved as a result of running the script. My goal was to have each frame from the recording exported as a separate PLY file.

4. **Current Outcome**: Unfortunately, I did not find any PLY files saved at the specified file save location. This is the main issue I'm facing.

5. **Using the Export Button**: I also explored the RealSense Viewer tool, which has an "Export" button in 3D mode. However, it seems to allow exporting only one frame of a PLY file at a time, which is not practical for my requirement of exporting multiple frames.

I would greatly appreciate your guidance and support in resolving this issue. My thesis project heavily relies on accurately exporting and analyzing the point cloud data from the RealSense camera recording.

If you could provide step-by-step instructions or advice on how to configure the RealSense camera or the Python script to export multiple PLY files from a recording, it would be immensely helpful.

Thank you for your time and assistance.

MartyG-RealSense, 2023-11-03:
You may be able to achieve your goal with **export_ply_example.py** if you edit its code so that it loops the script by commenting out the **pipe.stop()** instruction at the end of the script.

You would also have to rewrite the filename code so that instead of using the fixed ply name "1.ply", it increases the filename by 1 for each time that the script loops (1.ply, 2.ply, 3.ply, etc).  This can be done by generating a filename from separate strings that are then assembled together into a single name (e.g "1" and ".ply" makes "1.ply").  The principles of doing so are discussed at the link below.

https://bobbyhadz.com/blog/python-create-file-name-using-variables

sytong02, 2023-11-04:
Hi, thank you for your response. 

I'm pleased to let you know that I've successfully exported all my .ply files using the rs-convert tool. I managed this by moving the .bag file in the same folder as the tool (C:\Program Files (x86)\Intel RealSense SDK 2.0\tools), and then running it in the command prompt. I hope this might be helpful for your reference in the future.

MartyG-RealSense, 2023-11-04:
You are very welcome.  It's great to hear that you were successful with ply export via rs-convert.  Thanks so much for the the update and for sharing the details of your solution!   As your problem is solved, I will close this case.  Thanks again!

closed 2023-11-04 by MartyG-RealSense