realsenseai/librealsense #12362: Python ImportError: No module named pyrealsense2

mericgeren, 2023-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435  | 
| Firmware Version                   | 5.15.00.02 | 
| Operating System & Version |   L4T 32.7.1 (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  4.9.337-tegra                                         | 
| Platform                                 | NVIDIA Jetson AGX Xavier  |
| SDK Version                            |  2.50.0                          | 
| Language                            |  Python                          | 
| Segment			|  RobotVR/AR/others                          | 
| realsense-ros Version     |  2.3.2

### Issue Description
Hello, 

I am trying to run a ROS1 node I have written previously in Python using RealSense SDK v2.50.0, it's Python wrapper and realsense-ros version 2.3.2. For that, by following the [guide](https://github.com/IntelRealSense/librealsense/tree/v2.50.0/wrappers/python) I have installed and built RealSense SDK 2.50.0 with Python wrapper successfully. Then, I have added the following line to `.bashrc`:

`export PYTHONPATH=$PYTHONPATH:/usr/local/lib`

Then, sourced .bashrc. Then, by followin this [guide](https://github.com/IntelRealSense/realsense-ros/tree/2.3.2) installed and did catkin_make for ddynamic_reconfigre, realsense-ros (version 2.3.2) and then, my package. Then, I have done `source devel/setup.bash`. After sourcing I have tried to launch my node with roslaunch. But, it exited and the process have died. While looking at the output, I saw these lines:

 ```

    import pyrealsense2 as pyrs
ImportError: No module named pyrealsense2

```

Later on, I have tried to install the `pyrealsense2` package with `pip install` command and tried following:

`sudo pip3 install pyrealsense2==2.50.0`

`sudo pip3 install pyrealsense2==2.50.0.3812`

`pip install pyrealsense2==2.50.0.3812`

and 

`pip install pyrealsense2==2.50`

But unfortunately, after each trial, I saw this error message:

`Could not find a version that satisfies the requirement pyrealsense2==X (from versions: )
No matching distribution found for pyrealsense2==X`

> Note that X represents versions of pyrealsense2.

Could you offer your help please?

Kindest regards.

MartyG-RealSense, 2023-11-06:
Hi @mericgeren  The import instruction should be `import pyrealsense2 as rs` rather than pyrs.  Can you confirm please whether you used pyrs or rs in your import instruction, please? 

When installing from PyPi packages with `pip install pyrealsense2`, installation on Arm devices such as Jetson with the **aarch64.whl** packages is currently only supported up to Python 3.9.  Newer Python versions such as 3.11 are supported when building pyrealsense2 from source code.  

Which Python 3 version are you using?

In regard to pip versions: whilst the release history page lists older versions, the download page only lists packages for SDK 2.54.2.

https://pypi.org/project/pyrealsense2/#files

mericgeren, 2023-11-06:
Hi @MartyG-RealSense

Thank you for your kind response. Ofcourse, I can confirm that I have used `pyrs` at the import. However, when I have tried another ROS node I have written in Python with the same version of RealSense SDK, in which I have used `import pyrealsense2 as rs` rather than pyrs, it exited and the process have died again and I have seen this output:

```
    import pyrealsense2 as rs
ImportError: No module named pyrealsense2
```

MartyG-RealSense, 2023-11-06:
On some computers, the import instruction `import pyrealsense2.pyrealsense2 as rs` will sometimes work when **import pyrealsense2 as rs** has not.

The **aarch64.whl ** packages were introduced from SDK 2.54.1 onwards.  Before that, the pip install process only supported x86 / x64 computers such as desktop / laptop PCs but not Arm devices.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-707501049 has a guide for installing librealsense and pyrealsense2 on Jetson and Python 3.6 if you have not seen it already.

mericgeren, 2023-11-06:
Thank you for all the help you have offered.

Building librealsense after adding `-DPYTHON_EXECUTABLE=/usr/bin/python2.7` to cmake command and then, installing request package with pip install, have solved the issue and as i did roslaunch, it started and ran without any error and any processes getting killed as it turned out ROS was installed and built for Python 2.7 and  RealSense python wrapper have built for Python 3.6 and that version and direction difference made ROS unaware of RealSense python wrapper. To summarize, in order to make RealSense python wrapper work with ROS, you should build and install both for the same python version. If you don't mind i am closing this issue as completed.

Kindest regards.

closed 2023-11-06 by mericgeren

MartyG-RealSense, 2023-11-06:
You are very welcome.  Thanks so much for sharing the details of your solution!