realsenseai/librealsense #12365: "No frames received" issue for D435i IP device

SeanShi13, 2023-11-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i IP device | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Linux (Ubuntu 20.04) | 
| Kernel Version (Linux Only)    | 5.15.0-88                                     | 
| Platform                                 | PC  |
| SDK Version                            |   2.53.1                       | 
| Language                            |  python                        | 
| Segment			|  Robot                        | 

### Issue Description
I used a Raspberry Pi 4B board with the pre-built image to connect ubuntu system to a D435i camera through its ip address. However, in ubuntu realsense-viewer, it reported "No frames received" after successful connection. 
[image]
With the associated messages after I click on the triangular symbol it suddenly exited.
[image]

[image]

I wonder what's the problem and how can I fix it.

MartyG-RealSense, 2023-11-06:
Hi @SeanShi13  May I first ask which camera firmware driver version is installed on your camera, please?  You can find the firmware by clicking on the **Info** option at the top of the RealSense Viewer's options side-panel.  RealSense SDK version 2.53.1 should be used with firmware version **5.14.0.0**.

SeanShi13, 2023-11-06:
@MartyG-RealSense Thanks for your timely reply. The firmware of the camera is 05.15.00.02. Plus, it always shows that I am using USB 2.1 port to connect the camera, whereas I am using USB 3.0 cable all the time.

> Hi @SeanShi13 May I first ask which camera firmware driver version is installed on your camera, please? You can find the firmware by clicking on the **Info** option at the top of the RealSense [...]

MartyG-RealSense, 2023-11-06:
Using a firmware version that is not designed for a particular SDK version can result in errors.  Firmware 5.15.0.2 is designed for SDK 2.54.1, the next one after the 2.53.1 that you are using.

If you are using a USB 3 cable but the camera is detected as USB 2.1, this could indicate that the camera is plugged into a USB 2 port on the computer or is plugged into a USB hub that is USB 2.1 istead of USB 3.  

Are you using the official 1 meter USB cable supplied with the camera?  If using your own choice of cable then it is possible to accidentally use a USB 2.1 cable unless it is certain that the cable is USB 3.  A USB 3 device with a USB 2.1 cable will be detected as 2.1.

The camera can also sometimes be mis-detected as USB 2.1 if it is not inserted all of the way into the USB port with a quick, firm insertion.  If the connector is only around three-quarters into the port then the camera will often be detected as 2.1.

SeanShi13, 2023-11-06:
Thanks. I see. Now if I connect the camera directly to the ubuntu system, it will work normally. However, if I use the Raspberry Pi as a medium, it won't show any frame. Could it be the problem of the image of the Raspberry Pi, since I compile it by myself and I am not sure whether I did it correctly. By the way, I used the pre-built image provided in https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras. And it can also show frames in the Raspberry Pi system with realsense-viewer.

> Using a firmware version that is not designed for a particular SDK version can result in errors. Firmware 5.15.0.2 is designed for SDK 2.54.1, the next one after the 2.53.1 that you are using.
>
> If [...]

MartyG-RealSense, 2023-11-06:
What resolution and FPS speed are you using for depth and color on the network connection, please?  A network device connection only supports a very limited number of resolution and FPs combinations (see the link below), so the best settings to test a network device with would be **640x480** resolution and **30** FPS for both depth and color.

https://dev.intelrealsense.com/docs/open-source-ethernet-networking-for-intel-realsense-depth-cameras#36-software-limitations

You mentioned the pre-built SD card image and that it worked in the Viewer.  My recollection is that this image has a very old RealSense SDK version of 2.34.0.  Do you mean that you tried this network-equipped image and it worked correctly, but you compiled SDK 2.53.1 and it didn't show the frames, please?

If the SDK is compiled from source code with CMake then the CMake build instruction should include the flags **-DBUILD_NETWORK_DEVICE=ON** and **-DFORCE_RSUSB_BACKEND=ON**.  

Compile the SDK with the -DBUILD_NETWORK_DEVICE=ON network support flag on Pi so that the **rs-server** and ** realsense2-net** components are installed on the Pi, and then create an SSH connection from the main host computer to the Pi to start rs-server on the Pi and set up the network device connection in the realsense-viewer on the **host computer**.

SeanShi13, 2023-11-10:
> Do you mean that you tried this network-equipped image and it worked correctly, but you compiled SDK 2.53.1 and it didn't show the frames, please?

I have been trying to fix the problem these days but failed. The Realsense sdk version of my Raspberry Pi image is  v2.34.1. The device can show frames on the realsense-viewer of my Pi server but not on my host PC. I wonder whether the setting static step is necessary, because I just type in the ip of my server in the realsense-viewer of my host and there's no frames received on my host.

SeanShi13, 2023-11-10:
Also, do I need to compile the realsense package in the pre-built image of Raspberry Pi? Or it has been compiled.

MartyG-RealSense, 2023-11-10:
May I confirm that you have not got the realsense-viewer active on your Pi server when trying to access the frames from the realsense-viewer on the host computer?  Once a stream is enabled by a program then it prevents a second program from accessing the same stream.  

If the realsense-viewer on the host was opened secondly after the Pi viewer was opened then this could manifest as No Frames Received when trying to access the streams on the host.   This principle is demonstrated in the image below, where I enabled the depth and color streams in the **rs-capture** example first and then tried enabling the same streams in realsense-viewer afterwards whilst rs-capture was still using the streams.

[image]

SeanShi13, 2023-11-13:
> May I confirm that you have not got the realsense-viewer active on your Pi server when trying to access the frames from the realsense-viewer on the host computer?

It's confirmed that the realsense-viewer on the Pi server was not opened firstly. I only opened the viewer on my host pc. Plus, no matter I run the "rs-server" command on the Pi server or not, there's no frame received on my host. Also, I am using the offered cable with USB 3.2 port.

MartyG-RealSense, 2023-11-13:
The instructions in the ethernet guide are admittedly presented in a format that is difficult to read, so I will outline the steps to take when setting up the connection from the host to the Pi.

1.  Open a secure shell on the host computer and login to the Pi via SSH using the initial username and password **realsense**

`ssh realsense@10.55.0.1`

2.   While logged-in to the Pi via secure shell, enter the fllowing instruction:

`sudo nano /etc/dhcpcd.conf`

3.  After entering your password, navigate to the “Example static IP configuration” section to set a static IP address.

The image below demonstrates how to set a static IP address.

[image]

Uncomment ip_address, routers and domain_name_servers lines.

4.  Enter new intended static IP address in **ip_address line**. The line must end with /24, do not remove it.  

The image above suggests using the IP address **10.0.0.99/24**

5. Enter the IP address of your network router under **routers** and **domain_name_servers**.

To find out your router IP, on Linux type-in `ip route | grep default` in a new terminal window.

6.  Press Control + X, Y to save changes.  Then restart the system (using sudo reboot) to apply the new configuration.

7.  Open the RealSense Viewer on the host computer and click **Add Source > Add Network Device**.  Enter the camera IP address.  The white-paper guide suggests the IP address **10.0.0.99**

8. Toggle the depth and RGB streams on in the Viewer to view them.

SeanShi13, 2023-11-13:
Thanks for your further extraction. Maybe I am using a display to directly show the Pi server interface so that I don’t need to log into it via ssh approach. More concerns of mine are about the operations needed on the Pi server assuming that I am starting up with the pre-built image by realsense.  For example, do I need to compile the build file of the realsense? Or what else should I do. Plus I wonder do I need to type in “rs-server” each time in the command window of the Pi server before connecting the ip device?

MartyG-RealSense, 2023-11-13:
I have not used this ethernet tool myself but some RealSense users have reported that they had to build librealsense from source code with networking enabled on the host computer too in order for it to work.

If you are using the pre-made Pi SD card image then all the necessary SDK files and networking components should already be installed.  Though as mentioned above, you might need to build librealsense from source on the host computer with networking support enabled by including the **-DBUILD_NETWORK_DEVICE=ON ** build flag.

I have wondered myself why there is not a launch command for rs-server mentioned in the instructions.  Perhaps the Viewer's network device interface automatically launches the rs-server component on the Pi when establishing a connection to the camera attached to the Pi but I have no information to confirm that.

This rs-server networking system has now been retired as of SDK 2.54.1 onwards.  Intel plan to introduce a new networking interface in an upcomng SDK release.

Which version of librealsense is on the Pi SD card image?  It was 2.34.0 when the image was originally created, which is very old now and not fully compatible with current-day RealSense camera firmware driver versions.

SeanShi13, 2023-11-13:
I see. Now I am using sdk 2.34.1 on the Pi server. When I first type in “rs-server” in the command window of my Pi server, it did not recognize this command. So later I compiled the build file in the pre-made image myself, with the needed cmake flags, which may not be necessary?

MartyG-RealSense, 2023-11-13:
You do not need to compile the SDK if you have installed the SD card image on the Pi and should not install it on top of the card image, as you could end up with two versions of librealsense on the same Pi that conflict with each other.

Because of the significant age of the 2.34 SDK in the SD card image, it is not very useful in the present day except for using it as a reference to see what a self-compiled network enabled build of the SDK should look like.

SeanShi13, 2023-11-14:
I see. I formatted the SD card and burned the pre-made image into the SD card. Then I set the static ip address of the Pi server but still received the same no frames error. Could you confirm that I don't need to compile the sdk on the Pi server, please? Because the Pi server cannot recognize nor run any "rs" or "realsense" command now. Also, is there any way through which I can check the network device support on my host is valid? Thank you very much.

MartyG-RealSense, 2023-11-14:
Burning the pre-made image with **balanaEtcher** should be all that is required, with no compilation of the SDK necessary as everything needed should be included in the image.  

A complicating factor though is that this image of the 2.34 SDK was originally designed to be used with the very old camera firmware driver version **5.12.3.0**.    The D435i camera model only supports a minimum of the later firmware version **5.12.7.100**.  After the 5.12.7.100 firmware, SDK **2.39.0** became the minimum supported SDK.  The situation increased in complexity later, as from around SDK 2.50.0 onwards each SDK version has had a single specific recommended firmware version to use with it.

So in summary, forgetting about using the SD card image and instead compiling librealsense yourself with network support is likely to be the best option.  You can only do this up until SDK 2.53.0 though, as support for the ethernet networking system was removed in 2.54.1 onwards.  So if you compiled the latest version of the SDK by compiling it from the GitHub instead of downloading and extracting the source code of 2.53.1 or older from a zipped file then the networking support would not be available.

I am not aware of a means to check the status of the networking connection on the host, unfortunately.  The network camera access in the Viewer either works or it doesn't.

MartyG-RealSense, 2023-11-21:
You are very welcome.  It is planned that the next librealsense SDK release will be in the first quarter of 2024.  It is also planned that a new networking system will be released to replace the rs-server one.