realsenseai/librealsense #12373: failed to set power state error

Nimaro76, 2023-11-07:
Hello,

I've recently upgraded my system to Ubuntu 22 with kernel 6.02. Consequently, I enabled the Force_RSUSB_Backend during the build process. However, since this change, my c++ codes that functioned correctly on Ubuntu 20 and I used to get the serial number per camera in it is no longer working. The typical error message I encounter is:

terminate called after throwing an instance of 'rs2::error' what(): failed to set power state Aborted (core dumped)

Could you assist me in identifying the potential cause of this issue? If the problem stems from the absence of a kernel patch, could you inform me about Intel's schedule for updating the SDK and releasing new patches, particularly for kernel 6.02? Your guidance would be greatly appreciated.

MartyG-RealSense, 2023-11-07:
Hi @Nimaro76  There should not be a kernel related conflict when RSUSB = true as an RSUSB-enabled build of librealsense bypasses the kernel and so it should not matter which kernel version is installed on the computer as librealsense will be non-dependent on Linux versions or kernel versions and not require kernel patching.

If RSUSB = true does not work for you and you require a quick solution, an approach that could be considered is to downgrade your kernel version, as the 2.54 SDK already officially supports use of kernel **5.15** with Ubuntu 22.04 Jammy by applying the patch script **patch-realsense-ubuntu-lts-hwe.sh**

**************

The **development** branch of librealsense (which becomes the next master branch of the SDK once pushed to release) currently has a patch script for kernels 6.0 to 6.4 called **realsense-metadata-jammy-hwe-6.2.patch**

https://github.com/IntelRealSense/librealsense/blob/development/scripts/realsense-metadata-jammy-hwe-6.2.patch

For kernel 6.5 and newer, a patch will not need to be applied.

A member of the Intel RealSense team with expertise in kernel patches provides advice about these new kernel developments at https://github.com/IntelRealSense/librealsense/issues/11745#issuecomment-1695116725

MartyG-RealSense, 2023-11-13:
Hi @Nimaro76  Do you require further assistance with this case, please?  Thanks!

Nimaro76, 2023-11-13:
No , problem solved. thank you for your assistance.

closed 2023-11-13 by Nimaro76