realsenseai/librealsense #12375: Realsense d435i IMU  orientation is fluctuating, is that normal?

ok-kewei, 2023-11-08:
Hi, 
Here is the required Info: 
| Camera Model | d435i |
| Firmware Version | 5.14.0 |
| Operating System & Version | Linux (Ubuntu 20) |
| Language | ROS} |
----------------------------------------------------------------------------------------------------

### Issue Description
I would like to get the orientation of the camera, where it is mounted on a static place (no motion).  here was the step i did. 
1. roslaunch realsense2_camera rs_camera.launch enable_gyro:=true enable_accel:=true unite_imu_method:="linear_interpolation" 
2. roslaunch imu_filter_madgwick imu_filter_madgwick_kw.launch 
3. rosrun imu_filter_madgwick quaternion_to_euler.py 

[image]
[image]

Here is the simple code of my imu_filter_madgwick_kw.launch 

```
<launch>
  <node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_madgwick" output="screen" respawn="false" >
          <param name="fixed_frame" value="base_footprint" />
          <param name="use_mag" value="false" />
          <!-- <param name="publish_tf" value="false" /> -->
          <param name="publish_tf" value="true" />
          <param name="use_magnetic_field_msg" value="true" />
          <param name="world_frame" value="enu" />
          <param name="orientation_stddev" value="0.05" />
          <remap from="/imu/data_raw" to="/camera/imu"/>
          <!-- <remap from="/imu/data" to="/imu/data"/> -->
      </node>

</launch>
```
So my question is the IMU reading is fluctuating, esp in Yaw axis, is that normal ? is the data reliable? 

Thank you for your help :)

MartyG-RealSense, 2023-11-08:
Hi @ok-kewei  Raw IMU data has inherently noisy characteristics.  You could try setting unite_imu_method to **copy** instead of linear_interpolation, as the copy mode should provide more stable IMU frames.

MartyG-RealSense, 2023-11-19:
Case closed due to no further comments received.

closed 2023-11-19 by MartyG-RealSense

ok-kewei, 2023-11-20:
> Hi @ok-kewei Do you require further assistance with this case, please? Thanks!

Sorry for the late reply. Yes, by setting unite_imu_method to copy looks much more stable than the linear interpolation. You may close the case. Thank you so much @MartyG-RealSense  :)

ok-kewei, 2023-11-20:
> Hi @ok-kewei Do you require further assistance with this case, please? Thanks!

Sorry for the late reply as I was away for holiday. I tried by changing the setting of unite_imu_method to **copy** . Unfortunately, the readings of Yaw keeps increasing? Here is the result i got in 4 mins apart where the camera had been putting on the table in static :- 

[image]

[image]

Appreciate it if you can help to address this issue. Thank you. @MartyG-RealSense

ok-kewei, 2023-11-21:
Got it. Thanks for your clarification. @MartyG-RealSense  <3