realsenseai/librealsense #12377: Not able to pyrealsense2. "Module not found"

ta-jetson, 2023-11-08:
@MartyG-RealSense 
----------------------------------------------------------------------------------------------------

| Required Info
|---------------------------------|------------------------------------------- |
| Camera Model - D435i
| Firmware Version - 05.15.01.00
| Operating System & Version - Ubuntu 18| 
| Release- 4.9.201 - tegra
| Jetpack - 4.5.1
| OpenCV - 3.4.3 with CUDA : YES                                   
| Platform - Nvidia Jetson (Seed studio cariier)
| SDK Version- 2.48.0
| Realsense-ros - 2.3.1

### Issue Description
I built librealsense according to the issue "RealSense SDK & pyrealsense2 Installation on Nvidia Jetson Modules #10891" . I used selim35's bash file.
In the DPYTHON_EXECUTABLE, i have used /usr/bin/python and /usr/bin/python3 (as well as python2.7 and python3.6). The build was completed successfully every time. I even got the .so files for pyrealsense2 as well.
I still can't import pyrealsense2 in both approaches. I tried 
import pyrealsense2 as rs
as well as
import pyrealsense2.pyrealsense2 as rs
Both give the "Module not found error"

and 
I noticed
"/usr/local/lib/python2.7/" or "/usr/local/lib/python3.6/"  doesn't have pyrealsense2

Could you please suggest how I could proceed?

NOTE - I used libsense 2.48.0 and JP- 4.5.1 as in one of the issues this was suggested as a fix for the imu data which isn't received in the lastest version. (Tested)

I had another question
I plan on using the IR cameras through the ros wrapper for VIO and pyrealsense2 to access the rgb camera for detection. Would it be possible to use it simultaneously ?

MartyG-RealSense, 2023-11-08:
Hi @ta-jetson  If you have the .so files then an alternative to using a Python path to connect Python and pyrealsense2 together is to place the .so files **librealsense2.so** and **pyrealsense2.so** in the same folder that your project script is in.

If you are using Python 2.7 then place **pyrealsense2.so** in the project folder.

If using Python 3.6 then the pyrealsense2.so file that should be placed in the project folder is the one that will have a more complex auto-generated filename, such as **pyrealsense2.cpython-36m-arm-linux-gnueabihf.so** (one that starts with 'pyrealsense2' and ends with '.so').

The number in the middle of the filename, such as '36m', refers to the Python 3 version that you have got installed on your computer.  36m stands for Python 3.6.

******

It is also worth mentioning that the recommended camera firmware driver version to use with librealsense 2.48.0 is **5.12.14.50**.  Using a newer firmware such as 5.15.1.0 with an older SDK version may result in errors, as 5.15.1.0 is designed for use with SDK **2.54.2**.

MartyG-RealSense, 2023-11-08:
In regard to using infrared and RGB simultaneously, yes you can as the D435i camera has its own dedicated RGB sensor separate from the IR sensors.

ta-jetson, 2023-11-08:
Thank you @MartyG-RealSense 
Error was resolved.

But I can't run them simultaneously. 
I have a python file which uses the rgb camera. Just viewing for now.
I have launch file in realsense-ros for the infra and imu topics. 
When I run realsense-ros launch file and then the python file,

Traceback (most recent call last):
  File "rs_py.py", line 39, in <module>
    pipeline.start(config)
RuntimeError: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error

When I run the python file, and then realsense-ros file,

[ERROR] [1699446990.264410758]: An exception has been thrown: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
[ERROR] [1699446990.264631492]: Exception: xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error
 08/11 18:06:30,263 ERROR [547656499584] (types.h:310) xioctl(VIDIOC_S_FMT) failed Last Error: Input/output error

MartyG-RealSense, 2023-11-08:
The librealsense SDK operates under a set of rules called **Multi-Streaming Model** (see the link below) that determines how access to the sensors by more than one application is governed.

https://github.com/IntelRealSense/librealsense/blob/master/doc/rs400_support.md#multi-streaming-model

The first program to access a particular stream type places a 'claim' on it and prevents another application from accessing that stream until the claim is released by the first program (such as by disabling streaming).  So if realsense-ros enables a stream first, a Python script that is subsequently launched would be unable to use that stream.  

The reverse is also true if the Python script was run first and realsense-ros second - the Python script could use the stream but realsense-ros couldn't.

*********

As you are only using RGB in the Python script, let's make sure that realsense-ros is not also publishing the RGB stream and locking its access.  Please try adding **enable_color:=false** to your launch instruction.  If enable_color is true inside the launch file then the launch instruction command will override it and set it to false.

ta-jetson, 2023-11-09:
Hey @MartyG-RealSense ,
All of your advice has worked flawlessly.
Thank for your help.

Just a review on what I did to resolve (with the help of your advice) :
Used : to build my SDK with pyrealsense2 according to  " https://github.com/IntelRealSense/librealsense/issues/10891#issue-1369393054"

The only change being -DPYTHON_EXECUTABLE=/usr/bin/python3.6  

After that having the .so files in the project folder solved the "Module not found error problem".
To make sure I could use it anywhere, I used the following commands in the terminal(credit to https://github.com/35selim/RealSense-Jetson/issues/1#issuecomment-1293894851)

In terminal :
```
sudo mkdir /usr/local/lib/python3.6/pyrealsense2

sudo ln -s ~/librealsense/build/wrappers/python/pybackend2.cpython-36-aarch64-linux-gnu.so /usr/local/lib/python3.6/pyrealsense2/

sudo ln -s ~/librealsense/build/wrappers/python/pyrealsense2.cpython-36-aarch64-linux-gnu.so /usr/local/lib/python3.6/pyrealsense2/
```

And I sourced my bashrc again. (Remember to make the changes in the bashrc according to " https://github.com/IntelRealSense/librealsense/issues/10891#issue-1369393054"

Now, import pyrealsense2 works without the need of adding the .so files in your project folder.

Using both the python and ros wrapper simultaneously issue was resolved by setting enable_color :=false.

Thanks a lot again @MartyG-RealSense 

I'll be closing the issue.

closed 2023-11-09 by ta-jetson

MartyG-RealSense, 2023-11-09:
You are very welcome.  It's great to hear that you were successful.  Thanks so much for sharing the details of your solution!