realsenseai/librealsense #12379: L515

enrfbezo, 2023-11-08:
18:41:13 [Info]  - librealsense version: 2.54.2

18:41:13 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2697 - Framebuffer size changed to 2560 x 1346
18:41:13 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2697 - Window size changed to 2560 x 1346
18:41:13 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2697 - Scale Factor is now 1
18:41:13 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:312 - null pointer passed for argument "device"
18:41:13 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:2700 - Couldn't refresh devices - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."
18:41:13 [Info] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\context.cpp:336 - Found 1 RealSense devices (mask 0xfe)
| | |
|---|---|
|**librealsense**|2.54.2 RELEASE|
|**OS**|Windows|

Hi All,
I have two brand new RealSense L515 devices one of them returns the above error in realsense viewer2.54.2 on windows 10. The USB cable is the one supplied I swapped them between the devices but the same erro occurs.
The L515 is however listed correctly in windows but as I cannot access it i'm unable to update its firmware.
Do you have anny tips / advices on how to get it started please?
Best regards,
Fred

MartyG-RealSense, 2023-11-09:
Hi @enrfbezo  If you have installed the full RealSense SDK for Windows 10 using the installer program **Intel.RealSense.SDK-WIN10** on the SDK's Releases page, could you right-click on the RealSense Viewer launch shortcut icon on the Windows desktop and select the drop-down menu option **Open file location** please?

This will open the SDK's **Tools** folder.  Please double-leftclick on a file called 'rs-capture' in the Tools folder whilst the camera is plugged in to launch the program.  If rs-capture runs successfully then it will suggest that the camera's problem is related to the RealSense Viewer specifically.

If the camera does not work with rs-capture then unplug the micro-sized USB connector from the base of the camera, turn it around the opposite direction and plug it back in (USB-C cables are two-way insertion at the micro-sized end).  This will confirm whether or not the problem is related to the direction that the connector was inserted in.

MartyG-RealSense, 2023-11-10:
If the affected new camera's firmware has not yet been updated then it probably contains the **1.5.5.0** L515 firmware by default.  So let's try the camera with an older RealSense Viewer version (2.45.0) that is compatible with firmware 1.5.5.0.

Please visit the link below and download from the 'Assets' file list **Intel.RealSense.Viewer.exe**

https://github.com/IntelRealSense/librealsense/releases/tag/v2.45.0

[image]

After the file has downloaded, run it to launch the old 2.54.0 version of the Viewer (do not use the Viewer launch shortcut on the desktop to run it).  Is the L515 now detected in the Viewer?

If it is detected then you can try updating its firmware in the Viewer to the latest **1.5.8.1**.

MartyG-RealSense, 2023-11-16:
Hi @enrfbezo  Do you require further assistance with this case, please?  Thanks!

enrfbezo, 2023-11-16:
Hi @MartyG-RealSense  Yes, It partially works as 2.54.0 detects the camera. However, i only receive the streams from RGB and IMU. I do not receive the depth or infrared streams It shows "no frames received". Also i do not have the option popping up (top right) for the firmware update so where can i find the "signed firmware image" file so i can download and update myself to 1.5.8.1 ?
Thanks in advance

MartyG-RealSense, 2023-11-17:
The L515 firmware files have their own page at the link below.

https://dev.intelrealsense.com/docs/firmware-releases-l500

enrfbezo, 2023-11-18:
I have successfully updated the firmware to 1.5.8.1 using the Viewer 2.45.0. 

Here are the current issues:
in **Viewer version 2.45.0:**
i only receive the streams from RGB and IMU. I do not receive the depth or infrared streams It shows "no frames received".

in **Viewer version 2.54.2:**
rs.cpp:2700 - Couldn't refresh devices - hr returned: HResult 0x8007001f: "A device attached to the system is not functioning."

MartyG-RealSense, 2023-11-19:
It seems likely that the L515 with the problems has a hardware fault with its internal USB components.  As it has only been two weeks since the camera was purchased, I would recommend enquiring to the retailer that you purchased it from regarding returning it for refund or replacement.

enrfbezo, 2023-11-22:
@MartyG-RealSense I will try to figure it out with the supplier. 
Thank you.
Best regards,
Fred

MartyG-RealSense, 2023-11-22:
@enrfbezo  Thanks very much for the update.  Good luck!

closed 2023-12-10 by MartyG-RealSense