realsenseai/librealsense #12381: Request for LabVIEW Wrapper Update and D455 Camera Support

MirrorCY, 2023-11-09:
### Required Info
| Required Info                     | Information |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 |
| Operating System & Version | Windows 10 |
| Platform                                 | PC |
| SDK Version                            | 2.33.0|
| Language                            | LabVIEW |
| Segment			| Others |

### Issue Description

The LabVIEW wrapper for Intel RealSense has not been updated for three and a half years, and it does not support the D455 camera. We kindly request an update for the LabVIEW wrapper to include support for the D455 camera.

#7920 #10064

MartyG-RealSense, 2023-11-09:
Hi @MirrorCY  Some RealSense users have been able to use the LabVIEW wrapper with RealSense camera models released after SDK 2.33.0 if the SDK version used with the wrapper is updated.  D455 support was introduced in SDK 2.38.1.

An example of this was a D405 user at https://github.com/IntelRealSense/librealsense/issues/12093 who was able to use the wrapper with SDK version 2.53.1 (official D405 support was first introduced in 2.51.1).

MirrorCY, 2023-11-13:
Thank you for your response.
I have successfully compiled the 32-bit SDK DLL and can run it in LabVIEW. However, I encountered the same issue as described in #10064. When capturing both depth and left IR streams simultaneously, the program freezes, and LabVIEW crashes. I have tried the suggested workaround discussed in that thread, but the 05.12.14.50 firmware does not support the D455 camera. I'm looking forward to receiving some suggestions for a solution.

The diagram below shows a portion of routines that are functioning correctly.
[image]

MirrorCY, 2023-11-17:
Thank you for your reply. I have now switched to using Python to implement the functionality I needed. Thanks again.

closed 2023-11-17 by MirrorCY

MartyG-RealSense, 2023-11-17:
You are very welcome.  Thanks very much for the update.