realsenseai/librealsense #12382: profile = pipe.start(cfg) RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy terminate called without an active exception

Anthonia2001, 2023-11-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                       
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400series/D435 } | 
| Firmware Version                   | (5.15.1) | 
| Operating System & Version |   {Ubuntu 20.04 LTS| 
| Kernel Version (Linux Only)    | 5.15.0-88-generic | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.54.2 }  | 
| Language                            |{python 3.8.10}   | 
| Segment			      |  {Robot/Smartphone/VR/AR/others}| 

### Issue Description
<Describe your issue / question / feature request / etc..>

When I run the source code predict.py of the github above with source 4 is the Intel Realsense camera.
https://github.com/ultralytics/yolov5/blob/master/segment/predict.py
The program run **normally** without any **errors.**

But after I **imported pyrealsense2** and adding 
```
def active_camera():
    cfg = rs.config() #ok
    cfg.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 60) #ok
    cfg.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 60) #ok

    pipe = rs.pipeline()
    profile = pipe.start(cfg)

    #pipe.start(cfg)
    align_to = rs.stream.color
    align = rs.align(align_to)
    return pipe, align

@smart_inference_mode()
def run(
    weights=ROOT / 'yolov5s-seg.pt',  # model.pt path(s)
    source=ROOT / 'data/images',  # file/dir/URL/glob/screen/0(webcam)
    data=ROOT / 'data/coco128.yaml',  # dataset.yaml path
    imgsz=(640, 640),  # inference size (height, width)
    conf_thres=0.25,  # confidence threshold
    iou_thres=0.45,  # NMS IOU threshold
    max_det=1000,  # maximum detections per image
    device='',  # cuda device, i.e. 0 or 0,1,2,3 or cpu
    view_img=False,  # show results
    save_txt=False,  # save results to *.txt
    save_conf=False,  # save confidences in --save-txt labels
    save_crop=False,  # save cropped prediction boxes
    nosave=False,  # do not save images/videos
    classes=None,  # filter by class: --class 0, or --class 0 2 3
    agnostic_nms=False,  # class-agnostic NMS
    augment=False,  # augmented inference
    visualize=False,  # visualize features
    update=False,  # update all models
    project=ROOT / 'runs/predict-seg',  # save results to project/name
    name='exp',  # save results to project/name
    exist_ok=False,  # existing project/name ok, do not increment
    line_thickness=3,  # bounding box thickness (pixels)
    hide_labels=False,  # hide labels
    hide_conf=False,  # hide confidences
    half=False,  # use FP16 half-precision inference
    dnn=False,  # use OpenCV DNN for ONNX inference
    vid_stride=1,  # video frame-rate stride
    retina_masks=False,
):
    source = str(source)
    save_img = not nosave and not source.endswith('.txt')  # save inference images
    is_file = Path(source).suffix[1:] in (IMG_FORMATS + VID_FORMATS)
    is_url = source.lower().startswith(('rtsp://', 'rtmp://', 'http://', 'https://'))
    webcam = source.isnumeric() or source.endswith('.streams') or (is_url and not is_file)
    screenshot = source.lower().startswith('screen')
    if is_url and is_file:
        source = check_file(source)  # download

    # Directories
    save_dir = increment_path(Path(project) / name, exist_ok=exist_ok)  # increment run
    (save_dir / 'labels' if save_txt else save_dir).mkdir(parents=True, exist_ok=True)  # make dir

    # Load model
    device = select_device(device)
    model = DetectMultiBackend(weights, device=device, dnn=dnn, data=data, fp16=half)
    stride, names, pt = model.stride, model.names, model.pt
    imgsz = check_img_size(imgsz, s=stride)  # check image size

    # Dataloader
    bs = 1  # batch_size
    if webcam:
        view_img = check_imshow(warn=True)
        dataset = LoadStreams(source, img_size=imgsz, stride=stride, auto=pt, vid_stride=vid_stride)
        pipe , align = active_camera()
        num = threading.active_count()
        print(num)
        bs = len(dataset)
    elif screenshot:
        dataset = LoadScreenshots(source, img_size=imgsz, stride=stride, auto=pt)
    else:
        dataset = LoadImages(source, img_size=imgsz, stride=stride, auto=pt, vid_stride=vid_stride)
    vid_path, vid_writer = [None] * bs, [None] * bs
```
The compiler gave this error:
```
 File "predict.py", line 63, in active_camera
    profile = pipe.start(cfg)
RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
terminate called without an active exception
```
I do not know why the Device is busy because I do not open any other app to run camera except this code.
This problem just happens when I run on **Ubuntu 20.04**; when I execute on **Windows**, the program works normally.

MartyG-RealSense, 2023-11-09:
You should be able to install pyrealsense2 from packages with the **pip install pyrealsense2** instruction even if you install librealsense from source code first.

If you have a desktop / laptop PC and a Python version between 3.7 and 3.11 then you can pip install. 

If your computer is an Arm-based device such as Nvidia Jetson or Raspberry Pi then you can pip install if you have Python 3.7 to 3.9, otherwise you will need to build pyrealsense2 from source code to use Python versions newer than 3.9 at the present time.

Anthonia2001, 2023-11-09:
> You should be able to install pyrealsense2 from packages with the **pip install pyrealsense2** instruction even if you install librealsense from source code first.
>
> If you have a desktop / laptop PC [...]

Thanks for your support. I will follow your instruction. Just one more question that I am using Python 3.8.10 which is distributed with ROS Noetic on Ubuntu, Does it have any affect to this problem?

MartyG-RealSense, 2023-11-09:
The minor number of your Python version (.10) should not matter so long as the major number is 3.8.

Anthonia2001, 2023-11-10:
> The minor number of your Python version (.10) should not matter so long as the major number is 3.8.

After doing your instruction, I followed this link to install librealsense by source: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#make-ubuntu-up-to-date

And then, I run the code and this give me the new error:
```
 profile = pipe.start(cfg)
RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=5 Last Error: Input/output error
terminate called without an active exception
Aborted (core dumped)
```
I've searched this new error but I cannot find anything

MartyG-RealSense, 2023-11-10:
What happens if you try running the simple test script in the Python wrapper documentation, please?

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

Anthonia2001, 2023-11-10:
> What happens if you try running the simple test script in the Python wrapper documentation, please?
> 
> https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#streaming-using-rspipeline

Hi, thanks for your reply. After running the simple scripts. I got this result:
```
 if y%20 is 19:
       :.          ...    :   ...BWWXn  :    hnWh               
    .            .           :hBBXXBXh    :.nX.:.   .:          
        .          .n  .   :hWXXhWWWWX    hBXh..:BWWWX:         
                  .         .nhXXXWXWWX   hBX.   nBXBn  :BBh    
         n            .    .::h:hXXWXBBBX.nBBXhhhBXX:::hnn:     
             .          n   : hBXXWWWWWWWWWX:.BWWXXWWXhXh:n  :hn
      nhX.                   nXWWXWWXXBWWXXWXXWWWWWXXWXWWWXWWBhh
                       nhn nWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWhn
       n       :.n  :nnBWXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWXW
      XXhhBXBBXXXXXnXXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWBW
     :WWXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWBX
     nWWWWWW:hXWXXWWXXWWXWWWWWXXWXXXWWXBBXXBBXXhXBXXXBBXXXXWWXBX
    .hBhXXWWXXWWWWWWWWWWXWWXWWWWWXWWXXWWhnhB::hWWXWWWWWWWWXWWWWB
    hWWWWWXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWB.   BWWWWWWWWWWWWWWWWX
     :.::nhWWWXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWB:WWWWWWWWWWWWWXXWWW
     .hhWWWhhhhnBXWXWWWXhWWWWWWWWWWWWWXWWWWWWWWWWWWWWWWWWWWWXWWW
      .nBhn:n: nhh:hnhBXXXXWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWWXWWX
       :B:             ::BXXXWWWWWWWWWWWWWWWWWWWWWWWWWWWWXWWXXXh
        .          .      . :BBBXXWWWWWWWWWWWWWWWWWWWWWWWWWXXnB.
```
[image]

MartyG-RealSense, 2023-11-11:
This case can probably be closed as we are discussing your busy pipeline problem with predict.py in your other current case at https://github.com/IntelRealSense/librealsense/issues/12391

closed 2023-11-23 by MartyG-RealSense