realsenseai/librealsense #12383: semantic segmentation with YOLOv8

sanjaiiv04, 2023-11-09:
I am trying to do semantic segmentation with Intel RealSense. I am new to Intel RealSense and so I took an example script from the examples that retrieve the rgb and depth values. Now I am trying to get the rgb frames and then send it to the YOLOv8 segmentation model that then does inference on that. Then I extract the mask coordinates from the inference results and use cv2.fillPoly() to fill the mask coordinates on the rgb image. This is the code: 

from ultralytics import YOLO

import pyrealsense2 as rs
import numpy as np
import cv2

model=YOLO("yolov8s-seg.pt")
# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

#  device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)

try:
    while True:

        # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()
        if not depth_frame or not color_frame:
            continue

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        results=model.predict(source=color_image,stream=True)
        for person in results:
            mask=person.masks.xy
            mask=np.round(mask).astype(int)
            cv2.fillPoly(color_image,[mask],color=(0,0,255))

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.05), cv2.COLORMAP_JET)

        depth_colormap_dim = depth_colormap.shape
        color_colormap_dim = color_image.shape

        # If depth and color resolutions are different, resize color image to match depth image for display
        if depth_colormap_dim != color_colormap_dim:
            resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
            images = np.hstack((resized_color_image, depth_colormap))
        else:
            images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)

finally:

    # Stop streaming
    pipeline.stop()
But I get the following error: 

0: 480x640 1 person, 170.5ms
Speed: 2.9ms preprocess, 170.5ms inference, 2.1ms postprocess per image at shape (1, 3, 480, 640)

0: 480x640 1 person, 198.6ms
Speed: 1.3ms preprocess, 198.6ms inference, 1.5ms postprocess per image at shape (1, 3, 480, 640)

0: 480x640 1 person, 162.6ms
Speed: 4.2ms preprocess, 162.6ms inference, 1.8ms postprocess per image at shape (1, 3, 480, 640)

0: 480x640 1 person, 165.5ms
Speed: 1.4ms preprocess, 165.5ms inference, 1.6ms postprocess per image at shape (1, 3, 480, 640)

Traceback (most recent call last):
  File "/home/kukahtic/Sanjaii/yolov8-test.py", line 54, in <module>
    mask=np.round(mask).astype(int)
  File "<__array_function__ internals>", line 200, in round_
  File "/home/kukahtic/.local/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 3763, in round_
    return around(a, decimals=decimals, out=out)
  File "<__array_function__ internals>", line 200, in around
  File "/home/kukahtic/.local/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 3337, in around
    return _wrapfunc(a, 'round', decimals=decimals, out=out)
  File "/home/kukahtic/.local/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 54, in _wrapfunc
    return _wrapit(obj, method, *args, **kwds)
  File "/home/kukahtic/.local/lib/python3.8/site-packages/numpy/core/fromnumeric.py", line 43, in _wrapit
    result = getattr(asarray(obj), method)(*args, **kwds)
ValueError: setting an array element with a sequence. The requested array has an inhomogeneous shape after 1 dimensions. The detected shape was (2,) + inhomogeneous part.
Exception ignored in: <generator object BasePredictor.stream_inference at 0x7fb16877c5f0>
Traceback (most recent call last):
  File "/home/kukahtic/.local/lib/python3.8/site-packages/torch/utils/_contextlib.py", line 44, in generator_context
  File "/home/kukahtic/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 270, in clone
  File "/home/kukahtic/.local/lib/python3.8/site-packages/torch/autograd/grad_mode.py", line 251, in __init__
AttributeError: 'NoneType' object has no attribute 'is_scripting'

How to solve this error?

MartyG-RealSense, 2023-11-09:
Hi @sanjaiiv04  Whilst there is unfortunately not reference material available regarding how to solve the problem described above, an alternative approach that might provide useful insights is the **Image Capture With RealSense** Python project at https://github.com/Incalos/Image-Capture-With-RealSense which can be used to create a custom dataset for semantic segmentation.

If you prefer to continue with your own method above then I would recommend focusing on debugging the first error that appears in the log - **mask=np.round(mask).astype(int)** - as subsequent errors after that may only be occurring because of the first error (i.e they result from a 'cascade failure' caused by the first error) and may automatically disappear if the first error is solved.

sanjaiiv04, 2023-11-10:
@MartyG-RealSense Hey as you said, the mask was the problem. Instead of using **mask=np.round(mask).astype(int)** I changed the script to: 

**act_mask = mask[i].tolist()
mask_contours = [np.array(act_mask, dtype=np.int32)]**. Then I used the mask coordinates in the mask_contours variable to do **cv2.fillPoly()**. Now it works!

sanjaiiv04, 2023-11-10:
I am trying to get the mask image from the YOLOv8 model and then overlay the mask image on the depth_colormap to crop the depth_map. But it does not work due to issues in shape and type of the two arrays when trying to do bitwise_and or cv2.addWeighted(). This is the following code: 

from ultralytics import YOLO
import pyrealsense2 as rs
import numpy as np
import cv2

#Loading the yolov8 segmentation model
model=YOLO("yolov8s-seg.pt")

# Configure depth and color streams
pipeline = rs.pipeline()
config = rs.config()

#  device product line for setting a supporting resolution
pipeline_wrapper = rs.pipeline_wrapper(pipeline)
pipeline_profile = config.resolve(pipeline_wrapper)
device = pipeline_profile.get_device()
device_product_line = str(device.get_info(rs.camera_info.product_line))

found_rgb = False
for s in device.sensors:
    if s.get_info(rs.camera_info.name) == 'RGB Camera':
        found_rgb = True
        break
if not found_rgb:
    print("The demo requires Depth camera with Color sensor")
    exit(0)

config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

if device_product_line == 'L500':
    config.enable_stream(rs.stream.color, 960, 540, rs.format.bgr8, 30)
else:
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

# Start streaming
pipeline.start(config)
try:
    while True:

    # Wait for a coherent pair of frames: depth and color
        frames = pipeline.wait_for_frames()
        depth_frame = frames.get_depth_frame()
        color_frame = frames.get_color_frame()

        # Convert images to numpy arrays
        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())
        mask_image = np.zeros_like(depth_image, dtype=np.uint8)

        #Segmentation
        results=model.predict(source=color_image,stream=True)
        for person in results:
            mask=person.masks.xy #getting mask cordinates for each object detected in each frame
            #print(mask)
            for i in range(len(mask)): #iterating through each mask 
                
                act_mask = mask[i].tolist()  # Assuming the mask data is a list of coordinates
                
                # Convert the mask data into a list of contours
                mask_contours = [np.array(act_mask, dtype=np.int32)]
                #print(mask_contours[0].shape)

                #Creating an empty frame with the same size as depth image and filling it with the mask cordinates
                cv2.fillPoly(mask_image, mask_contours, color=255)
                cv2.fillPoly(color_image, mask_contours, color=(0, 0, 255))

        # Apply colormap on depth image (image must be converted to 8-bit per pixel first)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.05), cv2.COLORMAP_JET)
        print(mask_image.shape)
        print(depth_colormap.shape[:2])
        print(mask_image.dtype)
        print(depth_colormap.dtype)
        #added_image = cv2.addWeighted(mask_image,0.5,depth_colormap[:2],0.1,0)
        depth_colormap_dim = depth_colormap.shape
        color_colormap_dim = color_image.shape

        # If depth and color resolutions are different, resize color image to match depth image for display
        if depth_colormap_dim != color_colormap_dim:
            resized_color_image = cv2.resize(color_image, dsize=(depth_colormap_dim[1], depth_colormap_dim[0]), interpolation=cv2.INTER_AREA)
            images = np.hstack((resized_color_image, depth_colormap))
        else:
            images = np.hstack((color_image, depth_colormap))

        # Show images
        cv2.imshow("",mask_image)
        cv2.namedWindow('RealSense', cv2.WINDOW_AUTOSIZE)
        cv2.imshow('RealSense', images)
        cv2.waitKey(1)
        if cv2.waitKey(10) & 0xFF=='q':
            cv2.destroyAllWindows()

finally:

    # Stop streaming
    pipeline.stop()

MartyG-RealSense, 2023-11-10:
I wonder if the **bitwise_and** YOLOv8 example at https://github.com/ultralytics/ultralytics/issues/1407#issuecomment-1616287320 is similar to what you are trying to achieve.

closed 2023-11-23 by MartyG-RealSense