realsenseai/librealsense #12385: D435 detected as USB 2.1, failing after ~3.5 days

stwirth, 2023-11-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.13.00.50 | 
| Operating System & Version |   Ubuntu 20.04.1  | 
| Kernel Version (Linux Only)    |  5.15.0-88-generic        | 
| Platform                                 | PC |
| SDK Version                            |  v2.50.0                           | 
| Language                            |  C++                          | 
| Segment			|  Robot       | 

### Issue Description

On one of our robots a camera failed after running for ~3.5 days. The kernel reported:
```
Nov 03 14:03:15 robot kernel: usb 4-2.3: reset full-speed USB device number 3 using xhci_hcd
Nov 03 14:03:16 robot kernel: usb 4-2.4: Process 3985813 (nodelet) called USBDEVFS_CLEAR_HALT for active endpoint 0x82
```
Furthermore, the process said:
```
Nov 03 14:03:16 robot ros[2910]:  03/11 14:03:16,915 ERROR [139736383465216] (uvc-streamer.cpp:106) uvc streamer watchdog triggered on endpoint: 130
Nov 03 14:03:16 robot ros[2910]:  03/11 14:03:16,918 WARNING [139736383465216] (messenger-libusb.cpp:30) reset_endpoint returned error, index: 130, error: Protocol error, number: 71
```

Looking at logs from the startup of the camera ~3.5 days before, I found that it was detected as a USB 2.1 device:
```
Oct 30 20:27:58 robot ros[2910]: [ WARN] [/camera/nodelet_manager] [1698722878.859793010]: Device 231122072215 is connected using a 2.1 port. Reduced performance is expected.
...
Oct 30 20:28:07 robot ros[2910]: [ INFO] [/camera/nodelet_manager] [1698722878.859532944]: Device with physical ID 4-2.4-5 was found.
Oct 30 20:28:07 robot ros[2910]: [ INFO] [/camera/nodelet_manager] [1698722878.859539597]: Device with name Intel RealSense D435 was found.
Oct 30 20:28:07 robot ros[2910]: [ INFO] [/camera/nodelet_manager] [1698722878.859770867]: Device with port number 4-2.4 was found.
Oct 30 20:28:07 robot ros[2910]: [ INFO] [/camera/nodelet_manager] [1698722878.859781969]: Device USB type: 2.1
```
After a restart, the camera was detected as a USB 3.2 and works fine since then.

My question is: what can make the camera be detected (intermittently) as USB 2.1 device? How can this be avoided?

MartyG-RealSense, 2023-11-09:
Hi @stwirth  RealSense 400 Series cameras can run indefinitely as long as they remain within recommended operating temperature tolerances.  If the camera operates for days whilst remaining within recommended temperatures and then experiences a problem, the cause is typically not with the camera hardware but with a glitch in a non-camera component such as the computer's USB port or its operating system.  Your log suggests that it is a USB related issue in your particular case, especially as the camera performs normally again after a restart.

One possible cause of the camera restarting and then being mis-detected as USB 2.1 is if the USB cable is moved / collided with in a way that causes the USB connector to shift position within the USB port.

MartyG-RealSense, 2023-11-14:
Hi @stwirth  Do you require further assistance with this case, please?  Thanks!

stwirth, 2023-11-14:
Thanks @MartyG-RealSense. We swapped the camera and the cable and it has been working fine since then. A bit unsatisfactory to not know exactly the reason but I don't think you can do anything about that :)

closed 2023-11-14 by stwirth

MartyG-RealSense, 2023-11-14:
Yes, sometimes you have to be satisfied with a good outcome without knowing what the cause of the problem was.  :)  Thanks very much for the update!