realsenseai/librealsense #12387: Pipeline async vs synchronous.

GrubbyHalo, 2023-11-09:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400  | 
| Firmware Version                   | N/A | 
| Operating System & Version |  not really applicable | 
| Kernel Version (Linux Only)    |                                           | 
| Platform                                 | Any  |
| SDK Version                            |  2.54                           | 
| Language                            |  {C/C++ }                          | 

### Issue Description
What are the pro's/cons of receiving frames via a callback or from polling or waiting on the frames besides being a convenience for the developer based on how they structure their code. Why are there two mechanisms to retrieve frames when either a polling/blocking call or alternatively  a callback could have been exclusively provided to do job ? 

In the rs-callback example the only reason given for using a callback is if the user has to deal with high frequency data such as an IMU. Is that it ? Is that the only reason ?

Also, I noticed that when stalling within a pipeline callback for a long time I get an app that eventually hangs. This makes sense after examining the source code in ```playback_device.cpp``` and ```playback_sensor.cpp``` for bag file sourced frames. So there seems to be some precautions that need be exercised when using the callback however using the synchronous API it seems that frames are dropped if not handled by user code .

Thoughts please.

MartyG-RealSense, 2023-11-10:
Hi @GrubbyHalo  Documentation about the synchronous and asynchronous methods can be found at the documentation link below.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0

https://github.com/IntelRealSense/librealsense/issues/3817#issuecomment-485663496 is a very good reference for further reading, where a RealSense team member discusses synchronous and asynchronous operation.

https://github.com/IntelRealSense/librealsense/issues/2662#issuecomment-442589025 has more information about the syncer class.

My understanding is that synchronous and asynchronous provide similar performance and so there is no significant advantage in choosing one approach over the other.

A common reason for using a callback is if you need to stream depth, color and IMU data simultaneously, otherwise depth+color or IMU on its own may work but not all three stream types together.

GrubbyHalo, 2023-11-10:
@MartyG-RealSense thanks for the feedback.

closed 2023-11-10 by GrubbyHalo