realsenseai/librealsense #12388: Occasional lag between RGB and D data in high resolutions / low frame rates

jev-aleks, 2023-11-09:
| Required Info                         |                                                                |
|-------------------------------|-------------------------------------------|
| Camera Model                       | D457 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version  |  Linux (Ubuntu 22.04.3)  | 
| Kernel Version (Linux Only)     |  5.15.0   | 
| Platform                                 | PC  |
| SDK Version                           |  2.54.2  | 
| Language                              |  Python  | 
| Segment			       |  Robot  | 

### Issue Description
I have run into the following issue. 
In some images, there seems to be a lag between the recorded RGB image and depth image. I have tried out different combinations of settings. The issue seems to happen mostly with _higher resolutions_ and _lower frame rates_. When something is in movement, this leads to misaligned RGB and depth data. 

To record, I use the function `rs.pipeline.wait_for_frames()`. This should already provide time-synchronized frames. Yet I still observe this misalignment in some of the images. What could be the issue?

MartyG-RealSense, 2023-11-10:
Hi @jev-aleks  If you are using a low FPS then you may get less lag in the RGB image if you use **5** FPS and **78** manual RGB exposure instead of auto-exposure.

When setting RGB exposure in Python code, you should tell the script to access the RGB color sensor specifically or it will default to using the depth sensor.  You can do this by defining a **color_sensor** and either setting it to an index number of '1' (the index for depth is '0') or by using the **first_color_sensor** instruction.

MartyG-RealSense, 2023-11-16:
Hi @jev-aleks  Do you require further assistance with this case, please?  Thanks!

jev-aleks, 2023-11-16:
Thank you for your response!
Does that mean that I have to switch to manual exposure to not get any lag? Why does it happen only with low FPS?

The problem is that we are working in dynamic lighting conditions and rely on the auto exposure to get good images. Is there a solution there?

MartyG-RealSense, 2023-11-16:
With very low frame rates such as 5 or 6, there is a chance that the camera connection will 'time out' whilst waiting for new frames to arrive because of the slowness of frame delivery.  15 FPS should reduce the risk of problems occurring.

If you need to use auto-exposure then having auto-exposure enabled and an RGB option called **Auto-Exposure Priority** disabled can force the SDK to try to maintain the depth and color streams at a constant FPS instead of permitting FPS to vary.  A Python script at https://github.com/IntelRealSense/librealsense/issues/11246#issue-1506464871 has an example of code for doing so.

`color_sensor.set_option(rs.option.auto_exposure_priority, 0)`

jev-aleks, 2023-11-16:
Thank you, we will try this!