realsenseai/librealsense #12389: Is it possible to check for cracks in walls using the L515?

Delircan, 2023-11-10:
I am considering using the L515 for a robot project, but I have concerns about its ability to detect damages on a wall. Can you recommend this product for finding cracks, even those as small as millimeters?

MartyG-RealSense, 2023-11-10:
Hi @Delircan  I ran tests on a hairline wall crack with L515 and other RealSense camera models.  Such a crack is too thin to be rendered visibly on a depth image though, unfortunately.

Delircan, 2023-11-10:
You're welcome, and thank you for your response. Do you think it would be useful for cracks that are in centimeters in size?

MartyG-RealSense, 2023-11-10:
Centimeters are able to be captured.  I tested with an L515 and D435f on a radiator grille to simulate cracks.  The L515 provided the clearest results.  

**L515**

[image]

**D435f**

[image]

The L515 model is now retired though, but remains available in the official RealSense Store whilst stocks last.

https://store.intelrealsense.com/buy-intel-realsense-lidar-camera-l515.html

The listing for D435f is here:

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d435f.html

MartyG-RealSense, 2023-11-16:
Hi @Delircan  Do you require further assistance with this case, please?  Thanks!

Delircan, 2023-11-16:
Thank you for the answer . We will use this product on an inspection robot and I am comparing it with other "slamtec Mapping Sensor LiDAR" do you have any ideas? Our robot to be used in concrete pipelines.

MartyG-RealSense, 2023-11-16:
There was a past article on the official RealSense blog site about using RealSense cameras for inspection of materials such as concrete.

https://www.intelrealsense.com/the-future-of-construction/

The article provided a link to a paper called "Using a 3D Computer Vision System for Inspection of Reinforced Concrete Structures" and stated " The authors propose using Intel RealSense D435 depth cameras to create an alternative to manual visual inspection.  The cameras were mounted on a robot, allowing for inspection not only of areas that were already accessible to a human engineer, but also allowing for inspection of inaccessible areas such as crawl spaces or other spaces difficult for a person to access safely". 

https://rd.springer.com/chapter/10.1007/978-3-030-17798-0_49

closed 2023-11-23 by Delircan