realsenseai/librealsense #12390: ROS2 Spinnaker SDK disconnects Realsense

dariodenardi, 2023-11-10:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.8.1 | 
| Operating System & Version |   Ubuntu 22.04.03  | 
| Kernel Version (Linux Only)    |  6.2.0-36-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.2                           | 
| Language                            |  ROS2                          | 
| Segment			|  Robot                          | 

### Issue Description
I've a similar problem to: https://github.com/IntelRealSense/realsense-ros/issues/2569 but I'm using a different version of ROS and different devices.

When I installed the SDK and compiled the ROS2 Intel RealSense workspace, I encountered the following error:

[realsense2_camera_node-1]  10/11 10:43:59,863 ERROR [140496559138368] (backend-v4l2.cpp:2771) xioctl(VIDIOC_QBUF) failed when requesting new frame! fd: 29 error: No such device

I attempted to resolve this issue by referring to https://github.com/IntelRealSense/realsense-ros/issues/2569, but unfortunately, I faced a similar problem:

[realsense2_camera_node-1]  10/11 11:36:51,564 WARNING [139872469845568] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: Device or resource busy, number: 16

Additional information: I have installed the ROS2-Beta Wrapper for Intel® RealSense™ Devices (build 4.51.1).

MartyG-RealSense, 2023-11-10:
Hi @dariodenardi  At the time of writing this the librealsense SDK does not officially support the 6.2 kernel.  The latest supported is 5.15.  Patch support for kernel 6.2 is currently on the **development** branch of librealsense, which becomes the next master branch when pushed to release.

https://github.com/IntelRealSense/librealsense/tree/development

In the meantime, if librealsense is built from source code with CMake with the **-DFORCE_RSUSB_BACKEND=TRUE** flag included in the CMake build instruction then librealsense bypasses the kernel so it does not matter which kernel your computer is using.  Information about this can be found in the Spinnaker case that you quoted, at https://github.com/IntelRealSense/realsense-ros/issues/2569#issuecomment-1339495126

MartyG-RealSense, 2023-11-16:
Hi @dariodenardi  Do you require further assistance with this case, please?  Thanks!

dariodenardi, 2023-11-16:
Hi, thank you for your response.

I'm currently conducting some tests, and I'll update you once I've finished.

I can tell you that when using kernel 6.2 with the setup described in https://github.com/IntelRealSense/realsense-ros/issues/2569#issuecomment-1339495126, I encounter an error. I have now installed Ubuntu 20.04 with kernel 5.15, and I'm currently in the process of testing it.

MartyG-RealSense, 2023-11-17:
Okay, thanks very much for the update.  I look forward to your next report after testing.  Good luck!

dariodenardi, 2023-11-20:
Hi @MartyG-RealSense,

I've tried on Ubuntu 20.04 and have finally resolved the issue. I'd like to note two things:
- I'm using BlackFly S and have installed Spinnaker SDK 3.0.79
- I followed this guide to install the RealSense SDK: https://github.com/IntelRealSense/realsense-ros/issues/2569#issuecomment-1339495126

An important point to highlight: when launching the nodes, make sure to start the Spinnaker node **first and then** the RealSense node. Otherwise, you'll encounter the same error consistently

MartyG-RealSense, 2023-11-20:
It's great to hear that you achieved a solution.  Thanks so much for sharing the details of it!

closed 2023-11-26 by dariodenardi