realsenseai/librealsense #12391: realsense camera run normally in yolov5s but not yolov5s-seg on Ubuntu 20.04

Anthonia2001, 2023-11-10:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info |
|---------------------------------|------------------------------------------- |
| Camera Model | { D400series/D435 } |
| Firmware Version | (5.15.1) |
| Operating System & Version | {Ubuntu 20.04 LTS|
| Kernel Version (Linux Only) | 5.15.0-88-generic |
| Platform | PC|
| SDK Version | { legacy / 2.54.2 } |
| Language |{python 3.8.10 distributed with ROS Noetic} |
| Segment | {Robot/Smartphone/VR/AR/others}|
### Issue Description

Hi! I'm applying yolov5 to detect object and publish data to my ROS node. 
And I also install **librealsense** from this link: (https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) with  CMake and including the flag **-DFORCE_RSUSB_BACKEND=TRUE in the CMake build instruction**. And **pyrealsense2 with pip install**

At first, when I modified and run this code from yolov5s.pt model
```
import argparse
import os #ok
import sys #ok
from pathlib import Path #ok
import pyrealsense2 as rs #ok
import numpy as np #ok
import cv2 #ok
import torch #ok
import torch.backends.cudnn as cudnn #ok
import time #ok
from tkinter import ttk 
import tkinter as tk
import tkinter.font as font
import math
import threading

import rospy
from std_msgs.msg import Float64MultiArray
from std_msgs.msg import Int16

FILE = Path(__file__).resolve() 
ROOT = FILE.parents[0]  # YOLOv5 root directory
if str(ROOT) not in sys.path:
    sys.path.append(str(ROOT))  # add ROOT to PATH
ROOT = Path(os.path.relpath(ROOT, Path.cwd()))  # relative

from models.common import DetectMultiBackend
from utils.datasets import IMG_FORMATS, VID_FORMATS, LoadImages, LoadStreams
from utils.general import (LOGGER, check_file, check_img_size, check_imshow, check_requirements, colorstr,
                           increment_path, non_max_suppression, print_args, scale_coords, strip_optimizer, xyxy2xywh)
from utils.plots import Annotator, colors, save_one_box
from utils.torch_utils import select_device, time_sync

import threading

cv_font = cv2.FONT_HERSHEY_SIMPLEX

object_coordinates = [] #label, x, y
delete_objects = False

signal_1 = int()

[... 789 lines cut ...]
[ WARN:0@3.658] global cap_v4l.cpp:982 open VIDEOIO(V4L2:/dev/video4): can't open camera by index
[ERROR:0@3.659] global obsensor_uvc_stream_channel.cpp:156 getStreamChannelGroup Camera index out of range
Traceback (most recent call last):
  File "test_seg_2.py", line 515, in <module>
    main(opt)
  File "test_seg_2.py", line 511, in main
    run(**vars(opt))
  File "test_seg_2.py", line 181, in run
    dataset = LoadStreams(source, img_size=imgsz, stride=stride, auto=pt, vid_stride=vid_stride)
  File "/home/anthonia/Thesis/src/robot_project/yolov5-master/utils/dataloaders.py", line 367, in __init__
    assert cap.isOpened(), f'{st}Failed to open {s}'
AssertionError: 1/1: 4... Failed to open 4
```
And with the code above, and move the code to open realsense camera into def get_frame stream() function. The camera is open but when the camera can detect a thing, the program is crashed and gives the errors:

```
profile = pipe.start(cfg)
RuntimeError: xioctl(VIDIOC_S_FMT) failed, errno=16 Last Error: Device or resource busy
terminate called without an active exception
Aborted (core dumped)
```
### Environment
YOLOv5 🚀 2023-10-3 Python-3.8.10 torch-2.1.0+cu118 CUDA:0 (NVIDIA GeForce RTX 3050 Laptop GPU, 3910MiB)
OS: Ubuntu 20.04
Python version: 3.8.10 distributed with ROS Noetic

MartyG-RealSense, 2023-11-11:
Hi @Anthonia2001  Is the modified code from yolov5s.pt model and the modified predict.py two separate scripts?  Both contain a pipe start instruction.  

So if the modified yolov5s.pt script was run first and then the modified predict.py script was run secondly afterwards, the camera would be 'busy' for the predict.py script because the yolov5s.pt script would have already started the camera's pipeline.

Anthonia2001, 2023-11-11:
Hi! Thanks for your reply. yeah, They are two different scripts. The code
from yolov5s.pt model and the code from yolov5s-seg.pt model are in two
different projects.
After I stop execute the script from the yolov5s model, and run the
modified predict.py script from other project. The camera is busy.

MartyG-RealSense, 2023-11-11:
If you reboot your computer and then run predict.py first without running the yolov5s.pt script beforehand and it works, that could indicate that the method by which you stop executing the yolov5s.pt script is not properly closing down the pipeline, preventing it from working properly when predict.py is run.

Anthonia2001, 2023-11-11:
Yeah, actually I have tried what you say before. I shut down my PC and then
start with the predict.py first but it still doesn’t work.
I think that maybe it is some conflicts between rs and yolov5s-seg.pt model
on Ubuntu right?

MartyG-RealSense, 2023-11-11:
You are the first RealSense user to report having used yolov5s-seg so there is no way to be certain.  It would be reasonable to conflude though that if yolov5 works and yolov5s-seg does not then there is an issue with yolov5s-seg specifically.

Anthonia2001, 2023-11-11:
Really appreciate for your help. I have uploaded the issue on the yolov5
github. Hope that I will find a solution.

MartyG-RealSense, 2023-11-11:
You are very welcome.  I'm pleased that I could be of help.  Good luck!

Anthonia2001, 2023-11-11:
Thanks a lot. Have a nice weekend!

closed 2023-11-23 by MartyG-RealSense