realsenseai/librealsense #12393: Issues with starting Intel D435i in USB 2.0 mode

chutsu, 2023-11-11:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 5.13.0.50 | 
| Operating System & Version |  Ubuntu 22.04  | 
| Kernel Version (Linux Only)    | 6.2.0-36-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.54.1.0                       | 
| Language                            |  C++                          | 
| Segment			| Robot                          | 

### Issue Description

I'm trying to use the Intel D435i in **USB 2.1 mode** (specifically not USB 3). While I have created a C++ program that is capable of connecting to the device's Depth, Stereo IR and RGB and IMU streams properly in **USB 3.0**, when I connect the device using a **USB 2.1 cable** I can't seem to get the combination of Stereo IR + IMU data in a timely manner. 

Using the realsense-viewer when I enable just the stereo module it works fine, as soon as I enable the motion module, I get messages such as "Frame didn't arrive within 5 seconds"... How do I configure the device so both stereo and motion module are working in USB 2.1 mode?

MartyG-RealSense, 2023-11-11:
Hi @chutsu  I was not able to replicate your issue in USB 2.1 mode.

Does the **No Frames Received** error still occur if the depth FPS speed is reduced from 10 to 6?

The frame drop graph in the log can be disregarded as its information will not be helpful to diagnosis of the issue.

If you are using a USB 2.1 cable then this likely means that you are using your own choice of USB cable rather than the official USB 3 one supplied with the camera.  It is important to use a high quality cable with RealSense cameras that is designed for data transfer.  An inexpensive charging / data sync cable may not be able to perform correctly.

chutsu, 2023-11-12:
> Does the No Frames Received error still occur if the depth FPS speed is reduced from 10 to 6?

Problem still occurs when FPS is down to 6.

> If you are using a USB 2.1 cable then this likely means that you are using your own choice of USB cable rather than the official USB 3 one supplied with the camera. It is important to use a high [...]

Fully aware, I'm working with hardware constraints that prevent me from using USB 3.

> When using SDK version 2.54.1, the camera firmware version installed on your camera should be 5.15.0.2 rather than 5.13.0.50 which is designed for use with SDK 2.51.1. Using an older firmware than 5.15.0.2 with 2.54.1 may result in errors.

Upgraded device firmware to match recommended 5.15.0.2

[image]

> It is possible to use the SDK with an unsupported kernel but there may be unforeseen consequences in regards to stability. The kernel can be bypassed by building the SDK from source code with CMake using the -DFORCE_RSUSB_BACKEND=TRUE build flag so that the SDK is not dependent on Linux versions or kernel versions and does not need to be kernel patched.

Compiled `librealsense` from source and still face the same issues. Particularly I've noticed that even with just the Motion Module turned on the number of "dropped frames" is large. It does not report IMU measurements in a timely manner:

[image]

MartyG-RealSense, 2023-11-12:
Yes, my computer's test environment is Windows, though I'm very familiar with how the camera behaves in Ubuntu.

Which computer / computing device are you using the camera with, please?  For example, IMU problems with Raspberry Pi is a known issue whilst depth and color work fine on it.

chutsu, 2023-11-12:
I'm using a Framework Laptop 11th Gen Intel. So compute shouldn't be the issue here.

MartyG-RealSense, 2023-11-12:
I note in the Viewer log that you are receiving **Resource temporarily unavailable** messages, indicating a communication problem between the camera and computer (typically, a data logjam at the USB port).  

Not all computers' USB ports are made equal, despite USB being a specification, and some computers' ports behave better with RealSense cameras than others.

Do you have a USB hub that you could attach the camera to?  A hub will have its own USB controllers separate from the ones used in the computer's own built-in USB ports.

chutsu, 2023-11-12:
Thank you for the suggestion, I used an external USB hub as you suggested and it seems to be working now. For reference I used an Anker 4-Port Ultra Slim USB 3.0 Data Hub. [[Amazon Link](https://www.amazon.co.uk/Anker-4-Port-Macbook-Surface-Notebook/dp/B00Y25XFGK)]

Thanks for the help :)

closed 2023-11-12 by chutsu

MartyG-RealSense, 2023-11-12:
You are very welcome.  It's great to hear that a USB hub resolved your issue.  Thanks very much for the update and the details about the hub that worked for you.  :)