realsenseai/librealsense #12404: Segmentation fault in pyrealsense2 Raspberry PI

pfcouto, 2023-11-15:
Hello I am trying to install librealsense and pyrealsense2 in a raspberry pi compute module. I followed the guide below. When I tried to run the code below the guide I get `Segmentation Fault` like in the print. This is due to the line with `pipeline.start()`. How can I fix this issue? 

I even tried increasing the swap size to 4GB, but still got the error `Segmentation Fault`

And is there a better way to save the data I am trying to save (RGB image and depth map)? Thanks!

pyrealsense2 version: 2.54.1.5217
Raspberry PI info:
`cat /etc/os-release`:
```
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```
---
`hostnamectl`:
```
Static hostname: raspberrypi
         Icon name: computer
        Machine ID: 81a7b702fe8a48c8af6dd77a976a1959
           Boot ID: 97fdb8cea5cf43c5bfe115ce154a8fd5
  Operating System: Debian GNU/Linux 11 (bullseye)
            Kernel: Linux 5.15.61-v8+
      Architecture: arm64
```

# INSTALACAO REALSENSE

```
sudo apt-get update  //&& sudo apt-get dist-upgrade

sudo apt-get install automake libtool vim cmake libusb-1.0-0-dev libx11-dev xorg-dev libglu1-mesa-dev
```

# Expand the filesystem by selecting the`Advanced Options`menu entry and then "Expand filesystem", then flick finish, and select yes to reboot:

```

sudo raspi-config
```

# Increase swap to 2GB by changing the file below to`CONF_SWAPSIZE=2048`:

```

sudo nano /etc/dphys-swapfile
sudo /etc/init.d/dphys-swapfile restart swapon -s

cd ~
git clone https://github.com/IntelRealSense/librealsense.git
cd librealsense
sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/

sudo su
udevadm control --reload-rules && udevadm trigger
exit
```

# Modify the path by adding the following line to the`.bashrc`file:

```
echo "export LD_LIBRARY_PATH=/usr/local/lib:$LD_LIBRARY_PATH" >> .bashrc
source ~/.bashrc

cd ~
git clone --depth=1 -b v3.10.0 https://github.com/google/protobuf.git
cd protobuf
./autogen.sh
./configure
python --version # 3.9.2
make -j1
sudo make install
cd python
export LD_LIBRARY_PATH=../src/.libs
python3 [setup.py](http://setup.py/) build --cpp_implementation
sudo python3 [setup.py](http://setup.py/) install --cpp_implementation
export PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=cpp
export PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION_VERSION=3
sudo ldconfig
protoc --version

cd ~
wget https://github.com/PINTO0309/TBBonARMv7/raw/master/libtbb-dev_2018U2_armhf.deb
sudo dpkg -i ~/libtbb-dev_2018U2_armhf.deb
sudo ldconfig
rm libtbb-dev_2018U2_armhf.deb

sudo apt install clang llvm
sudo apt-get install libssl-dev
cd ~/librealsense
mkdir build
cd build
export CC=/usr/bin/clang
export CXX=/usr/bin/clang++
cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true -DOTHER_LIBS="-latomic"

make -j2
sudo make install

cd ~/librealsense/build
cmake .. -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=$(which python3)
make -j2
sudo make install
```

# Adicionar a linha ao ficheiro bashrc

```
echo "export PYTHONPATH=$PYTHONPATH:/usr/local/lib" >> .bashrc
source ~/.bashrc

sudo pip install opencv-python==4.5.3.56

sudo pip install pyrealsense2
```

---

```
import pyrealsense2.pyrealsense2 as rs
import numpy as np
import cv2

def captura_realsense():

    pipeline = rs.pipeline()

    # Create a config and configure the pipeline to stream
    # different resolutions of color and depth streams
    config = rs.config()

    config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
    config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)

    colorizer = rs.colorizer()

    # Start streaming
    profile = pipeline.start(config)

    # Getting the depth sensor's depth scale (see rs-align example for explanation)
    depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
    #print("Depth Scale is: " , depth_scale)

    # Create an align object
    # rs.align allows us to perform alignment of depth frames to other frames
    # The "align_to" is the stream type to which we plan to align depth frames.
    align_to = rs.stream.color
    align = rs.align(align_to)

    # when stream is finished, RuntimeError is raised, hence this
    # exception block to capture this
    t_end = time.time() + 4
    while time.time() < t_end:
        try:
            # frames = pipeline.wait_for_frames()
            frames = pipeline.wait_for_frames()
            if frames.size() <2:
                print("Not enough frames")
                return
                # Inputs are not ready yet
        except (RuntimeError):
            pipeline.stop()

        # align the deph to color frame
        aligned_frames = align.process(frames)

        # get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame()
        color_frame = aligned_frames.get_color_frame()

        # validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            print("No frames received")

        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

    

    data = time.strftime("%Y-%m-%d_%H-%M-%S")

    cv2.imwrite(f"/home/pi/Desktop/smartfarm/CAM-INTEL/CAM-INTEL_{idlastsensorinsert}_RGB_smartfarm_{data}" + ".png", color_image)
    np.save(f"/home/pi/Desktop/smartfarm/CAM-INTEL/CAM-INTEL_{idlastsensorinsert}_DEPTH_smartfarm_{data}", depth_image)

captura_realsense()
```

[image]

MartyG-RealSense, 2023-11-15:
Hi @pfcouto  Bullseye is known to have problems with librealsense.  The previous version **Buster** works with it.

pfcouto, 2023-11-16:
Hello @MartyG-RealSense, I have other requirements that require the need of the Bullseye version.

I tried to install an older version of realsense and then checkout to the tag https://github.com/IntelRealSense/librealsense/releases/tag/v2.53.1

Then I tried to install pyrealsense2 with the command `pip install pyrealsense2==2.53.1.4623` but it does not install giving me an error but when I do the command `pip install pyrealsense2==2.54.1.5217` the installation works, but I still got the `Segmentation Fault` error.

Is there a way you can help me? Thanks!

MartyG-RealSense, 2023-11-17:
Another RealSense Pi user at https://github.com/IntelRealSense/librealsense/issues/9943#issuecomment-966048785 who had Bullseye problems resolved it by compiling Buster and then copying 2 binary files from Buster over to Bullseye.  They did not specify which files it was that they copied though.

Raspberry Pi boards are Arm based computers.  Arm files for **pip install pyrealsense2** were only introduced as recently as 2.54.1.  Also, only Python 3.7 to 3.9 can be used with the Arm pip files.

closed 2023-11-29 by MartyG-RealSense