realsenseai/librealsense #12406: Is it possible to get the batch-support version rs.rs2_project_point_to_pixel() and rs.rs2_deproject_pixel_to_point() in python library

Jianghanxiao, 2023-11-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux (Ubuntu 22.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| SDK Version                            |  1.13.15                      | 
| Language                            |  python }                          | 
| Segment			|  {Robot}                          | 

### Issue Description
Currently in the python library, I find that if I want to consider the distort coefficient and get the projection and deprojection matrix, the easiest way is to call the functions: rs.rs2_project_point_to_pixel() and rs.rs2_deproject_pixel_to_point(). However, I find that it doesn't support the batch input like a numpy list of points or a numpy list of pixels with depth. This will make it very slow to call in the python end for a batch of points. Is there some solution for this? For example, I can directly calculate the projection matrix considering the dist coefficient in some way? 

Thanks a lot for all potential help!!!

MartyG-RealSense, 2023-11-15:
Hi @Jianghanxiao  A possibility that you could consider is the **Keep()** instruction, which allows frames to be stored in the computer's memory and then have batch-processed actions performed on all of the frames at the same time after the pipeline has been closed.  Such actions could include application of post-processing filters, depth-color alignment or batch-saving to file.

The main disadvantage of Keep() is that because the frames are stored in the computer's available free memory capacity, it can limit recording sessions to up to only 30 seconds (or 10 seconds on low-end devices such as single board computers).

Examples of Python code for Keep() can be found at https://github.com/IntelRealSense/librealsense/issues/6146

Jianghanxiao, 2023-11-15:
> Hi @Jianghanxiao  A possibility that you could consider is the **Keep()** instruction, which allows frames to be stored in the computer's memory and then have batch-processed actions performed on [...]

Hi, thanks for the quick reply. Maybe my question is not that clear. My current goal is to project some 3d points back to the image coordinate to do some depth check. (Arbitrary 3d points) Therefore, I try finding some efficient way to do this instead of calling the rs2_project_point_to_pixel multiple times in a loop. Im not sure if I fully understand ur answer, but it seems that the answer is about how to keep frames and do post processing. 

Hope that this time the question is clear. Thanks a lot for ur help!!

MartyG-RealSense, 2023-11-16:
Thanks very much for the further information.  The intention behind suggesting Keep() was the idea of performing calculations on all frames at the same time once they had been collected together instead of performing calculations on each frame in a loop.

If you are performing deprojection of 2D to 3D and then projection back from 3D to 2D then there is likely not a more processing-efficient way of doing that, unfortunately.  

The link below describes a different approach that a RealSense user took to getting the 3D world coordinates and then returning to 2D pixel coordinates.

https://community.intel.com/t5/Items-with-no-label/How-to-project-points-to-pixel-Librealsense-SDK-2-0/m-p/479197

Jianghanxiao, 2023-11-17:
Thanks for the answer, I find that opencv provide some projection functions to consider the dist_coef in the meanwhile. I check the results, and it's nearly the same to the results from the projection function from realsense. The advantage is that the opencv version supports the batch input in python. 

Here are some example code
```
# Use the opencv projection
# The width and height are inversed here
pixels = cv2.projectPoints(
    points,
    np.zeros(3),
    np.zeros(3),
    self.intrinsic_matrix,
    self.dist_coef,
)[0][:, 0, :]

return pixels[:, ::-1]
```

closed 2023-11-17 by Jianghanxiao

MartyG-RealSense, 2023-11-17:
It's great to hear that you achieved a solution.  Thanks very much for the update and for sharing your code!