realsenseai/librealsense #12411: Decoding messages in realsense-viewer

shervinkoushan, 2023-11-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.5.0 | 
| Operating System & Version |   PopOs 22 | 
| Platform                                 | Laptop  |
| SDK Version                            |  2.54.2                  | 

### Issue Description
How can we decode these messages?
[image]

MartyG-RealSense, 2023-11-15:
Hi @shervinkoushan  The list of numbers are camera firmware driver logs.  They will only be meaningful to an Intel RealSense camera engineer and so can be disregarded.

The **Frame Drops Per Second** log can also be disregarded as it is typically not a useful tool for analyzing performance.  

The most useful part of the debug window is the warning and error log, which can be viewed by toggling off the firmware log option.

shervinkoushan, 2023-11-15:
I see, thank you for the prompt response @MartyG-RealSense

closed 2023-11-15 by shervinkoushan