realsenseai/librealsense #12416: Distance between objects in real time

willyum00, 2023-11-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D401 | 
| Firmware Version                   | 5.12.14.100 | 
| Operating System & Version |  Win 11  | 
| Platform                                 | PC |
| SDK Version                            | Latest   | 
| Language                            |  python/opencv  | 

### Issue Description
Hi Marty. 

I want to find the distance between a pointing stick and the surface it points to. In other words, I want to find the distance from the the stick and the point where its "trace" hits.  

I am having trouble figuring out how to detect when and where the trace hits the surface. The surface is non uniform, and messy. So I cannot approximate it with a simple plane equation. 

I should have the surface's depth data since I am streaming everything using my realsense. But I suspect there is still some information missing that might prevent any implementation from working. 

One idea I had was to loop through the xyz points of the trace and loop through the 3D data of the camera stream and "check" if any of the xyz match an xyz on the surface, but this seems like it would have very inefficient time complexity. 

I suppose this problem can be likened to detecting collisions between objects and point clouds. 

I was also looking into triangulation, but not sure if it applies here. 

If you have any ideas, pointers, or any ideas on how to minimize runtime, I would really appreciate the help. 

Thanks!

MartyG-RealSense, 2023-11-16:
Hi @willyum00  If the camera is on the end of the pointing script then all that you might need is a script that returns the depth value in meters of the depth coordinate at the center of the camera's view, like the RealSense SDK's C++ **rs-hello-realsense** example.

https://github.com/IntelRealSense/librealsense/tree/master/examples/hello-realsense

For Python there is the **distance_to_object** example.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

The most processing-efficient approach for getting the depth value of a single specific coordinate without burdening the computer's CPU with depth-color alignment is likely to be to use the **rs2_project_color_pixel_to_depth_pixel** SDK instruction to convert a 2D color pixel to a 3D depth pixel - see https://github.com/IntelRealSense/librealsense/issues/5603#issuecomment-574019008 for a Python example of the instruction.

willyum00, 2023-11-16:
I think I have found a solution. The camera is not on the end of the pointer stick, it's on the side. I can change the position of the camera but I cannot put it on the end of the stick due to restrictions. 

A different question I have now, but still related to my current task is:

I can easily obtain the z distance in mm from each pixel on the depth camera using get_depth. However, I have not been able to figure out how to get x-y distance data from the camera. In other words, how can I make it so that if I were to measure the distances between objects in x,y coordinates of the camera screen, how can I make it accurate to the real world in millimeters rather than pixels?

MartyG-RealSense, 2023-11-17:
If you wish to make real-world measurements between objects instead of between the camera and an object, there is a Python script for doing so at the link below.

https://github.com/soarwing52/RealsensePython/blob/master/separate%20functions/measure_new.py

This is a Python conversion of a C++ SDK example called **rs-measure**.

https://github.com/IntelRealSense/librealsense/tree/master/examples/measure

MartyG-RealSense, 2023-11-26:
Hi @willyum00  Do you require further assistance with this case, please?  Thanks!

willyum00, 2023-11-28:
Hi Marty, yes. I am looking at the code now and it's trying to open a certain bag file. I don't know what bag files are. Also, I want to run this in real time, not a recording. How would I modify this code to do this? Thank you!

willyum00, 2023-11-29:
I was able to get the code to take the live stream from a camera, but it only works one frame at a time, and hangs on a single frame until I press "q"

willyum00, 2023-11-29:
Also, there is a strange blue tint on the image, I'm not sure what's causing that.

willyum00, 2023-11-29:
I was able to solve the blue tint problem using color spaces,  from this article: https://realpython.com/python-opencv-color-spaces/.

Not really sure why this would happen though. I am still trying to figure out how to make this code work with a live stream video.

MartyG-RealSense, 2023-11-29:
Bag files are like a video recording that contains raw camera data.  RealSense SDK scripts can treat the data in a bag file as though it is coming from a live camera, and so bag files can be used by scripts with minimal changes to scripting.  

By default the SDK uses a live camera, and a **cfg.enable_device_from_file()** instruction must be deliberately added to scripts in order for them to read data from a bag instead.  So if a script contains cfg.enable_device_from_file(), it can be changed to a live-camera script by removing this instruction.

In regard to the blue tint, OpenCV uses **BGR8** as its default colorspace, so a RealSense script that uses OpenCV code should use BGR8 instead of RGB8 in order to avoid the colors of the image being reversed.  https://github.com/IntelRealSense/librealsense/issues/9304#issuecomment-869712832 has further information about this.

willyum00, 2023-12-02:
Thanks for the help

closed 2023-12-02 by willyum00

MartyG-RealSense, 2023-12-02:
You are very welcome.  I'm pleased that I could help.  Thanks very much for the update!