realsenseai/librealsense #12419: No device connected

niknadzirie, 2023-11-17:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435i  | 
| Firmware Version                   | 5.15.0.2 | 
| Operating System & Version |   Ubuntu 22.04.3 LTS (Jammy Jellyfish)  | 
| Kernel Version (Linux Only)    |  5.10.110-rockchip-rk3588                                         | 
| Platform                                 | Orange PI 5B  |
| SDK Version                            |  2.54.2                         | 
| Language                            |  python                         | 
| Segment			|  others                          | 

### Issue Description
Camera works well with realsense-viewer, but when i try with python code it state that no device connected. I installed the SDK using LibUVC-backend installation method.

MartyG-RealSense, 2023-11-17:
Hi @niknadzirie  Have you installed the RealSense Python compatibility wrapper, please?  On Arm based devices such as Orange Pi, you can install the wrapper with the pip command **pip install pyrealsense2** if you have Python version 3.7, 3.8 or 3.9.

niknadzirie, 2023-11-17:
> Hi @niknadzirie Have you installed the RealSense Python compatibility wrapper, please? On Arm based devices such as Orange Pi, if you are using SDK 2.54.1 or newer (you are using 2.54.2, so that is fine) then you can install the wrapper with the pip command **pip install pyrealsense2** if you have Python version 3.7, 3.8 or 3.9.

Yes, i using python 3.9 and already installed pyrealsense2 using pip

niknadzirie, 2023-11-17:
> is there any difference if you change your script's pyrealsense2 import instruction to **import pyrealsense2.pyrealsense2 as rs**

no difference, still 'no device connected'

MartyG-RealSense, 2023-11-17:
When you run a Python script, is the RealSense Viewer closed?  The Viewer should be inactive when a Python script is launched.

niknadzirie, 2023-11-17:
> When you run a Python script, is the RealSense Viewer closed? The Viewer should be inactive when a Python script is launched.

The viewer already closed when i run the python script

MartyG-RealSense, 2023-11-17:
The recommended firmware driver for librealsense SDK version 2.54.2 is **5.15.1.0** rather than 5.15.02 (which is the recommended firmware for SDK 2.54.1).  

Could you try a firmware update please in order to eliminate the firmware version as the cause of the Python problem.

niknadzirie, 2023-11-18:
> The recommended firmware driver for librealsense SDK version 2.54.2 is **5.15.1.0** rather than 5.15.02 (which is the recommended firmware for SDK 2.54.1).
> 
> Could you try a firmware update please in order to eliminate the firmware version as the cause of the Python problem.

Hi, i already update camera firmware to 5.15.1 still have problem with python script => no device connected

niknadzirie, 2023-11-18:
Also got libGL error

fireinice, 2023-11-18:
I encounter the same problem by using `pip install pyrealsense2` exactly as you described, except no libgl error.
but it is now working by using `conda install -c conda-forge librealsense`

notice start with a fresh environments with python3,9 and do **NOT** install pyrealsense2 by pip but only librealsense which also provides python wrapper.

MartyG-RealSense, 2023-11-18:
@fireinice  Thanks so much for sharing the method that worked for you in your particular situation!

@niknadzirie  The 'resource temporarily unavailable' type of the **control_transfer returned** warning can indicate that the computer is having difficulty keeping up with the volume of new data arriving from the camera at the USB port of the computer.  This could be because of an issue on the port, such as the brand of **USB controller** chosen by the computer's manufacturer.  Since USB hubs have their own separate USB controller which is likely a different brand from the computer's one, attaching the camera to a hub instead of directly to the computer's USB port can sometimes make a positive difference to performance.

fireinice, 2023-11-18:
> @fireinice Thanks so much for sharing the method that worked for you in your particular situation!
>
> @niknadzirie The 'resource temporarily unavailable' type of the **control_transfer returned** [...]

There should be some compatible problem for pyrealsense2. My OS is Debian bookworm, with libusb version 1.0.26 which meets the requirements by conda lirealsense described. I'm not sure which library is not compatible with the pyrealsense2

MartyG-RealSense, 2023-11-18:
Hi @fireinice  Are you asking about the libusb library, please?  If you are then the official instructions for installing librealsense from source code recommend libusb version **1.0.0**.

If pyrealsense2 is installed from source code and a PYTHONPATH has to be set, I recommend using the command **which python3** to find the path to use, as the path for Conda can differ from the one for Python 3.

fireinice, 2023-11-18:
> Hi @fireinice Are you asking about the libusb library, please? If you are then the official instructions for installing librealsense from source code recommend libusb version **1.0.0**.
>
> If [...]

the pyrealsense2 does not work on debian bookworm, but work on a conda environment. that's weired.

niknadzirie, 2023-11-20:
> I encounter the same problem by using `pip install pyrealsense2` exactly as you described, except no libgl error. but it is now working by using `conda install -c conda-forge librealsense=2.54.2=py39*`
> 
> **Notice**: starting with a fresh environments with python3,9 and do **NOT** install pyrealsense2 by pip but only `librealsense` which also provides python wrapper.

Thank you for the suggested method using conda!!!!!! Now i already can run the python script.

closed 2023-11-20 by niknadzirie